eyal0/OctoPrint-PrintTimeGenius #21: Read M commands from the printer and store

eyal0, 2018-07-09:
Those m commands set the printer's jerk, acceleration, etc.  By saving them, we can add them to scripts that do analysis

eyal0, 2018-07-12:
We want to be able to notice whenever the user or a script sends the option to the printer.  We also want to record instances when there might have been an M503 command and the printer outputs all the settings.

We need a way to store those settings so that we can recall them when analyzing a file.  I would just put it on the front of the file but that would modify all the progress numbers.

If the progress output was in bytes instead of percentage, the analyzer could attach it to the front of the file, analyze, and then subtract from the progress.  That seems like a lousy solution.

A better solution would be to have those options as a separate input.  Currently there is a `{gcode}` tag.  We would add another tag, maybe called `{config}` and it would be a file with all the commands saved in it.  I guess that the analyzer would store this file before analyzing?  Having to make a temporary file isn't great.

jneilliii, 2018-07-14:
Found it...I'm using this hook in the BedLevelVisualizer Plugin to get the mesh data back from the server.  Seems like there is a built in helper function though shown in the example given; `from octoprint.util.comm import parse_firmware_line`. Wonder if that would work for you?

https://github.com/OctoPrint/Plugin-Examples/blob/master/read_m115_response.py

jneilliii, 2018-07-14:
Or another approach would be building the whole thing out like EEPROM Editor does...

https://github.com/amsbr/OctoPrint-EEPROM-Marlin/blob/16f9242342d4e5ecbae285c793e99f8da383178d/octoprint_eeprom_marlin/static/js/eeprom_marlin.js#L203

closed 2018-07-17 by eyal0