eyal0/OctoPrint-PrintTimeGenius #23: Plugin missing?

TheEagle13, 2018-07-13:
I installed via the plugin manager install from URL section, and everything seemed to install properly (according to the terminal window.) I then restarted Octoprint, but the plugin appears to be nowhere to be found. So I tried installing again, and it told me it was already installed, but re-installed. I then restarted the entire RaspberryPi to make sure it wasn't something that needed a full restart, but I still cannot find the Plugin anywhere (Either in the plugin manager list or elsewhere). I'm sure I probably missed some instruction somewhere, but I'd greatly appreciate your help figuring out what went wrong. 

Thanks!

eyal0, 2018-07-14:
That often means that there was an error during start-up of the plugin. Sorry about that, it's probably my fault. Can you download the log called OctoPrint.log and see if there are any error messages in there?

TheEagle13, 2018-07-14:
Matter of a fact, yes.... Let me know if you need more or want me to try something to troubleshoot.
Thanks!

`2018-07-13 21:49:18,347 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip
2018-07-13 21:49:18,555 - octoprint.plugin.core - INFO - Plugin Virtual Printer is disabled.
2018-07-13 21:49:37,367 - octoprint.plugin.core - INFO - Plugin Virtual Printer is disabled.
2018-07-13 21:49:38,238 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py:1643: UserWarning: No PKG-INFO found for package: octoprint_PrintTimeGenius
  warnings.warn('No PKG-INFO found for package: %s' % self.package_name)

2018-07-13 21:49:38,419 - octoprint.plugin.core - ERROR - Error loading plugin PrintTimeGenius
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 829, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 7, in <module>
    from octoprint.printer.estimation import PrintTimeEstimator
ImportError: cannot import name PrintTimeEstimator
2018-07-13 21:49:38,750 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. Please restart OctoPrint.
2018-07-13 21:50:14,060 - octoprint.server.api.system - INFO - Performing command for core:restart: sudo service octoprint restart
2018-07-13 21:50:14,258 - octoprint.server - INFO - Shutting down...
2018-07-13 21:50:14,660 - octoprint.plugins.HeaterTimeout - INFO - Shutting down...
2018-07-13 21:50:14,660 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2018-07-13 21:50:14,671 - octoprint.events - INFO - Event loop shut down
2018-07-13 21:50:14,685 - octoprint.server - INFO - Goodbye!
2018-07-13 21:50:16,618 - octoprint.startup - INFO - ******************************************************************************
2018-07-13 21:50:16,619 - octoprint.startup - INFO - Starting OctoPrint 1.3.8
2018-07-13 21:50:16,619 - octoprint.startup - INFO - ******************************************************************************
2018-07-13 21:50:17,891 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2018-07-13 21:50:18,633 - octoprint.startup - INFO - Blacklist processing done
2018-07-13 21:50:20,860 - octoprint.plugin.core - ERROR - Error loading plugin PrintTimeGenius
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 829, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 7, in <module>
    from octoprint.printer.estimation import PrintTimeEstimator
ImportError: cannot import name PrintTimeEstimator
2018-07-13 21:50:21,712 - octoprint.plugin.core - INFO - Found 23 plugin(s) providing 22 mixin implementations, 19 hook handlers
2018-07-13 21:50:21,822 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2018-07-13 21:50:21,834 - octoprint.server - INFO - Intermediary server started
2018-07-13 21:50:21,834 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-07-13 21:50:22,497 - octoprint.plugin.core - ERROR - Error loading plugin PrintTimeGenius
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 829, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 7, in <module>
    from octoprint.printer.estimation import PrintTimeEstimator
ImportError: cannot import name PrintTimeEstimator
`

daverdfw, 2018-07-14:
same error here

2018-07-14 04:48:23,915 - octoprint.plugin.core - ERROR - Error loading plugin PrintTimeGenius
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 829, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 7, in <module>
    from octoprint.printer.estimation import PrintTimeEstimator
ImportError: cannot import name PrintTimeEstimator

eyal0, 2018-07-14:
Ah, maybe your OctoPrint version is too old? I'm using brand new features from version 1.3.9rc1. I specified that is the requirement on the website but maybe I should also add that to the plugin somewhere.

Perhaps I could even handle it more gracefully and provide at least partial functionality for older versions. I'll look into it.

TheEagle13, 2018-07-14:
I don't think you necessarily need to support older versions, but I do feel you should support the most recent Regular release, and not strictly Release Candidates (Since RCs are still not official regular releases)

daverdfw, 2018-07-14:
upgraded to the RC of Octoprint and its installed now. Trying first print with it.

eyal0, 2018-07-14:
I'll work on a compatibility mode in the meantime for released OctoPrint.

If you can, perhaps check the logging to see that there were no errors. The plugin hasn't had exposure to many users and environments yet.

eyal0, 2018-07-15:
This is a duplicate of https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/12

closed 2018-07-15 by eyal0