eyal0/OctoPrint-PrintTimeGenius #24: Support for Simplify3D

vincent-ogury, 2018-07-14:
Hi, I'm trying this plugin to get more accurate estimated time and time left.

I have done a test print,
at the end of the GCODE I've got:

```
; Build Summary
;   Build time: 0 hours 23 minutes
;   Filament length: 1495.4 mm (1.50 m)
;   Plastic volume: 3596.78 mm^3 (3.60 cc)
;   Plastic weight: 4.50 g (0.01 lb)
;   Material cost: 0.11
```

The estimated time report by Simplify3D is 23 min.

The log for PrintTimeGenius is:
```
2018-07-14 08:53:29,240 Running: /home/pi/oprint/bin/python2 /home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py "/home/pi/.octoprint/uploads/test.gcode"
2018-07-14 08:53:29,465 Result: {}

2018-07-14 08:53:29,493 Merged result: {'estimatedPrintTime': 1437.0149036093135, 'printingArea': {'maxZ': 15.24, 'minX': 0.0, 'maxX': 165.2, 'maxY': 177.692, 'minY': 0.0, 'minZ': 0.2}, 'dimensions': {'width': 165.2, 'depth': 177.692, 'height': 15.040000000000001}, 'filament': {'tool0': {'volume': 0.0, 'length': 1540.3699000000167}}}
2018-07-14 08:53:30,676 Running: /home/pi/oprint/bin/python2 /home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py marlin-calc "/home/pi/.octoprint/uploads/test.gcode"
2018-07-14 08:53:32,460 Result: {"progress": [[0, 1466.3249773765344], [0.01524872407617128, 1461.245413164425], [0.0154607544289399, 1454.9430635151393], [0.03631456324241805, 1394.3309086797135], [0.04919852526653463, 1334.0202302834023], [0.07199802437600715, 1273.811949325037], [0.09269218680622421, 1213.3573441162794], [0.10929790514011464, 1152.5829148341647], [0.12587867872662054, 1092.227989505862], [0.14279870087755622, 1032.0685403157038], [0.15975115120009178, 971.8368054068674], [0.1767734468153041, 911.3491982163441], [0.20014418063988265, 850.8051963023566], [0.22824942751804753, 790.7701853169332], [0.2805261346118348, 730.7598928657432], [0.334499084527771, 670.5953975783674], [0.38684563691423496, 610.5880518805154], [0.44332303946757934, 550.1809075220499], [0.4906457197307963, 489.98570664904787], [0.5461702329340511, 429.9181507390317], [0.6016897571878289, 369.72721527719204], [0.6636674765394651, 309.72642562630904], [0.726473361504268, 249.58239533465735], [0.7932304944547827, 189.57704612657994], [0.8750542548255614, 129.5657518488233], [0.9531787091592123, 69.34013640893272], [0.9981041991987747, 4.522772884289225], [0.9986030941464655, 3.318527945748201], [1, 0]], "lastFilament": 0.9986030941464655, "firstFilament": 0.0154607544289399, "estimatedPrintTime": 1466.3249773765344}

2018-07-14 08:53:32,464 Merged result: {'printingArea': {'maxZ': 15.24, 'minX': 0.0, 'maxX': 165.2, 'maxY': 177.692, 'minY': 0.0, 'minZ': 0.2}, 'dimensions': {'width': 165.2, 'depth': 177.692, 'height': 15.040000000000001}, u'firstFilament': 0.0154607544289399, u'progress': [[0, 1466.3249773765344], [0.01524872407617128, 1461.245413164425], [0.0154607544289399, 1454.9430635151393], [0.03631456324241805, 1394.3309086797135], [0.04919852526653463, 1334.0202302834023], [0.07199802437600715, 1273.811949325037], [0.09269218680622421, 1213.3573441162794], [0.10929790514011464, 1152.5829148341647], [0.12587867872662054, 1092.227989505862], [0.14279870087755622, 1032.0685403157038], [0.15975115120009178, 971.8368054068674], [0.1767734468153041, 911.3491982163441], [0.20014418063988265, 850.8051963023566], [0.22824942751804753, 790.7701853169332], [0.2805261346118348, 730.7598928657432], [0.334499084527771, 670.5953975783674], [0.38684563691423496, 610.5880518805154], [0.44332303946757934, 550.1809075220499], [0.4906457197307963, 489.98570664904787], [0.5461702329340511, 429.9181507390317], [0.6016897571878289, 369.72721527719204], [0.6636674765394651, 309.72642562630904], [0.726473361504268, 249.58239533465735], [0.7932304944547827, 189.57704612657994], [0.8750542548255614, 129.5657518488233], [0.9531787091592123, 69.34013640893272], [0.9981041991987747, 4.522772884289225], [0.9986030941464655, 3.318527945748201], [1, 0]], 'estimatedPrintTime': 1466.3249773765344, u'lastFilament': 0.9986030941464655, 'filament': {'tool0': {'volume': 0.0, 'length': 1540.3699000000167}}}
2018-07-14 08:53:32,796 Compensated result: {'analysisFirstFilamentPrintTime': 11.381913861395105, 'printingArea': {'maxZ': 15.24, 'minX': 0.0, 'maxX': 165.2, 'maxY': 177.692, 'minY': 0.0, 'minZ': 0.2}, 'dimensions': {'width': 165.2, 'depth': 177.692, 'height': 15.040000000000001}, 'analysisPrintTime': 1466.3249773765344, 'analysisLastFilamentPrintTime': 1463.0064494307862, u'firstFilament': 0.0154607544289399, u'progress': [[0, 1466.3249773765344], [0.01524872407617128, 1461.245413164425], [0.0154607544289399, 1454.9430635151393], [0.03631456324241805, 1394.3309086797135], [0.04919852526653463, 1334.0202302834023], [0.07199802437600715, 1273.811949325037], [0.09269218680622421, 1213.3573441162794], [0.10929790514011464, 1152.5829148341647], [0.12587867872662054, 1092.227989505862], [0.14279870087755622, 1032.0685403157038], [0.15975115120009178, 971.8368054068674], [0.1767734468153041, 911.3491982163441], [0.20014418063988265, 850.8051963023566], [0.22824942751804753, 790.7701853169332], [0.2805261346118348, 730.7598928657432], [0.334499084527771, 670.5953975783674], [0.38684563691423496, 610.5880518805154], [0.44332303946757934, 550.1809075220499], [0.4906457197307963, 489.98570664904787], [0.5461702329340511, 429.9181507390317], [0.6016897571878289, 369.72721527719204], [0.6636674765394651, 309.72642562630904], [0.726473361504268, 249.58239533465735], [0.7932304944547827, 189.57704612657994], [0.8750542548255614, 129.5657518488233], [0.9531787091592123, 69.34013640893272], [0.9981041991987747, 4.522772884289225], [0.9986030941464655, 3.318527945748201], [1, 0]], 'estimatedPrintTime': 1466.3249773765344, u'lastFilament': 0.9986030941464655, 'filament': {'tool0': {'volume': 0.0, 'length': 1540.3699000000167}}}
```

The print result is:
```
test.gcode
Uploaded: an hour ago
Size: 391.5KB
Model size: 165.20mm × 177.69mm × 15.04mm
Filament: 1.54m
Estimated print time: 00:24:26
Last printed: 33 minutes ago
Last print time: 00:32:05
```

During the print, the  Print Time Left stay at "Still stabilizing ..."

What do this plugin is suppose to do ?
I guest it replace the Estimated print time with a proper time left (more accurate), and did is fix the time left also ?

Is they a training after the installation of this plugin ? or did it learn after the installation ?

is they a bug with the time left ?

Does it take in account the jerk settings ?

Does it work if it can't get the primer estimated time from simplify3d ?

Thanks for clarifiying.

mylife4aiurr, 2018-07-14:
I saw still stabilizing for entire print on Octoprint - Mk3, Slic3r prusa edition.
Slic3r estimate was: 14 hours 57mins

How to you know this pluggin is working? 
Does it replace the built in time estimate of Octoprint? 
Do i have to manually analyze every gcode file 1st?

eyal0, 2018-07-14:
I don't have support for Simplify3d yet. I will add it.

I'm not sure why you are not seeing the estimate. Please check in the OctoPrint logging section for errors in OctoPrint.log or plugin_PrintTimeGenius_engine.log .

The plugin should fix print time estimate and also print time remaining.

Also, please turn on debugging to see what estimate genius is making. You need to add PrintTimeGenius and set logging level to DEBUG and then press the blue plus button and then press save. Now the log will also show the estimate every second.

If you can, press F12 in the browser and look for errors in the console. You will need to reload the browser window with F5.

Yes, it does account for jerk. It uses Marlin code to run a printer simulation, including jerk and acceleration.

Sorry that you had trouble. I will look at it soon.

eyal0, 2018-07-14:
@vincent-ogury can you send me an example of slicer3d entire code so that I can make my plugin support simplify3d? You can make a small gcode and zip it and send it here, please. Thanks!

AurelioB, 2018-07-14:
@eyal0 Here's an example
[pegboard-hooks.gcode.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2195203/pegboard-hooks.gcode.zip)

vincent-ogury, 2018-07-15:
I have found this in the octoprint.log:

```
2018-07-15 04:45:56,623 - octoprint.plugin - ERROR - Error while calling plugin PrintTimeGenius
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 225, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 266, in on_event
    self._current_history["payload"] = payload
TypeError: 'NoneType' object does not support item assignment

```

I'm using Octoprint 1.3.9rc2.

eyal0, 2018-07-15:
@AurelioB Thanks, that was useful!  I added simplify3d support to the analyzer.  Enjoy!

vincent-ogury, 2018-07-15:
Tell me when it is ok, I will give it a shot.

vincent-ogury, 2018-07-15:
Wow already done, nice, I run an analysis

eyal0, 2018-07-15:
Thanks! Perhaps try to reupload any file before you print it, to force a reanalysis. Or you can analyze from the settings.

vincent-ogury, 2018-07-15:
I have upload fresh file. I can see the estimated time (from simplify gcode) set properly.
It is the same as with the simplify3d.
Then a marlin analyze, so it add few minute.

How can it guess the jerk settings of the printer, when the printer is not connected ?

I use JERK_DEVIATION, S_CURVE and ADAPTIVE_STEP_SMOOTHING.
JERK_DEVIATION will replace the JERK entirely.

I guess, more we print, better accurate is the result due to a learning step ?

vincent-ogury, 2018-07-15:
Printing in progress. Does it take G28 and G29 time also in the estimation ?

vincent-ogury, 2018-07-15:
Well I have this:
```
File: Wing.gcode 
Timelapse: -
Filament (Tool 0): 0.90m / 2.72g
Approx. Total Print Time: 00:20:56
Print Time: 00:14:14
Print Time Left: Still stabilizing... 
Printed: 260.8KB / 663.7KB
Current Height: 0.84 / 2.44mm 
Layer: 5 / 13 
```

Still have the stabilizing problem.
Also I check the log after I have set the log to DEBUG and restart octoprint. Nothing ...

vincent-ogury, 2018-07-15:
It say 1.0.3 is load in the log, and I have check the code, the simplify fix is here.

Did you test with 1.3.9RC2 ?

eyal0, 2018-07-15:
I will test with 1.3.9rc2. I am usually using a version that is a little bit newer.

Do you see errors in octoprint.log? Or plugin_PrintTimeGenius log?

vincent-ogury, 2018-07-15:
Also I can't see any errors in the console.log.

vincent-ogury, 2018-07-15:
No error in octoprint.log,
No log during the print in PrintTimeGenius.log.

I'm waiting the print to finish in order to see if it ends well.

vincent-ogury, 2018-07-15:
Print finish.
Estimated 20:56, Real: 24m (pure print) 27:49 (with G28 and G29)
No more logs when the DEBUG is activated.
I see no error also.

vincent-ogury, 2018-07-15:
I've go this if I reanalyze the same file:
```
2018-07-15 08:49:36,079 Running: /home/pi/oprint/bin/python2 /home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py "/home/pi/.octoprint/uploads/Wing.gcode"
2018-07-15 08:49:36,820 Result: {"estimatedPrintTime": 1140.0, "filament": {"tool0": {"volume": 2.17621, "length": 904.8}}}

2018-07-15 08:49:36,822 Merged result: {'printingArea': {'maxZ': 2.44, 'minX': 0.0, 'maxX': 220.273, 'maxY': 168.388, 'minY': 0.0, 'minZ': 0.2}, 'dimensions': {'width': 220.273, 'depth': 168.388, 'height': 2.2399999999999998}, 'estimatedPrintTime': 1140.0, 'filament': {u'tool0': {u'volume': 2.17621, u'length': 904.8}}}
2018-07-15 08:49:38,195 Running: /home/pi/oprint/bin/python2 /home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py marlin-calc "/home/pi/.octoprint/uploads/Wing.gcode"
2018-07-15 08:49:39,522 Result: {"progress": [[0, 1256.0909850190071], [0.00899347266937805, 1251.0114208068976], [0.00911854550591227, 1244.709071157612], [0.03032059846616559, 1184.574104762965], [0.04014396619207656, 1124.3332662163157], [0.06728624316513734, 1064.294315688252], [0.13425171129069283, 1004.1715265230779], [0.15699583875315626, 944.0891809678797], [0.2319703827523087, 884.0251878869594], [0.27154784256714204, 823.341797140728], [0.32373117285948877, 763.3411731633298], [0.38622491921766205, 703.2568644654353], [0.41663969017251223, 643.2213616429326], [0.5003340180458031, 583.1246295559706], [0.5272805928158163, 523.0538200802158], [0.6165987839977398, 461.08371860592024], [0.6396666293900566, 401.0108063198967], [0.7256667117909842, 340.9875143262574], [0.7512212994626283, 280.6542295814145], [0.8348464694145414, 220.6530827212689], [0.8645549467042571, 160.6449884583003], [0.9455683015403088, 100.63848761033978], [0.9872146132159316, 40.63737268330851], [0.9993540355854291, 3.261514739212089], [1, 0]], "lastFilament": 0.9993540355854291, "firstFilament": 0.00911854550591227, "estimatedPrintTime": 1256.0909850190071}

2018-07-15 08:49:39,524 Merged result: {'printingArea': {'maxZ': 2.44, 'minX': 0.0, 'maxX': 220.273, 'maxY': 168.388, 'minY': 0.0, 'minZ': 0.2}, 'dimensions': {'width': 220.273, 'depth': 168.388, 'height': 2.2399999999999998}, u'progress': [[0, 1256.0909850190071], [0.00899347266937805, 1251.0114208068976], [0.00911854550591227, 1244.709071157612], [0.03032059846616559, 1184.574104762965], [0.04014396619207656, 1124.3332662163157], [0.06728624316513734, 1064.294315688252], [0.13425171129069283, 1004.1715265230779], [0.15699583875315626, 944.0891809678797], [0.2319703827523087, 884.0251878869594], [0.27154784256714204, 823.341797140728], [0.32373117285948877, 763.3411731633298], [0.38622491921766205, 703.2568644654353], [0.41663969017251223, 643.2213616429326], [0.5003340180458031, 583.1246295559706], [0.5272805928158163, 523.0538200802158], [0.6165987839977398, 461.08371860592024], [0.6396666293900566, 401.0108063198967], [0.7256667117909842, 340.9875143262574], [0.7512212994626283, 280.6542295814145], [0.8348464694145414, 220.6530827212689], [0.8645549467042571, 160.6449884583003], [0.9455683015403088, 100.63848761033978], [0.9872146132159316, 40.63737268330851], [0.9993540355854291, 3.261514739212089], [1, 0]], 'estimatedPrintTime': 1256.0909850190071, u'lastFilament': 0.9993540355854291, 'filament': {u'tool0': {u'volume': 2.17621, u'length': 904.8}}, u'firstFilament': 0.00911854550591227}
2018-07-15 08:49:40,238 Failed to compensate
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 135, in compensate_analysis
    for ph in print_history]
KeyError: 'firstFilamentPrintTime'
2018-07-15 08:49:40,242 Compensated result: {'analysisPrintTime': 1256.0909850190071, 'analysisFirstFilamentPrintTime': 11.381913861395105, 'printingArea': {'maxZ': 2.44, 'minX': 0.0, 'maxX': 220.273, 'maxY': 168.388, 'minY': 0.0, 'minZ': 0.2}, 'dimensions': {'width': 220.273, 'depth': 168.388, 'height': 2.2399999999999998}, u'progress': [[0, 1256.0909850190071], [0.00899347266937805, 1251.0114208068976], [0.00911854550591227, 1244.709071157612], [0.03032059846616559, 1184.574104762965], [0.04014396619207656, 1124.3332662163157], [0.06728624316513734, 1064.294315688252], [0.13425171129069283, 1004.1715265230779], [0.15699583875315626, 944.0891809678797], [0.2319703827523087, 884.0251878869594], [0.27154784256714204, 823.341797140728], [0.32373117285948877, 763.3411731633298], [0.38622491921766205, 703.2568644654353], [0.41663969017251223, 643.2213616429326], [0.5003340180458031, 583.1246295559706], [0.5272805928158163, 523.0538200802158], [0.6165987839977398, 461.08371860592024], [0.6396666293900566, 401.0108063198967], [0.7256667117909842, 340.9875143262574], [0.7512212994626283, 280.6542295814145], [0.8348464694145414, 220.6530827212689], [0.8645549467042571, 160.6449884583003], [0.9455683015403088, 100.63848761033978], [0.9872146132159316, 40.63737268330851], [0.9993540355854291, 3.261514739212089], [1, 0]], 'estimatedPrintTime': 1256.0909850190071, 'analysisLastFilamentPrintTime': 1252.829470279795, u'lastFilament': 0.9993540355854291, 'filament': {u'tool0': {u'volume': 2.17621, u'length': 904.8}}, u'firstFilament': 0.00911854550591227}```

vincent-ogury, 2018-07-15:
I've got this in the settings file:

```  PrintTimeGenius:
    print_history:
    - analysisFirstFilamentPrintTime: 11.381913861395105
      analysisLastFilamentPrintTime: 1252.829470279795
      analysisPrintTime: 1256.0909850190071
      payload:
        file: /home/pi/.octoprint/uploads/Wing.gcode
        filename: Wing.gcode
        name: Wing.gcode
        origin: local
        path: Wing.gcode
        size: 679604
        time: 1669.8609111309052
      timestamp: 1531644120.163443
```

eyal0, 2018-07-15:
It's having trouble with the print_history . That is stored in .octoprint/config.yml . Maybe the print_history got corrupted somehow due to the old versions?

You can open that file, rename print_history to print_history_old, and then restart OctoPrint and see if it starts working?

The print history was supposed to store how long the heating time took but it's missing and maybe that is due to a previous error.

vincent-ogury, 2018-07-15:
Ok I have remove the print_history section. The analysis is ok.

So I have to print something again ?

eyal0, 2018-07-15:
Seems like the genius estimate is failing, but without error message, which is weird. And when it fails, it's supposed to fallback to using the original algorithm but it isn't doing that either. Weird. I'll try to figure out how that could possibly happen.

vincent-ogury, 2018-07-15:
Ok I will wait to reuse this again then.

eyal0, 2018-07-15:
@vincent-ogury Ah, i see a problem.  My code works with the latest, and it works with 1.3.9rc1, but it doesn't work with 1.3.9rc2.  Something was broken in the middle and then fixes.  I'll try to identify what @foosel did there!

eyal0, 2018-07-15:
@foosel knows about this already and fixed it here: https://github.com/foosel/OctoPrint/issues/2715

Maybe I can make a work around in the meanwhile?  Going back to 1.3.9rc1 would work, too.

vincent-ogury, 2018-07-15:
The estimator issue should have been fixed in the 1.3.9rc2. the problem was on 1.3.9rc1. But they is another fix for the rc3.
I will try again after the next update.

eyal0, 2018-07-15:
Now that we know that rc2 was the problem and I've added simplify3d support, this is closed.

closed 2018-07-15 by eyal0