eyal0/OctoPrint-PrintTimeGenius #25: No such file for marlin-calc estimate?

Billiam, 2018-07-14:
My PrintTimeGenius logs contain:

```
2018-07-14 16:04:24,127 Failed to run '/home/octoprint/src/OctoPrint/venv/bin/python2 /home/octoprint/src/OctoPrint/venv/lib/pytho
n2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py marlin-calc "/home/octoprint/.octoprint/uploads/wheel
_inner.gcode"'
Traceback (most recent call last):
  File "/home/octoprint/src/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 193, i
n _do_analysis
    results_text = subprocess.check_output(shlex.split(command))
  File "/usr/lib/python2.7/subprocess.py", line 219, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
CalledProcessError: Command '['/home/octoprint/src/OctoPrint/venv/bin/python2', '/home/octoprint/src/OctoPrint/venv/lib/python2.7/
site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py', 'marlin-calc', '/home/octoprint/.octoprint/uploads/wheel
_inner.gcode']' returned non-zero exit status 1
```

Running that command manually yields:
```
Traceback (most recent call last):
  File "/home/octoprint/src/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 53, in <module>
    main()
  File "/home/octoprint/src/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 18, in main
    output = subprocess.check_output(cmd)
  File "/usr/lib/python2.7/subprocess.py", line 212, in check_output
    process = Popen(stdout=PIPE, *popenargs, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```

Any suggestions for debugging further?

eyal0, 2018-07-15:
Yup, that's the case. If I provide instructions for building marlin-calc, would you do it and send me the result?

eyal0, 2018-07-15:
https://github.com/eyal0/Marlin/blob/estimate-bugfix-2.0.x/Marlin/Makefile.calc

That's the makefile to build calc. I only know how to build on Linux. If you know how to compile code, you can try that.

If not, tell me about your setup and I will try to help. What operating system?

Billiam, 2018-07-15:
Thanks! That seems to work. This was built on Debian 9, 32 bit: [marlin-calc.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2195476/marlin-calc.zip)

eyal0, 2018-07-15:
Also, please send me the output of: uname -m

That is what I use to select the correct binary.

Others have also reported that they don't see the estimate. I'm trying to get more info. If you see errors in your logs, let me know.

Billiam, 2018-07-16:
Started a new print with Octoprint 1.3.9rc3 and PrintTimeGenius 1.0.5:

One print (which I cancelled) shows the call to the analyzer:
```
2018-07-16 13:19:53,284 Running built-in analysis.
2018-07-16 13:19:54,065 , 0.12872004509, None, linear, 0, genius, 9.75886063018e-05
2018-07-16 13:21:46,736 Running built-in analysis.
2018-07-16 13:22:47,366 Result: {'estimatedPrintTime': 24474.97534900769, 'printingArea': {'maxZ': 32.88, 'minX': 36.432, 'maxX': 163.568, 'maxY': 163.165, 'minY': 54.569, 'minZ': 0.28}, 'dimensions': {'width': 
127.13600000000001, 'depth': 108.59599999999999, 'height': 32.6}, 'filament': {'tool0': {'volume': 62.02234023217766, 'length': 25785.892650000736}}}
2018-07-16 13:22:47,882 Running: /home/billiam/src/OctoPrint/venv/bin/python /home/billiam/src/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py "/home/bill
iam/.octoprint/uploads/moat.gcode"
2018-07-16 13:22:48,765 Result: {"filament": {"tool0": {"volume": 62.0, "length": 25785.9}}}

2018-07-16 13:22:48,767 Merged result: {'estimatedPrintTime': 24474.97534900769, 'printingArea': {'maxZ': 32.88, 'minX': 36.432, 'maxX': 163.568, 'maxY': 163.165, 'minY': 54.569, 'minZ': 0.28}, 'dimensions': {'w
idth': 127.13600000000001, 'depth': 108.59599999999999, 'height': 32.6}, 'filament': {u'tool0': {u'volume': 62.0, u'length': 25785.9}}}
2018-07-16 13:22:49,645 Running: /home/billiam/src/OctoPrint/venv/bin/python /home/billiam/src/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py marlin-calc "/home/billiam/.octoprint/uploads/moat.gcode"
2018-07-16 13:22:57,146 Result: {"progress": [[0, 24140.32285311002], [0.00016839225470377, 24134.879356082045], [0.00028695202959385, 24133.708284578643], [0.00345463666308427, 24073.19148488164], [0.00638354066413407, 24012.475364563164], [0.00974536433356533, 23952.456422806146], [0.01303430800126907, 23889.90719201273], [0.01685168359033051, 23829.64218586423], [0.01984973015712181, 23769.63936085756], [0.02195203789926198, 23709.235241774426], [0.02236647802304586, 23649.2162472759], [0.0226723248679024, 23588.20368302335], [0.02293145376293361, 23526.11084798296], [0.02317438710202536, 23462.489321578552], [0.02340112488517766, 23399.3556602639], [0.02361166711239051, 23337.962650952508], [0.02383030711757309, 23273.60199357334], [0.02404084934478595, 23211.709963774225], [0.02426758712793825, 23151.344726055213], [0.02454311921424146, 23087.898411378024], ....
```

While the next print never did:
```
2018-07-16 13:27:36,138 Running built-in analysis.
2018-07-16 13:27:36,745 , 0.146903991699, None, linear, 0, genius, 0.00015633922198
2018-07-16 13:29:26,662 , 110.075037003, None, linear, 0, genius, 0.000200867457098
2018-07-16 13:29:38,071 , 121.493254185, None, linear, 0, genius, 0.000252653598382
2018-07-16 13:31:43,073 , 246.473582029, None, linear, 0, genius, 0.000305616697421
2018-07-16 13:31:43,725 , 247.137530088, None, linear, 0, genius, 0.000448617064828
.... many more follow.
```

eyal0, 2018-07-16:
The output that you see with None, linear, genius, etc is from the debug logging.  It only happens while you are printing.

I think that OctoPrint might pause analysis while printing, so that it doesn't use up all the CPU.  So maybe you tried to analyze while you were printing and that's why you got no output?

The first output looks fine.

This is how you can know if everything is working okay:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems#is-it-working

Mostly, you should look for the gold star which tells you that there are good estimates.  From this line:

`2018-07-16 13:27:36,745 , 0.146903991699, None, linear, 0, genius, 0.00015633922198`

It looks like genius is not giving a genius print time estimate.  This might be because the analysis did not complete before you started running.  Analysis can take some time, especially if the file is very large.

Billiam, 2018-07-16:
I didn't try any manual analyzing, so that output is just direct from the printing process.

Should I be manually analyzing before running the print?

eyal0, 2018-07-16:
Running a print doesn't cause analyzing, I think.  Analyzing happens whenever you upload a file or when you slice a file on the RPi.  If you have old files that aren't yet analyzed, then you need to re-analyze.  But all new files will get analyzed for you.

I could add a button to re-analyze all files if you think that is something worthwhile.  For now, there is only a button to analyze a single file, in the settings.

I'm not sure why you saw that line `Running built-in analysis`

Billiam, 2018-07-16:
Does that mean that: if I upload a file and print immediately, rather than waiting for an analysis, I'm unlikely to see an estimate from this plugin?

eyal0, 2018-07-16:
Not sure. My analyzer first calls the built in one and then does the custom ones. The built-in one pauses when printing, I think. Mine doesnt. So it's possible that if you start printing while it's still analyzing, it won't finish the analysis and the print will proceed without having genius estimates. I need to test to find out.

I'm not sure what the correct behavior would be. Analyzing does potentially slow the CPU and maybe it slows it so much that printing suffers. In that case, maybe we *do* want to pause.

But maybe the slow down is never that severe so we should never pause?

Finally, I could leave it up to the user and make it so that pause is ignored.

Either way, I ought to run the analyzers as 'nice' so they don't hog the CPU.

I'll look into it.

eyal0, 2018-07-18:
marlin-calc for i686 is in place now.  Let me know if you have further problems.  I also fixed the analyzer to cancel when starting a print, which is how the built-in analyzer already works.

closed 2018-07-18 by eyal0