eyal0/OctoPrint-PrintTimeGenius #27: Version Compatibility Error

andrew-cullen, 2018-07-15:
I'm having trouble installing to plugin from version 1.3.9rc2. The following error message is issured:

> Could not find a version that satisfies the requirement OctoPrint==1.3.9rc1 (from OctoPrint-PrintTimeGenius==1.0.4) (from versions: )
> No matching distribution found for OctoPrint==1.3.9rc1 (from OctoPrint-PrintTimeGenius==1.0.4)

eyal0, 2018-07-15:
Unfortunately, 1.3.9rc2 is broken so only 1.3.9rc1 is supported for now. I hope that 1.3.9rc3 will return it to working again! And soon!

eyal0, 2018-07-15:
@andrew-cullen @TheEagle13 @daverdfw @vincent-ogury @AurelioB @rhaamo Sorry to everyone about the version issues.  I have sorted it out now.

OctoPrint 1.3.9rc1 was working.  1.3.9rc2 broke the code but 1.3.9rc3 will fix it again.  I have adjusted this plugin so it will work around the 1.3.9rc2 bug.  You should now be able to use PrintTimeGenius version 1.0.5 with all versions at lesat 1.3.9rc1 .

if you upload a file or slice it on RPi, it should be analyzed.  And when you try to print an analyzed file, you should see a gold star in the print time.

Good luck!

racenviper, 2018-07-15:
Thanks it is now working.

closed 2018-07-18 by eyal0