eyal0/OctoPrint-PrintTimeGenius #28: Analyze from settings failed

drfish, 2018-07-15:
Installed it on an octopi setup 1.3.9rc2 on a Monoprice MP Select mini v1 (M200 clone), picked a random design to try to analyze just to see what it returned. It failed, I'm analyzing another but I figured a failed test case might still be useful. GCode and log attached (assuming I do it correctly)... Slicing was done with Cura 3.4.1, let me know if there's anything else that'd help.

[plugin_PrintTimeGenius_engine (1).log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2196110/plugin_PrintTimeGenius_engine.1.log)
[elf-rect-175.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2196113/elf-rect-175.txt)

Elf-rect-175.txt is the gcode, but github doesn't like the extension :)

drfish, 2018-07-15:
If it helps, here's another attempt that didn't work (figured once was a fluke, twice is a problem)

[articulated_slug-100p.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2196122/articulated_slug-100p.txt)
[plugin_PrintTimeGenius_engine (3).log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2196123/plugin_PrintTimeGenius_engine.3.log)

eyal0, 2018-07-16:
Because you are using a quite new version of Cura, PrintTimeGenius is already taking the line-by-line analysis from cura without doing the marlin simulation.  So you should still see a gold star next to the print time when you run.

[image]

The marlin simulation might not be working because it isn't compiled for your computer.  I prepared it from Raspberry Pi 3 and also for linux 64-bit.  I need help from others to prepare it for other systems.

If you know how to compile, I can send you instructions.  Here's someone else that tried it: https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/25

eyal0, 2018-07-16:
This may have been fixed by https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/25

Please try again when I push a new version.  Thanks!

brandstaetter, 2018-07-20:
Still happening with 1.3.9 RC 4 and PrintTimeGenius Plugin (1.0.7)
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2214317/plugin_PrintTimeGenius_engine.log)

eyal0, 2018-07-21:
Thanks for the report.

This is because you are using armv6l.  I don't have an armv6l compiler so I don't have an armv6l version.  I only have arm7l.  Maybe arm7l will work for you?

Can you try this command:

`cp "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv6l"
`
That'll copy the arm7l to the arm6l.  Maybe it'll work?  Might not.

If it doesn't work, you can try to compile for arm6l  On your Raspberry pi, the same one that uses PrintTimeGenius:

```
git clone --single-branch -b estimate-bugfix-2.0.x https://github.com/eyal0/Marlin.git
cd Marlin/Marlin
make -f Makefile.calc
```

That should create a file called "calc".  It needs to be renamed to marlin-calc.arm6l and copied to here:

`/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv6l`

If you send it to me or write a PR, I can put it into the build for everyone else to enjoy, too.

brandstaetter, 2018-07-21:
looks good:

`2018-07-21 08:47:21,689 Running built-in analysis.
2018-07-21 08:50:55,563 Result: {'estimatedPrintTime': 6178.64533032558, 'printingArea': {'maxZ': 26.8, 'minX': 0.0, 'maxX': 182.675, 'maxY': 153.051, 'minY': -3.0, 'minZ': 0.0}, 'dimensions': {'width': 182.675, 'depth': 156.051, 'height': 26.8}, 'filament': {'tool0': {'volume': 19.717495575115358, 'length': 8197.582070000804}}}
`

I'll provide a PR...

drfish, 2018-07-27:
> I've got a process to build arm for multiple architectures now, thanks!

Thanks, soon as I get a chance to fire up my printer I'll check it out.
Sorry for the lack of further feedback, day jobs been keeping me from
having time to test anything