eyal0/OctoPrint-PrintTimeGenius #29: Cant find in Plugin Manager nor install manually

jonathanlundstrom, 2018-07-15:
Hi,

I would love to try this plugin but unfortunately I can't find it in the Plugin Manager. I can't install it manually from the provided URL either, when I do so, it says plugin "Unknown" installed and the plugin is nowhere to be found in the UI.

I'm running the latest stable OctoPrint (1.3.8).

Thanks,
Jonathan

mark-hahn, 2018-07-15:
Same here but getting `Command "python setup.py egg_info" failed with error code 255 in /tmp/pip-req-build-bzPGDs/`

eyal0, 2018-07-16:
@Tinchus2009 That is also due to being on an old OctoPrint. I'm surprised that you were able to install at all. The latest release doesn't even let you install with old OctoPrint. Maybe you installed an old release?

Tinchus2009, 2018-07-16:
@eyal0 Old? Mine is 1.3.8.... it is latest available

eyal0, 2018-07-16:
@Tinchus2009 1.3.8 is the latest stable.  This plugin is developed for 1.3.9, which is still the unstable version.  In your settings, you can switch from stable to development, if you want.  But it might be less stable for you.

[image]

Tinchus2009, 2018-07-16:
ok... no thanks, I refer to stay in "stable"version. Maybe you should mark this plugin to be used only in that version so pleople dont get confused

eyal0, 2018-07-16:
I did mark it that way, but despite the marking, it gets offered to people!  I think that OctoPrint doesn't know how to distinguish.

You can try restarting OctoPrint in safe mode and then try to remove the plugin.

closed 2018-07-16 by eyal0