eyal0/OctoPrint-PrintTimeGenius #30: Missing data? [] artifact at end of string

OutsourcedGuru, 2018-07-15:
[image]

I'm seeing what looks to be a yellow pair of square brackets after the `Print Time Left:` value. Not sure if there's something that's supposed to be there or what.

Version: 1.0.5
OctoPrint: 1.3.9rc2

racenviper, 2018-07-16:
[image]
Version 1.0.5
1.3.9rc2

Chrome, Windows 10.

Maybe the font of your browser.

eyal0, 2018-07-16:
The yellow brackets are supposed to be a gold star. Maybe the font of your browser is preventing it? Just a cosmetic issue. The fact that it even tried to show a gold star means that it's working. Great!

eyal0, 2018-07-16:
@racenviper You ought to see a gold star, too, if the advanced line-by-line analyzer worked.  Maybe it didn't?  You can check your log for errors.

eyal0, 2018-07-16:
[image]

closed 2018-07-16 by eyal0

OutsourcedGuru, 2018-07-17:
Safari 11.1, for what it's worth.

Is this a character (Unicode?) or what?

eyal0, 2018-07-17:
Unicode, yeah. I used one from this page called heavy star: http://xahlee.info/comp/unicode_stars.html

But on my mobile phone, I see that the heavy star isn't supported. Now I just use a different star and that one *is* working.

Hopefully the latest version of the plugin is working for you.

OutsourcedGuru, 2018-07-17:
Here's a couple of screencaps of their page.

[image]
[image]

So, some of them work and some of them don't.

eyal0, 2018-07-17:
The one that I chose is working on your page. It's the second row, far left. It seems like the most generic of the stars so it ought to have widest support. Is it working for you now?

OutsourcedGuru, 2018-07-17:
Haven't update anything yet...

[image]

[image]

I wonder if visiting that webpage from the font author put it (the font) in my browser's cache...?

No idea. Ignore, I guess.

eyal0, 2018-07-17:
I don't see the gold star so it looks like it is t using the genius estimate. Maybe check the logs to see why not?

eyal0, 2018-07-17:
Perhaps you need to wait for the analysis to complete?

OutsourcedGuru, 2018-07-17:
The latest restart of OctoPrint looks like this.

```
2018-07-17 09:43:57,817 - octoprint.startup - INFO - ******************************************************************************
2018-07-17 09:43:57,823 - octoprint.startup - INFO - Starting OctoPrint 1.3.9rc2
2018-07-17 09:43:57,827 - octoprint.startup - INFO - ******************************************************************************
2018-07-17 09:44:01,211 - octoprint.startup - INFO - Blacklist processing done
2018-07-17 09:44:01,319 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-07-17 09:44:05,478 - octoprint.plugin.core - INFO - Found 18 plugin(s) providing 18 mixin implementations, 20 hook handlers
2018-07-17 09:44:05,709 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2018-07-17 09:44:05,769 - octoprint.server - INFO - Intermediary server started
2018-07-17 09:44:05,770 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-07-17 09:44:07,168 - octoprint.plugin.core - INFO - Found 18 plugin(s) providing 18 mixin implementations, 20 hook handlers
2018-07-17 09:44:07,257 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2018-07-17 09:44:08,900 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2018-07-17 09:44:08,913 - octoprint.printer.standard - INFO - Using print time estimator provided by PrintTimeGenius
2018-07-17 09:44:09,283 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2018-07-17 09:44:13,033 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2018-07-17 09:44:16,701 - octoprint.util.pip - INFO - Version of pip is 10.0.0
2018-07-17 09:44:16,703 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2018-07-17 09:44:16,705 - octoprint.util.pip - INFO - ==> pip ok -> yes
2018-07-17 09:44:16,732 - octoprint.plugin.core - INFO - Initialized 18 plugin implementation(s)
2018-07-17 09:44:16,761 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO801 = /usr/bin/aplay /home/pi/sounds/r2d2/1.wav
2018-07-17 09:44:16,762 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO802 = /usr/bin/aplay /home/pi/sounds/r2d2/2.wav
2018-07-17 09:44:16,763 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO803 = /usr/bin/aplay /home/pi/sounds/r2d2/3.wav
2018-07-17 09:44:16,764 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO804 = /usr/bin/aplay /home/pi/sounds/r2d2/4.wav
2018-07-17 09:44:16,768 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO805 = /usr/bin/aplay /home/pi/sounds/r2d2/5.wav
2018-07-17 09:44:16,770 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO806 = /usr/bin/aplay /home/pi/sounds/r2d2/6.wav
2018-07-17 09:44:16,773 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO900 = /usr/bin/ssh -t pi@pisense.local "sudo pkill temp-graph.py && sudo /home/pi/scripts/cls.py && sleep 3 && sudo poweroff"
2018-07-17 09:44:16,775 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO902 = /usr/bin/ssh -t pi@picam1.local "sudo pkill mjpg_streamer && sleep 3 && sudo poweroff"
2018-07-17 09:44:16,777 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO901 = /usr/bin/ssh -t pi@pisense.local "/home/pi/run &"
2018-07-17 09:44:16,800 - octoprint.plugin.core - INFO - 18 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Bed Leveling Wizard (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelingwizard
|  Bed Visualizer (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  EEPROM Marlin Editor Plugin (1.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin
|  GCODE System Commands (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_gcodesystemcommands
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Navbar Temperature Plugin (0.9) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_navbartemp
|  OctoPi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/octopi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
[... cut ...]
2018-07-17 10:29:05,720 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-07-17 10:44:05,727 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-07-17 10:48:03,411 - octoprint.server.util.sockjs - INFO - New connection from client: 2600:8801:9908:a200:d1c6:4122:8e14:77a4
```

I disabled it for one print job (to determine what was causing an unrelated bug) but I re-enabled it and restarted everything to make sure that it was running.

Don't worry about it. I'll update your plugin and watch it for a few days.

OutsourcedGuru, 2018-07-18:
I updated to v1.0.6 and it appears to be happier. The gold star shows up now rather than the original green/red circle.