eyal0/OctoPrint-PrintTimeGenius #35: Use last print to correct estimation in case of second exact same print

vincent-ogury, 2018-07-17:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.9rc3

PrintTimeGenius Version (if you know): 1.0.5

What did you try:
I have print a gcode.

What happenned:
Estimated 8h40. Real 8h27.
The second print indicate (without another analyze) 8h40 again.

What did you expect to happen:
I expect a 8h27 estimated time for another print of the same gcode.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

eyal0, 2018-07-17:
Okay.

I want to see how far off the analysis was to determine if the problem was just the estimate or the heat up and cool down time. If you can, can you provide the "print_history" from ~/.octoprint/config.yaml ?

vincent-ogury, 2018-07-18:
```
  PrintTimeGenius:
    print_history:
    - analysisFirstFilamentPrintTime: 11.381913861394423
      analysisLastFilamentPrintTime: 26499.283372704325
      analysisPrintTime: 26502.511838754734
      firstFilamentPrintTime: 231.94330096244812
      lastFilamentPrintTime: 30212.71986413002
      payload:
        file: /home/pi/.octoprint/uploads/Moomin_Bookend_V2-TEXT.gcode
        filename: Moomin_Bookend_V2-TEXT.gcode
        name: Moomin_Bookend_V2-TEXT.gcode
        origin: local
        path: Moomin_Bookend_V2-TEXT.gcode
        size: 8132526
        time: 30262.93536400795
      timestamp: 1531799900.396925
    - analysisFirstFilamentPrintTime: 11.381913861394423
      analysisLastFilamentPrintTime: 14350.741889256875
      analysisPrintTime: 14354.45372391605
      firstFilamentPrintTime: 208.69597792625427
      lastFilamentPrintTime: 16966.431143045425
      payload:
        file: /home/pi/.octoprint/uploads/Quick_Change_Universal_Spool_Hub_Bolt_2.0.gcode
        filename: Quick_Change_Universal_Spool_Hub_Bolt_2.0.gcode
        name: Quick_Change_Universal_Spool_Hub_Bolt_2.0.gcode
        origin: local
        path: Quick_Change_Universal_Spool_Hub_Bolt_2.0.gcode
        size: 9972150
        time: 17010.402679920197
      timestamp: 1531766405.53533
```

eyal0, 2018-07-18:
Looking at the first print, it was extruding (after heating and before cool down) from 232 to 30212 seconds. That's 29980 seconds. Analysis predicted that extruding would take from 11 to 26499, a total of 26488 seconds. So it was low by 13%. It also learned that heating takes 232 seconds.

On the print before that, it ran from 208 to 16966. That's a total of 16758 seconds. The prediction was 11 to 14350, which is a total of 14339. This time it was too low by 17%. And heating took 208 seconds.

So for future estimates, it's going to assume that the heating is actually around 220 seconds and the printing estimate is low by about 15%. Whatever estimate come from the Marlin simulation, the plugin will adjust using 220seconds and 15%.

It's too bad that the estimate is off. Once by 13% and once by 15%. Things that might help:

First, version 1.0.7 reads the acceleration and jerk parameters from the printer and puts them in the simulation. That can make a difference but usually it doesn't change things too much.

Second, are you using a Marlin based printer? Maybe you're on smoothieware? There is another analyzer called gcodestat that is based off smoothieware and not Marlin. Maybe it would be more accurate for you. What firmware do you have?

Finally, I'm doing the simulation using the latest version of Marlin. Maybe your firmware is older and simulating an old one would work better? Someone made a webpage based on the older Marlin and you can find it here: www.gcodeanalyser.com

Those are my ideas. For my printer, simulation was pretty accurate... Let me know if you have something for me to try to make the simulation work more accurately.

vincent-ogury, 2018-07-18:
The analyze is :
```
"/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/shark_cup.gcode" "M201 Y2000 X2000 Z120 E10000
M203 Y250.00 X250.00 Z30.00 E55.00
M205 S0.00 B20000 T0.00
M204 P800.00 R10000.00 T1500.00"```

The J0.02 is removed ??? I have run manually with J0.02 and exactly the same result.

eyal0, 2018-07-18:
Okay, I see.  Maybe that is the cause of the big difference in estimation and actual.

Here's what I need to do:

1. Add jdev to the list of options that are picked up.  The code is [here](https://github.com/eyal0/OctoPrint-PrintTimeGenius/blob/master/octoprint_PrintTimeGenius/__init__.py#L396).  I just need to add the letter "J".  That will put the J in the command line.
2. I need to rebuild marlin-calc for JUNCTION_DEVIATION.  The code to enable it is [here](https://github.com/eyal0/Marlin/blob/estimate-bugfix-2.0.x/Marlin/Configuration_adv.h#L447).  I'm not sure if it's enough for me to uncomment it there or if I need to also add it [here](https://github.com/eyal0/Marlin/blob/estimate-bugfix-2.0.x/Marlin/src/module/hax.h).  I'll try it both ways and see what works.
3. Right now, jdev is a `#define`, which means that I would need to compile two versions of marlin-calc.  I'll see if it's possible for me to convert it to an option so that I can have a single binary that supports both jdev and jerk.  If so, that'll mean less code to compile.  All places where JUNCTION_DEVIATION is used as a compile time constant in [planner.cpp](https://github.com/eyal0/Marlin/blob/estimate-bugfix-2.0.x/Marlin/src/module/planner.cpp#L2201) need to be modified to be a boolean.  Then, by looking at the config (if it has a J or not), I can figure if jdev is on or not.
4. I need to be able to get the jdev and use it in configuring the simulation.  I need to copy the code from [here](https://github.com/eyal0/Marlin/blob/estimate-bugfix-2.0.x/Marlin/src/gcode/config/M200-M205.cpp#L134) and put it [here](https://github.com/eyal0/Marlin/blob/estimate-bugfix-2.0.x/Marlin/src/module/calc.cpp#L295).

I would like to do the same for S_CURVE and ADAPTIVE_STEP_SMOOTHING but there is no way for me to detect in the config if those are enabled or not.  :unamused: That's a problem.  Maybe I can open a bug report with the Marlin people and ask them to add it to the output of M503.  It would be easy for them.

For example, I can detect LIN_ADVANCE [here](https://github.com/eyal0/Marlin/blob/estimate-bugfix-2.0.x/Marlin/src/module/configuration_store.cpp#L2625).  When LIN_ADVANCE is enabled, it says so in the M503 output.  I could grab that, send it to marlin-calc, and let the calculation figure it out.  But S_CURVE isn't in that file.  Nor is ADAPTIVE_STEP_SMOOTHING.

Finally, I'm working from bugfix-2.0.x, not bugfix-1.1.x .  Both of them get the new features like jdev and it's possible that the differences between the estimation on one versus the other would be zero.  But maybe bugfix-1.1.x has a radically different planner.cpp.  I don't know.  One way to test, kind of clumsy, would be to install bugfix-1.1.x, 3d print stuff, then install bugfix-2.0.x, print stuff, and see if the time to print is about the same.  Better would be to port the estimation code to bugfix-1.1.x.  That might be super hard, maybe not.  Getting the calculator to the point it's at now was difficult and hacky so doing it again for 1.1.x might be too much.

vincent-ogury, 2018-07-19:
I have a long running print in progress. I can rerun the analyze with a fix to compare the estimated with the real time spent using JUNCTION_DEVIATION.
It is bad if you need 1 marlin-calc depending of the option activated above. That mean at least 8 differents versions ...
It would be great if the calc handle the option from M20* option.

Another way to do it to describe how can we build the marlin-calc with our own Marlin config. So we build it somewhere locally on our device, and it would be customize for our own config.

eyal0, 2018-07-19:
Turns out that adding junction deviation to marlin-calc was really easy!  I converted the `#define` into a variable.  If I see a "J" in M205, I turn on jdev.  If I see one of XYZE, I turn it off.  This is accurate according to the Marlin source code that parses the options [here](https://github.com/eyal0/Marlin/blob/estimate-bugfix-2.0.x/Marlin/src/gcode/config/M200-M205.cpp#L135).

This was the fix for the simulator: https://github.com/eyal0/Marlin/commit/ef12ae5fb1b0630d103a0687fbacefc9da9dd2ac

So all the steps are done for jdev.  I still need to do LIN_ADVANCE, ADAPTIVE_STEP_SMOOTHING, etc.

vincent-ogury, 2018-07-19:
Nice, I will run this simulation again. How can I install this version ?

eyal0, 2018-07-19:
Okay, I just finished it.  Install this zip file:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/jdev.zip

To make sure that jdev is working, try analyzing a file with different a few different jerk settings and see if the estimated total time changes.  You can run the command like you did last time:

```
"/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/shark_cup.gcode" "M201 Y2000 X2000 Z120 E10000
M203 Y250.00 X250.00 Z30.00 E55.00
M205 S0.00 B20000 T0.00
M204 P800.00 R10000.00 T1500.00"

```

And change the M205 command to include "J0.02" or something like that.

I didn't mess with the S_CURVE_ACCELERATION because according to my understanding, it should have no affect on the timing!  S_CURVE causes the straight acceleration path to instead take an S shaped one but the area under the new path, which is the distance traveled, is still the same.  So it shouldn't make a difference.

Let me know how it goes.  I think that the jerk setting will change your results but maybe not enough to account for the 15% error so far.  15% is a big error!  Is there any other option that you changed that could cause this?

eyal0, 2018-07-19:
One last possibility is that it's because you are on bugfix-1.1.x and not 2.0.x.  I could try to port my code to 1.1.x.

eyal0, 2018-07-19:
Okay, I ported calc to bugfix-1.1.x.  I tested two files and they give the exact same simulation results!  So probably that isn't the problem.  15% is a lot.  Maybe you can turn on the debugging like is described in the settings?  On your next print, it'll fill the logs with comma-separated values that can be imported into a spreadsheet.  Maybe if I see a graph of the amount of error over time, it'll give a good hint as to what may be the problem?

vincent-ogury, 2018-07-19:
First of all, I rerun the simulation and I can't see any parameters after marlin-calc. Then I remember that my printer was off. This is often the case, I shut it down, prepare stuff, then turn it on to print.

Can you save the M2* of the last time you saw them and reuse them if the printer is off ?

I'm running the analyze right now.

vincent-ogury, 2018-07-19:
Here the result:

My print history:
```
    - analysisFirstFilamentPrintTime: 11.381956737473956
      analysisLastFilamentPrintTime: 51627.18556473628
      analysisPrintTime: 51629.43820063222
      firstFilamentPrintTime: 238.36027908325195
      lastFilamentPrintTime: 56847.45377802849
      payload:
        file: /home/pi/.octoprint/uploads/shark_cup.gcode
        filename: shark_cup.gcode
        name: shark_cup.gcode
        origin: local
        path: shark_cup.gcode
        size: 8374428
        time: 56906.040696144104
      timestamp: 1531995697.616347
```

The analyze
```
2018-07-19 20:56:14,055 Subprocess output: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/shark_cup.gcode" "M201 Y2000 X2000 Z120 E10000
M203 Y250.00 X250.00 Z30.00 E55.00
M205 S0.00 B20000 T0.00 J0.02
M204 P800.00 R10000.00 T1500.00"
Junction deviation enabled
..Processed 289025 Gcodes and 742 Mcodes. 285555 blocks
Total time: 51952.792400
...
2018-07-19 20:56:14,420 Recent heat-up times in seconds: 238.360279083, 239.296042919, 231.943300962, 208.695977926
2018-07-19 20:56:14,445 Average heat-up: 229.573900223 seconds
2018-07-19 20:56:14,445 Recent cool-down times in seconds: 58.5869181156, 50.2300150394, 50.2154998779, 43.9715368748
2018-07-19 20:56:14,446 Average cool-down: 50.7509924769 seconds
2018-07-19 20:56:14,446 Time spent printing, actual vs predicted: 56609.0934989/51615.803608, 29706.1376681/26487.9014588, 29980.7765632/26487.9014588, 16757.7351651/14339.3599754
2018-07-19 20:56:14,446 Average scaling factor: 1.12968941262
...
'estimatedPrintTime': 58955.22133908896
```

vincent-ogury, 2018-07-19:
I will need to reset my history in order to have better estimation on the next 2 print.
I can remove all the history except the last one and fix the estimated time. It should be good enought for the next print.

vincent-ogury, 2018-07-19:
I fix my history. I run a print of 4h after the 1.08 linear regression fix.  I will post the result tomorrow.

vincent-ogury, 2018-07-20:
```
    - analysisFirstFilamentPrintTime: 11.565768018763265
      analysisLastFilamentPrintTime: 13182.763743663716
      analysisPrintTime: 13185.93685974154
      firstFilamentPrintTime: 237.9099678993225
      lastFilamentPrintTime: 15205.471760034561
      payload:
        file: /home/pi/.octoprint/uploads/car_mount_v3a.gcode
        filename: car_mount_v3a.gcode
        name: car_mount_v3a.gcode
        origin: local
        path: car_mount_v3a.gcode
        size: 2492469
        time: 15219.05532693863
      timestamp: 1532051249.349296
```

vincent-ogury, 2018-07-20:
I run a short print (<1h), with debug activated. I will send the report tonight.

eyal0, 2018-07-20:
Analysis says 13100 seconds to print but actual was 15000! It's so far off! I don't understand why.

eyal0, 2018-07-20:
I hope that there will be something in the graph that will explain it better.

vincent-ogury, 2018-07-20:
I join the log and the csv

[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2214800/plugin_PrintTimeGenius_engine.log)

[plugin_PrintTimeGenius_engine.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2214804/plugin_PrintTimeGenius_engine.txt)

vincent-ogury, 2018-07-20:
I'm thinking of moving to bugfixes-2.0.X
is it buggy ? and usable ?

vincent-ogury, 2018-07-20:
I have made the bugfix-2.0.x config, I may test the next print with this.

I think the ADAPTIVE_STEP_SMOOTHING may affect the speed on the small parts.

eyal0, 2018-07-21:
[image]

Here's a chart for predicted printing time.  Actual shows what the right answer should be.  We can see that Genius is staying pretty close most of the time.  The built-in OctoPrint one is bouncing around because it assumes that the progress in the file is relative to the print time.  Like 30% of the way through the file means 30% of the print time is done.  That's not accurate.  This is how genius is meant to work!

It's unfortunate that the compensation factor is so large!  That means that the simulation isn't very accurate.  For me it's been very accurate, within just seconds.  I can't figure out what's going wrong!

Is your printer doing anything out of the ordinary that would make the print must slower than the estimation?  Like maybe it pauses each layer to take a photo or something?  Maybe there are some strange g-codes in the file that affect print speed that I didn't take into account....?

Other than the S_CURVE and the others, what other modifications did you make to the firmware?  Did you make a lot of changes to it?  If you send me a diff showing the changes or a repo where you've stored them, I can take a look.

vincent-ogury, 2018-07-21:
I join the diff from bugfixes-2.0.x and from bugfixes-1.1.x. I haven't test the 2.0.x yet. I'm installing TLSmoother right now. I will do test print today.

[tevo-1.1.x.patch.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2216060/tevo-1.1.x.patch.txt)
[tevo-2.0.x.patch.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2216061/tevo-2.0.x.patch.txt)

Also my vref is set to 0.850v for X Z E, and 0.950 for Y.

Previously I have installed the TLSmoother inside the box with the motherboard, with 0.950v for XZE and 1.1 V for Y but the temperature of the TLSmoother was really high, and inside the box it was not good.

I'm installing in outside, only for X and Y. if the temperature is ok, I may raise the vref to 0.950 for XZE and 1.05 for Y.

I think the vref may limit the speed, even if I have a reasonable print speed (50mm/s) setup in simplify.

I will join my simplify3d settings (it doesn't change the acceleration on the fly)

vincent-ogury, 2018-07-21:
This is my simplify config, and a gcode generated by it.
[SimplifyConfig and Gcode.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2216070/SimplifyConfig.and.Gcode.zip)

eyal0, 2018-07-21:
Hey!  Thanks to your diffs, I found a bug!

I noticed that you modified the default acceleration and I wondered, how come when I give "M201" commands to the simulator, it doesn't work?
```

"/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/shark_cup.gcode" "M201 Y2000 X2000 Z120 E10000
M203 Y250.00 X250.00 Z30.00 E55.00
M205 S0.00 B20000 T0.00
M204 P800.00 R10000.00 T1500.00"
```

Try changing the M201, you'll see it has no effect.

That's no good!  It was due to a bug in the marlin-calc where is doesn't recalculate parameters when changing the acceleration.  I'll fix it and submit a patch soon.

eyal0, 2018-07-21:
That's only set to 650 ns. At most, it would happen once per line in the file. Even in a file with a million lines, it would only add less than 1 second.

eyal0, 2018-07-21:
Okay, it's ready.  Also, config will be saved to ~/.octoprint/config.yml .  You can try to analyze a file without connecting to the printer and it will still use your configuration.

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/save_config.zip

If you modify the M201 parameters, you should see that the total estimated time is changed.  Also if you modify jerk or other parameters.  Good luck!

eyal0, 2018-07-22:
I tried recompiling with S_CURVE_ACCELERATION but it had no affect on my print time.  It's possible that my print doesn't use the S_CURVE much so the difference is not noticeable but for me it is very very small.  Maybe just 5 seconds different in print time with S_CURVE.

Maybe something with your printer something is misconfigured so it takes longer to print?  Do you want to try this gcode:

`G1 X0 Y0:G1 X100 F100:G1 X0 Y0 F3000`

It should go to 0,0, then move 100mm, then go back.  The 100mm move should take exactly 60 seconds.  You could time it with a watch and see if it is accurate?

I have a few printers and I'm getting pretty accurate results so far.

vincent-ogury, 2018-07-23:
Here the last print:
```
Actual | Predicted
-- | --

306.5 | 25034.6 | 17.0 | 12.2 | 24443.0 | 4.5
311.7 | 15697.1 | 22.1 | 11.6 | 14613.5 | 5.2
280.2 | 3570.8 | 25.5 | 11.6 | 3527.8 | 2.4
265.9 | 2207.2 | 13.4 | 11.6 | 1996.4 | 3.5
282.5 | 2483.1 | 39.4 | 11.6 | 2386.2 | 3.5
```

I have run your test,
it take exactly 60s to do 100mm at 100mm/min.

vincent-ogury, 2018-07-23:
Why the predicted for warm up is 12 instead of 300 ?

vincent-ogury, 2018-07-23:
Did you fixed the acceleration settings? Mine set to 800.

vincent-ogury, 2018-07-23:
yeh, It is close enough for me ! I mean, it is awesome !

is it mandatory to run the octoprint computation ? The marlin-calc is by far faster.

is it possible to do the marlin-calc directly and if it is not working fall back to octoprint calc ?
can you add global estimated (including all) vs real print (including all) ? so we don't have any compute to do to see how is it accurate ?

eyal0, 2018-07-23:
I also think that the built-in calculation is too slow.  I actually added many submits to the OctoPrint project to try to improve it but I think that it is limited in how fast it can be because it is written in python.  marlin-calc is in c++, same as Marlin.

I'm scared to remove it because I'm afraid that it has features or will develop new features that might be needed...  I will look into it.

As for global estimated, you mean for me to add it to the table?  I will add 2 columns, one that is the total of the numbers on the left and one that is the total of the numbers on the right, after compensation.  Sure, I can do that, it's a good idea, not too hard to do!

I'll send a picture of how it looks when I'm ready.  Let's close this bug because the topic has changed and I'll open two new ones for the new ideas.

closed 2018-07-23 by eyal0

vincent-ogury, 2018-07-23:
Ok thanks.