eyal0/OctoPrint-PrintTimeGenius #36: Weird char instead of gold star

vincent-ogury, 2018-07-17:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.9rc3

PrintTimeGenius Version (if you know): 1.0.5

What did you try:
Print

What happenned:
The gold star is badly display

What did you expect to happen:
A real gold star

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

Can you use the html code instead of the utf8 (I guess) one ?
This code for instance: &#x2606;

[image]

eyal0, 2018-07-17:
I did that in version 1.0.6. Please let me know if that works for you.

vincent-ogury, 2018-07-18:
Ok thanks.

I will check on my next print (long running print in progress ;))

vincent-ogury, 2018-07-18:
fixed
[image]

closed 2018-07-18 by vincent-ogury