eyal0/OctoPrint-PrintTimeGenius #43: Needs more documentation

mosh1, 2018-07-19:
This seems like a really awesome plugin but it really needs more documentation to figure out what's going on.

For instance sometimes the Print Time is stuck at 'Preparing'. Sometimes I get a red dot, sometimes orange and sometimes a gold star. It's not clear why that's  inconsistent  and what these mean. The hover over describes it curtly but it'd be helpful to have more details on the project's README.

Also, is there a history of estimated time and actual time somewhere (a log?). 

Thanks for the work on this plugin!

kazibole, 2018-07-21:
Thanks for all of the great info! One more question: When is the analysis performed on a gcode file?

eyal0, 2018-07-22:
Analysis is done whenever a file is uploaded or sliced on OctoPrint. Analysis is cancelled if printing begins before analysis is complete.

mosh1, 2018-07-23:
Thanks, this is a great explanation. It explains why I don't see the gold star a lot as I generally upload the file and print right away. Keep up the good work on this plugin, really love it!

eyal0, 2018-07-26:
Glad to hear that you like it. Your appreciation helps me continue my efforts.

 The newest version has some features that might help you. First, analysis is much faster. So there is a better chance that you finish analysis before you print. Second, cancelling analysis while printing is now optional. If you see the printer stay to get very slow because of analysis, you shouldn't do this, but the start of the print is usually just heating time so you might be able to do it. Third, there are gold stars in the file list now so you can see when the analysis completes. If it completes while you are printing, the estimate will switch from OctoPrint to Genius midprint.