eyal0/OctoPrint-PrintTimeGenius #45: Bad interaction with "Exclude region" plugin

cmock, 2018-07-20:
OctoPrint Version: 1.3.9rc3

PrintTimeGenius Version (if you know): current as of 2018-07-18

[Exclude Region](https://plugins.octoprint.org/plugins/excluderegion/) allows to exclude regions of a print, typically when one part of a multi-part print comes loose.

That means the real print time no longer matches with any estimate, not only for this print, but also for future prints.

Example: I had one part of a two-part print come loose and excluded it, halving the effective print time. In the post-print analysis, PTG seems to have decided that the original estimates are about double of the real print time, and in later prints (without using "exclude region") estimates are very much off.

This effect may be increased by the fact that I hadn't done many prints with PTG before that problematic print, and only one print after.

I think the problem could be solved by either ignoring print jobs that had regions excluded for tuning the estimation parameters, or by limiting the size of factors used for tuning (that would also help in cases like when the user changes the print speed multiplier directly on the printer during a print).

eyal0, 2018-07-22:
Accommodating Exclude Region would be difficult because there might be many plugins that could affect print time and I would have to maintain a list of all the plugins that affect print time.  Worse, I would have to know what's active and what isn't.

That's hard!

I'm not sure if I could ignore prints that are too far off because it would be hard to know what's too far off.  Some users find the plugin very accurate, some have a large estimate difference.

Maybe I could add a delete button in the settings to remove bad rows from the print_history.  Do you think that would be useful?

TheEagle13, 2018-07-25:
@eyal0 I think being able to delete/remove a print history that is inaccurate would be a great way to work around other developer's plugins. I use a similar plugin that does similar things, and I think that would be useful in the event I do need to use it.

cmock, 2018-07-25:
Yes, being able to delete from the history would be an OK solution.