eyal0/OctoPrint-PrintTimeGenius #48: Enhancement request: Project what local time the print will complete

thehammer2, 2018-07-22:
No issue here, just a simple quality-of-life enhancement request.

Have the plugin add what it calculates as the projected time to complete the print to the current local time and reports a projected local time that completion will occur at.  Bonus points if you can find some way to poll a timeserver during analysis to correct the device's clock to current universal time and then correct for a time zone the user selects from within the plugin settings.

eyal0, 2018-07-22:
Have you considered this: https://plugins.octoprint.org/plugins/octoprint-eta/

or this: https://plugins.octoprint.org/plugins/detailedprogress/

thehammer2, 2018-07-22:
I'm unsure exactly how PrintTimeGenius interacts with the rest of the OctoPrint system. Will those plugins use PrintTimeGenius's estimates to display their ETA? If so, then that's exactly what I'm looking for.

eyal0, 2018-07-22:
Yes, they will.  PrintTimeGenius replaces that estimated print time and also the remaining print time calculations.  Plugins that use those will use the new values.  Should work together great!

thehammer2, 2018-07-22:
Outstanding, then. Thanks again.

I'll go ahead and close this issue.

closed 2018-07-22 by thehammer2