eyal0/OctoPrint-PrintTimeGenius #49: Don't run the built-in analysis, and use marlin-calc instead

eyal0, 2018-07-23:
@vincent-ogury idea from here https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/35#issuecomment-406976268

vincent-ogury, 2018-07-23:
For this one, I'm not sure you use the result of octoprint computation if you got the marlin-calc one.

If you don't use the result of octoprint, you may add an option:

 * force usage of octoprint calculation (to disable the marlin-calc check)

So if people had issue with it, they just skip the try to using it.

Of course, if marlin-calc isn't available, we can fallback to octoprint.

eyal0, 2018-07-23:
The problem is that the built-in analysis also provides other data, like this:

```
2018-07-22 11:18:51,738 Running built-in analysis.
2018-07-22 11:19:03,276 Result: {'estimatedPrintTime': 3279.5535919193026, 'printingArea': {'maxZ': 9.4, 'minX': 0.0, 'maxX': 188.376, 'maxY': 153.877, 'minY': -3.0, 'minZ': 0.0}, 'dimensions': {'width': 188.376, 'depth': 156.877, 'height': 9.4}, 'filament': {'tool0': {'volume': 9.309770858429276, 'length': 3870.552950000058}}}
```

You see, it has volume and length and size and other stuff.  However, I can add these to marlin-calc, no problem.  I also need to deal with multiple extruders.

eyal0, 2018-07-23:
This needs testing with a multi extruder job:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/39

eyal0, 2018-07-24:
Testing with multi-extruder complete.  Looks good, sometimes even better than OctoPrint results and always faster!

vincent-ogury, 2018-07-24:
I install this branch right now and test a print

vincent-ogury, 2018-07-24:
wow blazing fast ! congrats !!!

vincent-ogury, 2018-07-24:
frisbee.gcode
Uploaded: an hour ago
Size: 1.3MB
Model size: 132.26mm × 212.26mm × 2.40mm
Filament: 3.14m / 7.56cm³
Estimated print time: 01:02:21
Last printed: a few seconds ago
Last print time: 01:03:48

eyal0, 2018-07-25:
Yes? You tried the more_print_stats branch? You can try this, also, the same:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/dev.zip

closed 2018-07-25 by eyal0