eyal0/OctoPrint-PrintTimeGenius #51: Can't be installed as it is 'incompatible'

qwertpi, 2018-07-23:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version:
1.3.8
What did you try:
Search for it in the plugin repository to install it
What happened:
Didn't appear until I unticked only show compatible and then it said it was incompatible
What did you expect to happen:
The plugin be compatible and install

kazibole, 2018-07-23:
OctoPrint 1.3.9-rc1 or higher is required.

eyal0, 2018-07-26:
Cool. Let me know if the estimates are bad for you.

qwertpi, 2018-07-26:
The estimates seem ok, they're certainly better than the OctoPrint built-in ones. As much as I generally run beta software I didn't want to run OctoPrint in beta so I reverted back to stable and to my pleasant surprise, PrintTimeGenius still works.

BTW I was affected by the printer slowing down bug but am happy to report the fix in the recent update worked