eyal0/OctoPrint-PrintTimeGenius #53: Feature Request: Compute print time while printing

a-w, 2018-07-23:
OctoPrint Version: 1.3.9rc4

PrintTimeGenius Version (if you know): 1.0.9

What did you try: Upload GCode from Slic3r with option "Start printing". This uploads a GCode and immediately starts printing it.

What happenned: PrintTimeGenius starts analyzing and immediately aborts. This happens every time I start a print job with this workflow. If I upload the GCode separately, then PrintTimeGenius makes me wait around four minutes before I can actually start printing. This is a severe impact in usability.

What did you expect to happen: PrintTimeGenius should perform its calculation concurrently to a print in progress and update the display when a result is available.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

2018-07-23 10:57:27,743 Running built-in analysis.
2018-07-23 10:57:28,243 Probably starting printing, aborting built-in analysis.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 197, in _do_analysis
    results = super(GeniusAnalysisQueue, self)._do_analysis(high_priority)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/analysis.py", line 390, in _do_analysis
    raise AnalysisAborted(reenqueue=self._reenqueue)
AnalysisAborted
2018-07-23 10:57:47,904 Running built-in analysis.
2018-07-23 10:58:10,473 Probably starting printing, aborting built-in analysis.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 197, in _do_analysis
    results = super(GeniusAnalysisQueue, self)._do_analysis(high_priority)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/analysis.py", line 390, in _do_analysis
    raise AnalysisAborted(reenqueue=self._reenqueue)
AnalysisAborted
2018-07-23 10:58:23,568 Running built-in analysis.
2018-07-23 10:58:53,313 Probably starting printing, aborting built-in analysis.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 197, in _do_analysis
    results = super(GeniusAnalysisQueue, self)._do_analysis(high_priority)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/analysis.py", line 390, in _do_analysis
    raise AnalysisAborted(reenqueue=self._reenqueue)
AnalysisAborted
2018-07-23 10:59:08,428 Running built-in analysis.
2018-07-23 10:59:22,934 Probably starting printing, aborting built-in analysis.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 197, in _do_analysis
    results = super(GeniusAnalysisQueue, self)._do_analysis(high_priority)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/analysis.py", line 390, in _do_analysis
    raise AnalysisAborted(reenqueue=self._reenqueue)
AnalysisAborted

eyal0, 2018-07-24:
AnalysisAborted is a feature of OctoPrint. OctoPrint sends a cancel request to all analyzers when printing begins because analyzing while printing can slow down OctoPrint to the point of making prints go slow. I had a user complain of a print that was pausing a lot because of analysis.

https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/34

What do you suggest is a better solution? I could provide a checkbox in the settings to not abort and you could disable the OctoPrint abort request. Maybe I can find a way to run the analysis at lower priority, too, and they could run together with the printing without the degraded print performance?

a-w, 2018-07-24:
Hmm, I understand that these are conflicting requirements. Some thoughts:

- when the print begins, the printer first is heating up, and may be calibrates. That should not take much CPU time on the octoprint host and therefore it should not be too much a problem to run print time calculation concurrently during that time. That could be sufficient for small to medium sized models
- only if print time calculation takes longer than the printer needs to warm up, then both printing and print time calculation will compete for CPU time. So, indeed this will be a problem when printing long and complex model.
- a checkbox would then mean "yes I agree that there will stuttering print and artifacts on perimeters". That could probably generate complaints again.
- IMHO the best way to deal with competing CPU is to give the print time calculation a lower priority. Lowering priority should not have any impact otherwise.
- may be prefixing the PrintTimeGenius commands with `nice` could help.
- is there any means to delay the start of the print with a "Print time calculation in progress, please wait" message?
- I understand that there is only a problem when using a workflow like I do: sending and starting the print directly from the slicer. But at least slic3r offers the possibility to run post-processing scripts which can add comments to the GCode. Any chance to optionally offload the time consuming part to such a post-processing script and have PrintTimeGenius retrieve information from such comments? But that would probably be a plugin of its own.

eyal0, 2018-07-24:
That's a good analysis of the problem.  I think that you are an engineer!

I agree, it could be possible to analyze while still heating.  It would be hard to detect exactly when to cancel the analysis.

I can lower analysis priority, that's a good idea.  I'll use psutil and not nice because the latter is not cross-platform.  I worry that lowering priority won't be enough to solve the problem and for some users it'll still cause problems.  That's why I'd like to leave users with the ability to cancel, still, using the checkbox.  Default would be to cancel while printing.

I don't know if PrintTimeGenius should delay the start of printing.  Users might not want that.  But I could do it.  There is a hook for preprocessing uploaded files.  I could cause all uploading and slicing to wait.  The user experience isn't very good, though.  All of OctoPrint is halted!

Do you use slic3r on OctoPi or not?  slic3r on OctoPi already outputs a bit of analysis but not as good as PrintTimeGenius.  I think that it would be difficult to put the PrintTimeGenius analysis into all the OctoPrint slicers because there are too many of them.

So far, my ideas are:

1. "Nice" all analysis and have an option to analyze while running.
2. Somehow cause the file to be analyzed before it is available for printing so that all files listed in the directory are already analyzed.
3. Analyze before starting a print if analysis isn't done already....?

I think that I can do #1 either way, right?

eyal0, 2018-07-24:
I already took care of point 1, including handing windows.  I hope that I did it right: https://github.com/eyal0/OctoPrint-PrintTimeGenius/commit/208773ff1be7ff305175dc59753601295686d2ee

The gold star appears next to the time estimation, while printing, not next to the file itself.  Maybe I could add that?  I'll look into it.

[image]

If you want to add a python script to your own slic3r, that's easy enough!  But I don't know how I would distribute that as a plugin.  You could, for example, have a script on your Windows computer that you run on gcode and it would add into the gcode the Genius analysis.  That analysis would be stored in a comment in the gcode file.

Then you would need to add an analyzer to PTG that would extract the analysis from the comment and use it.  That would be easy to do, too.  But how would I get it to users?  Somehow they'd need to install a plugin to their slic3r on Windows.

The simulation can run without the printer's config but it's not as accurate.  It's still pretty good, even with defaults.  And if you supply the config to the analyzer, it'll use it (just copy and paste the output of M503 on the printer, without all the "echo").

> Since analyzing starts as soon as a file is uploaded, this does not make much sense to me?? Instead, when the user attempts to print a file whose time calculation is still in progress, I would prefer a warning "Print time calculation still in progress" with the options: 1. start now and abort calculation, 2. wait for calculation to complete and start print when done, 3. start now and continue calculation (may reduce print quality)

That's getting really complicated now!  It's too complex.

Like I said, I'll look into adding an icon.  Maybe files that are still being analyzed can get a little spinner on them in the file list and files that are already analyzed can have a gold star?  I think that I looked into it before and it was too hard so I gave up.  Maybe I should look harder!

eyal0, 2018-07-24:
The more that I think about it, the less that I think it would be possible or useful to add a gold star next to the files in the list.  The flies in the list are only updated when the user reloads the list.  And [push updates ](http://docs.octoprint.org/en/master/api/push.html?highlight=push)seem like they aren't so stable so I'm worried about using them....

Maybe it's okay despite that note.  I'll play around with it.  If that worked, then I could put a little star next to analyzed files and maybe a little spinner next to those that are still being analyzed.

eyal0, 2018-07-25:
I learned about push updates but figured it out without push updates anyway.  I'll have something soon.

closed 2018-07-25 by eyal0