eyal0/OctoPrint-PrintTimeGenius #54: PrintTimeGenius causing short pauses during print after first layer

nebbywan, 2018-07-24:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.9rc4

PrintTimeGenius Version (if you know): 1.0.9

What did you try: Printing a simple part (gcode attached)

What happenned: Print hesitated consistently after the first layer at what seemed like were it needed a z-hop causing print malformations. This occurred on two different octoprint setups w/ Prusa MK3's running firmware v3.3.1. Issue corrected itself when this plugin was disabled.

What did you expect to happen: A smooth print.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
[GCODE and Log.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2221522/GCODE.and.Log.zip)

KD7LRJ, 2018-07-24:
I can confirm that this issue has ruined several prints on each of three different printers that I have tested in the last couple of days. With the plugin disabled, things are fine.

PrintTimeGenius 1.0.9 installed on the following:

TronXY P802E with OctoPrint 1.3.9rc4 (generic Raspbian) with original Marlin firmware
CR-10 with OctoPrint 1.3.9rc4 (OctoPi 0.14.0) with firmware TH3D v?
Prusa MK3 with OctoPrint 1.3.9rc4 (OctoPi 0.15.0) with firmware 3.3.1

I tried printing with the same exact gcode on the MK3 from an SD card and that worked fine. It didn't happen with previous versions of PrintTimeGenius, but I'm not sure which version caused it to stop working because there were several that came rather quickly.

eyal0, 2018-07-24:
Very strange.  I don't get it!  I see some weird errors in your log about shlex but I don't think that it's related to your issues.

It sounds like maybe too much CPU is being used.  Perhaps the RPi isn't able to write to the log fast enough to keep up?  Could you try these:

* Disable the debug logging of PrintTimeGenius.
* While printing, during the problems, run "top" on the OctoPrint computer and see if the CPU usage is too high.  Compare with and without PrintTimeGenius.

Let me know if you get any results from those.  Thanks!

foosel, 2018-07-24:
I think I've spotted the issue in 26f787f291962e5cce0117671db301c0915272a7

You really should *not* save stuff to disk inside the send loop, which you are effectively doing through these kinds of config updates triggered by sent commands. Serializing yaml takes a load of time, serializing json (twice) does take some time as well, and while this time is spent no new commands can be sent to the printer since you are blocking the send loop thread here.

Better yet would be not saving this data during the print job but after and keeping it in memory until then, or saving automatically every n seconds or so. But definitely not inside the send or the receive loop, that's way too timing critical.

Rapsey, 2018-07-24:
Well spotted @foosel !

I'm also having this issue, M204/M205 instructions cause a 800-900ms delay before the next instruction is sent. I would also advocate for grouped/delayed config updates. In addition to the massive performance impact these constant i/O writes cause increased wear on a microSD card.

eyal0, 2018-07-24:
That makes sense but the saving only happens when the config changes and that's only when there are Mcodes, which are probably rare during printing. I thought about a delayed write or even doing it in a different thread but I didn't think that the delay would be that bad nor that frequent.

What would be a better solution? I only really need to write on shutdown. Maybe I'll just put it into a shutdown hook and that's it. Even if we lose the config because the machine crashes before we write, it's no big deal because we read it again when we load.

foosel, 2018-07-24:
Possibly stupid question (I haven't really done a deep dive into the code) but why do you even need to save this stuff?

nebbywan, 2018-07-24:
Hello @eyal0 and @foosel, just woke up to the flurry of replies 😀 and wanted to check in to see if the additional info requested above is still needed and/or if they was anything else that might be helpful in your efforts.

eyal0, 2018-07-24:
@foosel The print time analyzer in this plugin is affected by the acceleration, jerk, and other settings of the printer.  A user filled an issue that analysis isn't accurate when analyzing before the printer is connected because the settings aren't yet loaded.  So I save them into the config to have them at startup before connecting to the printer.

Rapsey, 2018-07-24:
@eyal0 I've been using Cura (latest versions) and it has an **acceleration control** setting which is enabled by default. I'm pretty sure this is the reason why all my G-codes are full of M204/M205, often several times per second. (not if it has to wait 800-900ms each time of course)

eyal0, 2018-07-25:
This should solve the problem:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/Dev

Let me know please!

Rapsey, 2018-07-25:
I can confirm: It Works on My Machine™

I did have to remove the plugin first since the dev version is still 1.0.9 so pip wouldn't install it while the original 1.0.9 was still installed.

eyal0, 2018-07-25:
Thanks for testing.  Version 1.1.0 has this fix and some others.

eyal0, 2018-07-26:
Thanks for your thorough testing.  Sorry that you're still having problems.

This development version turns off gcode processing entirely while printing.  Let me know if it solves your problem:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/dev.zip

nebbywan, 2018-07-26:
Sorry, was unable to test this exact issue as a new problem popped up after installing the dev build provided. I'm connecting and communicating with the printer fine, but I now get a communication error when attempting to send a print. The printer never acts like it received anything and eventually OctoPrint gives me a "too many consecutive timeouts" error and disconnects. This problem is alleviated when PrintTimeGenius is disabled. Logs attached.
[octoprint.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2233594/octoprint.log)
[plugin_PrintTimeGenius_engine(1).log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2233595/plugin_PrintTimeGenius_engine.1.log)

eyal0, 2018-07-27:
@nebbywan Sorry, I messed up that dev release.  I saw the error and fixed it locally but forgot to push it to github.  It's now in 1.1.1.  I'm very curious to hear if it solves your problem!  If it doesn't, we'll look for another solution.

eyal0, 2018-07-27:
@nebbywan Sweet.  So you can confirm that the pauses that used to exist no longer exist?  Great!

nebbywan, 2018-07-27:
Confirmed. I ran that same GCODE that was submitted here and there was no stuttering to be seen. Circles were nice and circular :)