eyal0/OctoPrint-PrintTimeGenius #58: Add support for Anet A8 firmware

vincent-ogury, 2018-07-25:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.9

PrintTimeGenius Version (if you know): 1.1.0

What did you try:
Analyze a gcode

What happenned:
Anet doesn't expose parameters on connect, and even with M503 the gcode generated is not standard

What did you expect to happen:
Automatically call M503 on connect, and convert anet a8 firmware into a marlin compatible code with the M2* command.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

Here the output

```
Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: ['/dev/ttyUSB0']
Connecting to: /dev/ttyUSB0
Changing monitoring state from "Detecting serial port" to "Opening serial port"
Connected to: Serial<id=0x72d1d090, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection...
Changing monitoring state from "Opening serial port" to "Detecting baudrate"
Trying baudrate: 115200
Recv: start
Send: N0 M110 N0*125
Changing monitoring state from "Detecting baudrate" to "Operational"
Send: N0 M110 N0*125
Recv: Info:External Reset
Recv: Info: 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000
Recv: Free RAM:12251
Recv: SD init fail
Recv: ok
Send: N1 M115*39
Recv: ok
Recv: ok 1
Recv: FIRMWARE_NAME:ANET_A8_20160701V2.0 http://www.anet3d.com/ VERSION:ANET V2.0 MACHINE_TYPE:ANET_A8-L EXTRUDER_COUNT:1
Recv: Printed filament:1474.67m Printing time:9 days 20 hours 57 min
Send: N2 M105*37
Recv: ok 2

### I send the M503 command manually ###

Send: N6 M503*35
Recv: ok 6
Recv: EPR:2 75 115200 Baudrate
Recv: EPR:3 129 1474.673 Filament printed [m]
Recv: EPR:2 125 853035 Printer active [s]
Recv: EPR:2 79 0 Max. inactive time [ms,0=off]
Recv: EPR:2 83 360000 Stop stepper after inactivity [ms,0=off]
Recv: EPR:3 3 100.0000 X-axis steps per mm
Recv: EPR:3 7 100.0000 Y-axis steps per mm
Recv: EPR:3 11 400.0000 Z-axis steps per mm
Recv: EPR:3 15 400.000 X-axis max. feedrate [mm/s]
Recv: EPR:3 19 400.000 Y-axis max. feedrate [mm/s]
Recv: EPR:3 23 8.000 Z-axis max. feedrate [mm/s]
Recv: EPR:3 27 100.000 X-axis homing feedrate [mm/s]
Recv: EPR:3 31 100.000 Y-axis homing feedrate [mm/s]
Recv: EPR:3 35 4.000 Z-axis homing feedrate [mm/s]
Recv: EPR:3 39 20.000 Max. jerk [mm/s]
Recv: EPR:3 47 0.300 Max. Z-jerk [mm/s]
Recv: EPR:3 133 0.000 X home pos [mm]
Recv: EPR:3 137 0.000 Y home pos [mm]
Recv: EPR:3 141 0.000 Z home pos [mm]
Recv: EPR:3 145 250.000 X max length [mm]
Recv: EPR:3 149 224.000 Y max length [mm]
Recv: EPR:3 153 240.300 Z max length [mm]
Recv: EPR:3 51 1000.000 X-axis acceleration [mm/s^2]
Recv: EPR:3 55 1000.000 Y-axis acceleration [mm/s^2]
Recv: EPR:3 59 100.000 Z-axis acceleration [mm/s^2]
Recv: EPR:3 63 1000.000 X-axis travel acceleration [mm/s^2]
Recv: EPR:3 67 1000.000 Y-axis travel acceleration [mm/s^2]
Recv: EPR:3 71 100.000 Z-axis travel acceleration [mm/s^2]
Recv: EPR:0 880 0 Autolevel active (1/0)
Recv: EPR:0 106 0 Bed Heat Manager [0-3]
Recv: EPR:0 107 255 Bed PID drive max
Recv: EPR:0 124 80 Bed PID drive min
Recv: EPR:3 108 196.000 Bed PID P-gain
Recv: EPR:3 112 33.000 Bed PID I-gain
Recv: EPR:3 116 290.000 Bed PID D-gain
Recv: EPR:0 120 255 Bed PID max value [0-255]
Recv: EPR:3 200 95.000 Extr.1 steps per mm
Recv: EPR:3 204 50.000 Extr.1 max. feedrate [mm/s]
Recv: EPR:3 208 20.000 Extr.1 start feedrate [mm/s]
Recv: EPR:3 212 5000.000 Extr.1 acceleration [mm/s^2]
Recv: EPR:0 216 3 Extr.1 heat manager [0-3]
Recv: EPR:0 217 230 Extr.1 PID drive max
Recv: EPR:0 245 40 Extr.1 PID drive min
Recv: EPR:3 218 7.0000 Extr.1 PID P-gain/dead-time
Recv: EPR:3 222 2.0000 Extr.1 PID I-gain
Recv: EPR:3 226 40.0000 Extr.1 PID D-gain
Recv: EPR:0 230 255 Extr.1 PID max value [0-255]
Recv: EPR:2 231 -3000 Extr.1 X-offset [steps]
Recv: EPR:2 235 -400 Extr.1 Y-offset [steps]
Recv: EPR:1 239 1 Extr.1 temp. stabilize time [s]
Recv: EPR:1 250 150 Extr.1 temp. for retraction when heating [C]
Recv: EPR:1 252 0 Extr.1 distance to retract when heating [mm]
Recv: EPR:0 254 255 Extr.1 extruder cooler speed [0-255]
```

eyal0, 2018-07-26:
I have never seen firmware like this. Anet A8 does not use a normal Marlin firmware?

eyal0, 2018-07-26:
Output appears similar to this:

https://github.com/jeremy-prater/Repetier-Firmware-ANet-3D/blob/master/src/ArduinoAVR/Repetier/Communication.cpp

It is repetier firmware

vincent-ogury, 2018-07-26:
It support the reprap command. I don't know if Marlin calc would work properly on this machine. 

The problem is that I can't change the firmware. It is lock by the constructor.

vincent-ogury, 2018-07-26:
I can add on connect the proper M20* command and check if it fix the computation. I will give it a try.

eyal0, 2018-07-26:
Because it is a different firmware, the simulation might not be accurate! Let me know.

If you type in all the M commands 200-206, PTG will notice.

vincent-ogury, 2018-07-27:
I have set the config manually in the config.
The calc takes care of it and don't remove it.

So I will try to see if it is accurate enough.
Also it is hard to be sure I have put the right parameter with the right gcode based on the output of the repetier firmware ...

I close it, not a marlin firmware.

closed 2018-07-27 by vincent-ogury

eyal0, 2018-07-28:
Was it accurate?