eyal0/OctoPrint-PrintTimeGenius #60: No Gold Star when sending to OP using Cura's OP's API.

Thisismydigitalself, 2018-07-27:
OctoPrint Version:
1.3.9 release
PrintTimeGenius Version (if you know):
1.1.1, latest. same with 1.1.0.

What did you try:
Send gcode from Cura using Cura's OP's API. 
What happenned:
Don't see a Gold Star. 
What did you expect to happen:
To see the Gold start much like when I drag the Gcode to OctoPrint which in this way it works OK.

eyal0, 2018-07-27:
I'm not familiar with cura printing mode but it sounds like you are right. If you read the PTG logs, you can see if analysis was cancelled by printing.

There is also a feature to let you analyze while printing in the settings. You can try it out and see if it doesn't cause you troubles.

Thisismydigitalself, 2018-07-29:
Eyal, Can you please explain this screen and what should be checked / unchecked?
Reset to default checks this line:
"/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "{gcode}" "{mcodes}"'

Marlin-Calc?
for what it worth I am running Smoothieware. 

Thank U.

[image]

eyal0, 2018-07-29:
Everything is great but you can try to allow analysis while printing so that it won't abort. Might cause problems while you print, maybe not.

Let me know if you have good accuracy with Smoothieware. If not, there is a software called gcodestat stat that is supposed to be accurate for smoothieware but I have not tested it.

Thisismydigitalself, 2018-07-31:
Thank you.
I must say that i still see my Smoothie struggles at times. print can be smooth for 2~3 sec, then stuck for a fraction of a sec, then continues. uninstalling the plug and print head moves completely smooth. what goes on while printing and can we lower these jitters to minimum if not to get rid of those completely?

eyal0, 2018-07-31:
When the print is getting stuck, are you allowing analysis while printing or not?  If you allow analysis, it will make the printing go slower, and this will affect your speed.  But if you don't allow analysis, then the analysis won't complete before you start printing, and then you won't have the gold star.

You can't have it both way: The analysis takes time and it runs on the Raspberry Pi, which is kind of a wimpy computer.  Either you wait for it to finish and then print, or you cancel it when you start printing, or you let it analyze while printing but perhaps suffer the slowdowns while the analysis is running while printing.  I don't have a way to make the analysis run any faster.

Do you have a better solution?

Thisismydigitalself, 2018-08-01:
My settings are default and as you can see in my screenshot above. I just installed the plug and went printing. I will try to make a short video with my default settings, e.g - will install the plug , change nothing and start printing. And I don't really understand why i see these 2 Commands to chose from. for those who are not programmers these means confusion.

Thisismydigitalself, 2018-08-02:
Hello Eyal, Two files attached.

WITH_PTG: https://expirebox.com/download/b7c98bac7f66f6d69e6fb9a9342f726f.html
WITHOUT_PTG: https://expirebox.com/download/90501b28a6d06ea75f6085bb562ff6fa.html

eyal0, 2018-08-06:
@Thisismydigitalself By the way, what kind of Raspberry Pi do you use for OctoPi?  If it's getting stuck every 2 seconds, maybe the calculation of the progress is really slowing things down?

Thisismydigitalself, 2018-08-07:
Dear Eyal, 
Raspberry Pi 3 Model B.
I did spend quite some time trying to debug PTG but since I need to keep my printer printing, i sadly had to remove it and move on.

Mortikhan, 2018-09-11:
I can confirm that checking the "Allow Analysis while printing" checkbox does in fact all PrintTimeGenius to analyze the file when sent from Cura. With the Mk3 printer, I have to wait about 2-3 minutes for the hotend and bed to heat up as well as mesh calibration, which seems like plenty of time for PTG to do its work and not delay the print. WAY better estimates than I was getting from OctoPrint (11 hours vs 9 days).
Thanks!

eyal0, 2018-10-12:
@Mortikhan @Thisismydigitalself I had an idea about how to let users print while the machine is warming up but then cancel if the analysis takes longer than the warm-up time.  If you're interested in trying it, let me know.  #91

Mortikhan, 2018-11-07:
@eyal0 I'm up for testing. Let me know if you still want this tested.

eyal0, 2018-11-07:
@Mortikhan  I've already released it!  So you can just see if it's working for you on the latest version.

s1mpleman, 2019-01-03:
I'm guessing this is a similar issue to what I'm having so I'll put it here instead. 

I've now decided to start uploading the print files directly to .uploads through SMB. I've noticed now though I've got to start a manual calculation from the settings to get it to detect files. 

Is this something solvable or do you think it's down to octoprint itself?

eyal0, 2019-01-03:
I answered this elsewhere.  Can you make sure that you upload to the watched directory and not the uploads directory?