eyal0/OctoPrint-PrintTimeGenius #61: Update pourcentage based on the time

vincent-ogury, 2018-07-28:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.9

PrintTimeGenius Version (if you know): 1.1.1

What did you try:
Print

What happenned:
Pourcentage is not sync with time left
What did you expect to happen:
Pourcentage is sync with time spent
If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
[image]

On the screen it should display 70% instead of 53%

eyal0, 2018-07-28:
Percentage is percentage of bytes in the file size, not percentage of time. That's how OctoPrint works. I could maybe fix it but I think that it might confuse people the other way!

vincent-ogury, 2018-07-28:
I would expect a % of completion of the impression. 60% means progress means 40% of the total time left to print.

It is not really useful to know we are only at half the file. The metric is all about knowing when everything will finish.

If it is possible, replace this % by % based on the time. Maybe optional, so people can activate or no this.

vincent-ogury, 2018-07-29:
Another excellent think to add if the completed is based on time, is the M117 and M73 support. 
M73 is for the progress bar. 
M117 for message like : X% HH:MM:SS (progress and time left)

eyal0, 2018-07-29:
There are already plugins that can do M117 and they work together with PTG. I don't think that we need another plugin for that.

Fixing the progress bar might be possible. I'll check.

According to this:

https://reprap.org/wiki/G-code#M73:_Set_build_percentage

M73 is not supported by many printers! Better to send M117 instead.

vincent-ogury, 2018-07-30:
Also something optional. I don't know if they is any kind of possible detection (M73 support).

But if so, updating the progress bar based on time left would be awesome.

The plugin about M117 I have found use for the progress the filepos. Which isn't what we want.
Also 1 plugin that support web and printer display with accurated time is awesome.

eyal0, 2018-07-31:
Do you want to try the dev branch?  I didn't use an option, I just made it apply for everyone because I think that percentage of file is not very interesting but percentage of time is interesting.

This is the plugin that I use to make M117 messages:

https://plugins.octoprint.org/plugins/detailedprogress/

It should work together with the new progress to put the good progress percentage on the LCD.

eyal0, 2018-07-31:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/dev.zip

vincent-ogury, 2018-08-01:
Yeah I use it, with M73 plugin also.

I try it tonight !

vincent-ogury, 2018-08-01:
I add a comment on the code, to clean up a little :)

vincent-ogury, 2018-08-01:
I have test the dev branch

It works for the progress inside the web interface.
The details use "completion" for the percentage and it is still based on filepos.
See: https://github.com/dattas/OctoPrint-DetailedProgress/blob/1eaf2589c15c1e4bb7f94e161ee4b0ab63a93067/octoprint_detailedprogress/__init__.py#L94

So the percentage on the printer is not right.

vincent-ogury, 2018-08-03:
I have create a fork on detailsprogress with base time completion and M73 progression.
https://github.com/celogeek/OctoPrint-DetailedProgress/tree/eta-time-based

This work fine :)
but better to find out a way to based completion on time or may be create a timeCompletion key so plugin may choose the one they want

celogeek, 2018-08-05:
Hi,

I have made a patch that fixes the completion!

Can you take a look?

https://github.com/celogeek/OctoPrint-PrintTimeGenius/commit/e78f34be0ac3ddf5fbacf0a23b7a39ae06ee8343

eyal0, 2018-08-05:
@celogeek yes I also made one: https://github.com/eyal0/OctoPrint-PrintTimeGenius/compare/progress_percentage

They are similar.

celogeek, 2018-08-05:
This one only changes the progress of the web interface. All plugins use "completion" in the progress data and this one is still based on time.

My change here: https://github.com/celogeek/OctoPrint-PrintTimeGenius/commit/e78f34be0ac3ddf5fbacf0a23b7a39ae06ee8343
override the state monitor, and instead of computing the completion from filepos, it use time left.

This patch will spread the change everywhere (in the web interface, for other plugins which display the progress ...)

eyal0, 2018-08-06:
Yes, we can see that it is a problem: https://github.com/celogeek/OctoPrint-PrintTimeGenius/commit/e78f34be0ac3ddf5fbacf0a23b7a39ae06ee8343#commitcomment-29967922

Maybe the best solution is like my original one? https://github.com/eyal0/OctoPrint-PrintTimeGenius/commit/d46c0846f8aefc97e602786b7953505ed8995e68

It only affects the web interface so it won't interfere with other plugins or with the functioning of PTG.  And if other plugins want to use a better progress meter, they ought to do it.  For example, the RemainingTime plugin should just offer a timeprogress or even allow users to write equations like "time / total_time".  They might not have done that in the past because time estimates weren't very accurate.

I'm glad to hear that you are getting accurate estimates!  I worked hard on the plugin and also marlin-calc.  If you have other ideas for improvement or you see gcode files that aren't analyzed properly, let me know.

celogeek, 2018-08-06:
I have double check the patch. The progress use for the estimator come from _comm.getProgress which is based on filepos.
So no issue here.
The only issue is that on_progress hook that is called too early and with the file progression.
This is bad for plugin that use that hook.
Everything else is fine.

I think it would be better to rewrite the _updateProgressDataCallback entirely. like copy paste with a slighly adjustment.

If you test android or ios app that is connected to octoprint, or details progress, you will see that the progress is not the same as in the web interface

celogeek, 2018-08-07:
I have made an addon to check the progress information for all plugins based on progress, api, and apps.
It still call the estimator with the filepos progression.

https://github.com/celogeek/OctoPrint-ProgressBasedOnTime