eyal0/OctoPrint-PrintTimeGenius #63: Negative filament length and volume

celogeek, 2018-08-05:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.9

PrintTimeGenius Version (if you know):1.1.1

What did you try:
Analyze a gcode

What happenned:
Filament length and volume of marlin calc go negative. This breaks the filament manager and let thinking that the analyze has not runned.

What did you expect to happen:
filament length and volume are properly analyzed

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
[Neck_01.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2260377/Neck_01.txt)

This:
```
"/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Neck_01.gcode" "M200 D175
M200 D0
M92 X8000 Y8000 Z40000 E41725
M203 X25000 Y25000 Z3000 E5500
M201 X2000 Y2000 Z120 E10000
M204 P80000 R1000000 T150000
M205 B20000 S000 T000 J002"
```

return
```
Analysis: {"estimatedPrintTime": 1287.88193850868447043, "printingArea": {"maxX": 163.51300000000000523, "maxY": 300.00000000000000000, "maxZ": 13.23999977111816406, "minX": 80.00000000000000000, "minY": 142.72700000000000387, "minZ": 0.20000000298023224}, "dimensions": {"width": 83.51300000000000523, "depth": 157.27299999999999613, "height": 13.03999976813793182}, "filament": {"tool0": {"length": -1094.96200631494184563, "volume": -26336.92341335244418588}}}
```

celogeek, 2018-08-05:
The problem occurs if I activate the Wipe During Retraction in simplify 3D.
That means it will move a retract at the same time:

```
G92 E0
G1 X160.025 Y150.002 E-1.1099 F1701
G1 X159.500 Y148.877 E-2.2050
G1 X159.705 Y148.780 E-2.4051
G1 X160.276 Y150.002 E-3.5952
G1 X159.698 Y151.240 E-4.7999
G1 X159.493 Y151.143 E-5.0000
```
I think here it believes that the retractation is cumulate, it ends up far from 5mm.

Do you take properly into account the absolute move here?

celogeek, 2018-08-05:
I have also observed a massive pause (like several seconds) during this print.

I have removed this option from simplify, it works without this feature. I can print, and no more massive pause.

I'm wondering if there is something wrong in the latest bugfixes-2 branches with progressive retractation.

eyal0, 2018-08-06:
Hmm.  Okay, first, I see that the volume of your print is enormous!  That seems strange.  It is because you have "M200 D175" in your call to marlin-calc.  That is indicating that your filament is 175mm wide, which is like, filament wider than my head!  That's crazy!  Usually we see "M200 D1.75".  That is a more normal number.

As for calculating the filament length, I am also seeing a crazy number.  I'm not sure why but I will investigate.

As for the pause, I'm not sure what causes it, yet.  It seems to be related to this:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/blob/master/octoprint_PrintTimeGenius/__init__.py#L449

and this:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/blob/master/octoprint_PrintTimeGenius/__init__.py#L443

For each line of gcode, I search for MCodes so that I can provide the inputs to marlin-calc, like you see in your command line above.  Those set things like steps per movement and acceleration and jerk rates.  By using those, marlin-calc can more accurately simulate Marlin firmware.

I read those in as they come out of the printer but I turn that off while printing so that printing is not slowed down.  I think that *that* processing is the cause of the slowdown.  But, it should be off while printing, so it shouldn't affect anything.  However, I can't figure it out.

I want to debug this issue but I'm not sure how because on my OctoPi, I see no slow down and everything works fine.  I think that you are a programmer.  Do you want to try to remove those lines for processing gcode (send and receive) and see if those are really the problem?  Maybe you have another idea why it is slow?  You are not the first to report this problem and I added the "if printing return" lines because someone else's report.  It make the problem go away for him.

Which Raspberry Pi do you use for OctoPrint?  Maybe you are using RPi Zero and it is too slow?  I use RPi 3, which is the latest and fast enough.  I would like PTG to work for all OctoPi users, including Pi Zero.  Maybe it's possible for me to test the processor speed and turn off the gcode processing?  But that is getting ahead of myself.  First I want to prove that the gcode processing, even with "if printing return", is the problem.

Thanks for your bug report and your help!

celogeek, 2018-08-06:
First of all, I have this in my config:
```
    printer_config:
    - M200 D1.75
    - M200 D0
    - M92 X80.00 Y80.00 Z400.00 E417.25
    - M203 X250.00 Y250.00 Z30.00 E55.00
    - M201 X2000 Y2000 Z120 E10000
    - M204 P800.00 R10000.00 T1500.00
    - M205 B20000 S0.00 T0.00 J0.02
```

so 2 m200 command: M200 D1.75 and M200 D0

That mean D = 0 here. And even then I have D1.75.
I don't know why marlin-calc use the D175 here.

celogeek, 2018-08-06:
the M200 is happening at the connection of Marlin (so it comes from the firmware itself).
the issue here is that marlin-calc is called with bad parameters here.

eyal0, 2018-08-06:
@celogeek I had a bug in my code causing the M200 D1.75 to be wrong.  I have fixed it.  It was really dumb because I even had unittests for it that were failing but I didn't run unit tests before committing.

Anyway, should be okay now.  Good luck!

eyal0, 2018-08-06:
By the way, what Raspberry Pi do you use for the OctoPrint?  Maybe that is related to the jitter while printing?

celogeek, 2018-08-06:
I use a Rasberry PI 3 B+ (the fastest one)

eyal0, 2018-08-09:
I think that the only issue left is the slowness.  I'll cover it here: https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/62

closed 2018-08-09 by eyal0