eyal0/OctoPrint-PrintTimeGenius #66: Smoothieware print settings are not sent to marlin-calc

eyal0, 2018-08-11:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/62#issuecomment-412255356

User's jerk/acceleration/etc settings were not copied correctly from the printer to the printer settings.  Not sure why.

eyal0, 2018-08-11:
> Hello again, After 10 hours prints i had some time to look a bit deeper (and also updated to latest 1.1.4). I Sent M503 manually. first thing to notice is that the Acceleration (M204) in PTG log is different than that of my firmware and M92 which is nowhere to be found. My firmware acceleration is set to 750 and in PTG log file reads S500. i can look for other settings but it seems there is an issue here. i can check the rest later today. Correct me if I'm wrong but it seem that PTG does not work in my case.

> BTW: what mechanism sends M503 before printing? i don't see it in OctoPrint nor in your plug settings. does this happens behind the scenes during connection?

eyal0, 2018-08-11:
First, from looking at the output of the M503, everything looks fine.  I don't see any weirdness in the output from M503 that would make me think that PTG would fail to pick-up all the necessary Mcodes.  So I'd expect it to work, yet it isn't.

You ask what mechanism sends M503.  The answer is: I don't know.  I use a marlin printer and everytime I connect to the printer, M503 output appears on the screen.  I don't know how.  I just assumed that it was OctoPrint but maybe it isn't.  Can you test this?  On OctoPrint, with the terminal open, disconnect from the printer and then reconnect.  Let me know if the printer spews all the M503 output.  Maybe it's just a feature unique to Marlin, that marlin is doing on OctoPrint's behalf?  If so, I can modify PTG to make it always do an M503 at printer connect.  No biggie.

Now, even if the printer doesn't do M503 at startup, **any** M503 issued ought to update the config.  Even if you just type it in yourself, which you did!  So I would expect that if you type an M503 and then wait 5 seconds and then analyze a file, the PTG log will show all those correct settings getting send to marlin-calc so that it can run an accurate simulation.  Could you try that?  Do an M503 from the terminal.  Wait until the output is finished spewing in the terminal.  Then go to the PTG settings and manually analyze a file.  And finally, find the log output for that manually analyzed file and see if the arguments to marlin-calc match what you see in M503 output.  (it won't have all lines, just M200-M206 and M92, I think.  But not all, most are irrelevant to speed calcuation.)  If you see a mismatch between what it shows in the log and what it shows in M503, that means PTG isn't properly reading the M503 info and saving it.  I'll have to figure out why.  If there are error messages in the log, I would want to know about those, too!

And finally, after we sort all that out and marlin-calc is running with the right M200-M205 and M92 values, I want to see what estimates marlin-calc is giving.  I expect them to be really close to the actual print time, not of by about 9% like you've been experiencing.  If after getting all the right values marlin-calc is still inaccurate then I want to try gcodestat and see if it's any better.

So, you've got tasks!  :smile: Here you go:

1. Disconnect and reconnect from your printer using OctoPrint's disconnect/connect button.  Watch the terminal while it happens to see if all the M502 output appears.  If you can, send me a log of everything that comes out of the printer after you connect.
2. Run M503, wait for it to finish, then analyze a file and check if the log shows all the right Mcommands passed to marlin-calc.  If you can, send me the output of the M503 and also the PTG log (you can just attach the whole thing or copy-paste lines).
3. Determine if marlin-calc is accurate enough for you.  Send me gcode for files that you have actually printed along with how long they actually took (not a prediction).  (You can get the print time from  ~/.octoprint/uploads/.metadata.json, too, or just send me that file and I'll find it in there.)  I'll run them through gcodestat to see if gcodestat is actually more accurate for Smoothieware than marlin-calc.

Thanks for your help!  We'll make PTG better for everyone together!

Thisismydigitalself, 2018-08-11:
Your wishes are my commands.
I see both progress and a sort of a regression.

The good news is: I now see a much more realistic time estimation using the very same g-code which I executed last night. last night it was reading ~3hr print time, but ended up to be like 4hr. today, after sending manually M503 command and reloading this very same gcode - it reads:

**AIRV2_v4.gcode
Internal: AIRV2_v4.gcode
Uploaded: a few seconds ago
Size: 7.9MB
Model size: 225.89mm × 215.45mm × 14.70mm
Filament: 26.09m / 184.45cm³
Estimated print time: 04:09:54**

These are good news! it means a change for the better which i need to learn and figure it out.
what puzzles me is this:  i wanted to load PTG before initiating M503 and compare the result with a gcode which was loaded after M503. the good and bad news is that it appears that once initiating M503 manually - it is in memory somehow. no matter what i tried: RESET the board, restart OP, clear PTG analyzer - nothing got me back to square one to see how it loads without M503. I don't know how computers think so I will leave it at that.

Now for the "bad" news:
i recently upgraded to 1.1.4. then loaded a gcode to see what i get and .. nothing. no matter what i tried i did not get PTG to generate this gold-star icon so my first impression was that 1.1.4 gone sour on me, so I had to downgrade to 1.1.3 dev and .. Nope. it did not work either. So, I left my desk to cool down and when i got back this gold-star was shining upon me - WTH?

to make a long story short - it now takes much longer to calculate an 8Mb gcode file. it takes no less than 42sec to analyze one single file and i have not yet tried to print while it is doing its thing - i need to test that but i would hate starting a 2hr print and a minute or so into the print to realize it will actually take 4hr - twice as long :)

So.. more tests ahead but i see progress. i think..

Thisismydigitalself, 2018-08-11:
Just started a new print. 
Cura Est: 1h 06min
OTG  Est: 1h 42min

Thisismydigitalself, 2018-08-11:
right after the print was done, i ran M503 manually and then reloaded the same g-code again. this time i get 1h 26min which is exactly the time it took to actually print the file. 

- Can you please add M503 to PTG?
- Can you please explain how to Analyze the files? it seems i can't generate a log file with the correct firmware setting in it, but print time looks to be calculated OK.

I am a bit lost. need guidance.

eyal0, 2018-08-11:
PTG memorizes the Mcodes and saves them in the config when OctoPrint shuts down. It loads them up when OctoPrint starts, too. You can find those in ~/.octoprint/config.yaml . That files only gets written when you shutdown OctoPrint and only gets read when OctoPrint is started. So if you want to reset it, you have to shut off OctoPrint, then delete those lines, then start up again. I store it so that you can analyze even before you've connected to OctoPrint.

If you send me the log. I can read from there everything. You can download it midprint, it won't affect printing. It's available in the settings.

Thisismydigitalself, 2018-08-12:
[STOPPER2.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2280806/STOPPER2.txt)
No change to M503
.metadata file includes personal data about all my prints / models so I will not share it here.
And i don't understand this analyzing gcode - once you press Analyze you have no clue what goes on and if the generated log file is what you just analyzed. as a user i need more feedback what goes on.

Thisismydigitalself, 2018-08-12:
PTG log file contains all my analyzed files so i don't want to attach it here.
What i can see is the i must manually send M503.  regarding the log, i can see on some of the files that it did pick up my firmware settings, like:

**_AIROSOL_V2_v3.gcode" "M221 S105
M205 S0.00000 X0.10000 Z1.00000
M92 X159.53000 Y159.53000 Z159.53000 E6000.0000
M200 D0.0000
M204 S750.00000 P57988
M203 X666.66669 Y666.66669 Z666.66669 E50.0000"'_**

These looks to be my fw settings.