eyal0/OctoPrint-PrintTimeGenius #67: Hanging after analysis of large files

NotExpectYet, 2018-08-11:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.9

PrintTimeGenius Version (if you know): 1.14

What did you try: Slicing a large file, 136mb

What happenned: It takes a lot longer than the default analyser. 

What did you expect to happen: It would run faster? 

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
Can't upload the file due to it been larger than 10mb. I've snipped the log's below and also run time on yours and the octoprint analyser. 

> > 2018-08-11 17:54:04,103 Not running built-in analysis.
> 2018-08-11 17:54:04,108 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Groot_Pot.gcode"
> 2018-08-11 17:54:04,110 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "/home/pi/.octoprint/uploads/Groot_Pot.gcode" "M92 X80.00 Y80.00 Z400.00 E400.00
> M203 X500.00 Y500.00 Z15.00 E25.00
> M201 X1000 Y1000 Z1000 E5000
> M204 P500.00 R1000.00 T500.00
> M205 B20000 E5.00 S0.00 T0.00 X10.00 Y10.00 Z0.40"
> 2018-08-11 18:04:28,113 Failed to run '"/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "/home/pi/.octoprint/uploads/Groot_Pot.gcode" "M92 X80.00 Y80.00 Z400.00 E400.00
> M203 X500.00 Y500.00 Z15.00 E25.00
> M201 X1000 Y1000 Z1000 E5000
> M204 P500.00 R1000.00 T500.00
> M205 B20000 E5.00 S0.00 T0.00 X10.00 Y10.00 Z0.40"'
> Traceback (most recent call last):
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 260, in _do_analysis
>     raise Exception(results_err)
> Exception: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Groot_Pot.gcode" "M92 X80.00 Y80.00 Z400.00 E400.00
> M203 X500.00 Y500.00 Z15.00 E25.00
> M201 X1000 Y1000 Z1000 E5000
> M204 P500.00 R1000.00 T500.00
> M205 B20000 E5.00 S0.00 T0.00 X10.00 Y10.00 Z0.40"
> ...........................................Processed 4557244 Gcodes and 622 Mcodes. 4384875 blocks
> Total time: 182068.107033
> Traceback (most recent call last):
>   File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 74, in <module>
>     main()
>   File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 51, in main
>     progress.append([filepos, time])
> 
>  MemoryError

The octoprint analyser did this: 

>  time /home/pi/oprint/bin/python /home/pi/oprint/bin/octoprint analysis gcode ~/.octoprint/uploads/Groot\ Pot.gcode
> DONE:0.000276803970337s
> RESULTS:
> dimensions:
>   depth: 0.0
>   height: 0.0
>   width: 0.0
> extrusion_length:
> - 0
> extrusion_volume:
> - 0
> printing_area:
>   maxX: null
>   maxY: null
>   maxZ: null
>   minX: null
>   minY: null
>   minZ: null
> total_time: 0
> 
> 
> real    0m4.218s
> user    0m3.045s
> sys     0m0.266s
> 
> 

Your plugin: 

> time /home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py marlin-calc ~/.octoprint/uploads/Groot_Pot.gcode

> ........................................................................
> .Processed 4557244 Gcodes and 617 Mcodes. 4381760 blocks
> Total time: 147039.868409

Then everything hangs, can't get on web and also can't reconnect to ssh. It doesn't freeze but it hangs everything. 

Any help would be appreciated as it's taking ages to start a print because of this. 

Wonderful work though man! For small files this thing is epically awesome! Can't thank you enough :). Hope this is just me been an dodo.

eyal0, 2018-08-12:
Thanks for your detailed bug report. You did a great job!

I never did any consideration of enormous files. My goal was always accuracy down to the minute but all that accuracy for a 40 hour print is excessively pedantic.

Your OctoPrint analysis failed, by the way. It was fast but reported the size of your model as 0. Maybe you ran it incorrectly?

Looks like marlin-calc succeeded but the processing of that output failed due to running out of memory. I'll work on a solution.

NotExpectYet, 2018-08-12:
Dude your awesome! Thann you. Yeah I totally agree it's pedantic, its just I use the filament tracker plugin same as another user and it stops that working if I can't get these analysis off. I don't often print large files but would be nice to track them with that and your plugin. 

I've skipped it this time and will manually input the filament used, just hope you can work out a fix. Thanks for the reply and interest in my issue anyway. 

When marlinc-calc finished it frozed the whole pi, and the octoprint on like you say finished but I guess its too big a file for that to handle. I'll do more investigating about large file sizes see if anyone has a work around. 

Would there be a way to run analasys off the pi and on my PC and upload them both to the correct folders to be read? 

I was tempted to run the octoprint software off an old i5 I've got laying about, might mitigate these issues some but this is way out of my scope of understanding hahaha.

eyal0, 2018-08-12:
For a 40 hour print, it's only 2400 lines of output. It's not that much! I wouldn't expect my script to choke on that. I'll have to find out why it did.

NotExpectYet, 2018-08-12:
https://drive.google.com/open?id=1CkQ29fi9Vb2CihhJ7xr0HCbhaCUGyt-O

That's the file if it helps. I'm just uploading it to the copy I made of the SD card and I'm going to run the analysis on that one as the other is printing at the moment. 

I will post you the PTG logs when it's run :).

NotExpectYet, 2018-08-12:
> > 2018-08-12 11:41:50,082 Not running built-in analysis.
> 2018-08-12 11:41:50,084 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Groot_Pot.gcode"
> 2018-08-12 11:41:50,086 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "/home/pi/.octoprint/uploads/Groot_Pot.gcode" ""
> 2018-08-12 11:46:50,340 Failed to run '"/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "/home/pi/.octoprint/uploads/Groot_Pot.gcode" ""'
> Traceback (most recent call last):
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 260, in _do_analysis
>     raise Exception(results_err)
> Exception: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Groot_Pot.gcode"
> ...........................................Processed 4557244 Gcodes and 617 Mcodes. 4381760 blocks
> Total time: 147039.868409
> Traceback (most recent call last):
>   File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 74, in <module>
>     main()
>   File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 51, in main
>     progress.append([filepos, time])
> MemoryError
> 
> 2018-08-12 12:01:30,058 Not running built-in analysis.
> 2018-08-12 12:01:30,066 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Groot_Pot.gcode"
> 2018-08-12 12:01:30,069 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "/home/pi/.octoprint/uploads/Groot_Pot.gcode" ""

Less than 15 minutes, There's no times been logged on the error messages sadly. I've just come back to it though, and checked the logs, then ran another analysis on a different logging level. It was on DEBUG before, I've changed it to CRITICAL see if it makes a difference.

NotExpectYet, 2018-08-12:
2018-08-12 12:01:30,058 Not running built-in analysis.
2018-08-12 12:01:30,066 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Groot_Pot.gcode"
2018-08-12 12:01:30,069 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "/home/pi/.octoprint/uploads/Groot_Pot.gcode" ""
2018-08-12 12:07:14,872 Failed to run '"/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "/home/pi/.octoprint/uploads/Groot_Pot.gcode" ""'
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 260, in _do_analysis
    raise Exception(results_err)
Exception: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Groot_Pot.gcode"
...........................................Processed 4557244 Gcodes and 617 Mcodes. 4381760 blocks
Total time: 147039.868409
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 74, in <module>
    main()
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 51, in main
    progress.append([filepos, time])
MemoryError

There's another attempt

NotExpectYet, 2018-08-12:
I looked up large file errors with octoprint and it seems version 4.0.0 of simplify3d pads out the file with `NULL NULL NULL` lines throughout the gcode. This has been sorted with 4.0.1 but my licence ran out for me to upgrade. 

This was causing issues with octoprint going through the file. 

I tried a Cura export from the latest version 3.4.1 and uploaded it. With the same settings it's 4MB smaller and when I ran the analysis it gave me the below: 

2018-08-12 12:33:06,056 Not running built-in analysis.
2018-08-12 12:33:06,063 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Groot_Pot_-_Cura.gcode"
2018-08-12 12:33:06,066 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "/home/pi/.octoprint/uploads/Groot_Pot_-_Cura.gcode" ""
2018-08-12 12:37:18,087 Failed to run '"/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "/home/pi/.octoprint/uploads/Groot_Pot_-_Cura.gcode" ""'
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 260, in _do_analysis
    raise Exception(results_err)
Exception: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Groot_Pot_-_Cura.gcode"
...........................................Processed 3696621 Gcodes and 827570 Mcodes. 4325617 blocks
Total time: 189379.193204
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 74, in <module>
    main()
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 51, in main
    progress.append([filepos, time])
MemoryError

Same memory error! damn! Ruled out the slicer at least ahaha

eyal0, 2018-08-12:
If you can, turn on the built-in analysis and see how long that takes.  Just for reference.  There is a checkbox in the settings to do it.

I think that I'm pretty close to a solution to your issue.

NotExpectYet, 2018-08-12:
Just waiting for it to run now for you :) sweet too!

eyal0, 2018-08-12:
Bothered?  About what?  Looks good to me!

Here's the built-in analyzer:

```
└──> time octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 ~/Downloads/Groot\ Pot.gcode      
DONE:120.039178133s
RESULTS:
dimensions:
  depth: 237.59500000000003
  height: 299.8
  width: 277.496
extrusion_length:
- 177450.94739984273
extrusion_volume:
- 0.0
printing_area:
  maxX: 277.596
  maxY: 257.595
  maxZ: 300.0
  minX: 0.1
  minY: 20.0
  minZ: 0.2
total_time: 2268.595453649973

octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10     120.28s user 0.18s system 99% cpu 2:00.50 total

```

120 seconds!  Mine took 30 seconds.  4 times faster and it's way more accurate.  I can't expect much better than that!

NotExpectYet, 2018-08-12:
On a Pi

Built in:

> DONE:1162.50298214s
> RESULTS:
> dimensions:
>   depth: 237.59500000000003
>   height: 299.8
>   width: 277.496
> extrusion_length:
> - 177450.94739984273
> extrusion_volume:
> - 0.0
> printing_area:
>   maxX: 277.596
>   maxY: 257.595
>   maxZ: 300.0
>   minX: 0.1
>   minY: 20.0
>   minZ: 0.2
> total_time: 2268.595453649973
> 
> 
> real    19m25.583s
> user    19m23.409s
> sys     0m0.570s

19 Minutes! 

Yours:

> real    3m4.394s
> user    4m50.634s
> sys     0m4.516s
> 

Much faster! Thanks man!

NotExpectYet, 2018-08-13:
I closed that way to early! sorry man. 

> 2018-08-12 19:54:55,227 Running built-in analysis.
> 2018-08-12 20:14:30,459 Result: {'estimatedPrintTime': 136115.7272189984, 'printingArea': {'maxZ': 300.0, 'minX': 0.1, 'maxX': 277.596, 'maxY': 257.595, 'minY': 20.0, 'minZ': 0.2}, 'dimensions': {'width': 277.496, 'depth': 237.59500000000003, 'height': 299.8}, 'filament': {'tool0': {'volume': 0.0, 'length': 177450.94739984273}}}
> 2018-08-12 20:14:30,468 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Groot_Pot.gcode"
> 2018-08-12 20:14:30,471 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "/home/pi/.octoprint/uploads/Groot_Pot.gcode" ""

Your's isn't finishing now since I updated the branch. 

It's now 05:48 this morning and the octoprint analysis finished within 20 mins, your's still hasn't completed yet. No crashes or anything though.

eyal0, 2018-08-13:
Do small files still work?

eyal0, 2018-08-13:
I only tested on my laptop. I'll try to test on RPi.

NotExpectYet, 2018-08-13:
Small files work fine :) Just tried one.

eyal0, 2018-08-13:
At least there's that! Maybe there's some other drain on resources with marlin-calc that only happens on RPi.

NotExpectYet, 2018-08-13:
How come it ran on cli but not when called through octoprint?

eyal0, 2018-08-13:
I appear to be experiencing this:

https://stackoverflow.com/questions/11937028/poll-method-of-subprocess-popen-returning-none-for-long-process

I'll look into using sarge instead of subprocess, which maybe doesn't have the same problem.

NotExpectYet, 2018-08-13:
Strange one, god speed man! All way over my head.

eyal0, 2018-08-13:
I wrote it in a hurry. No promises!

NotExpectYet, 2018-08-13:
I'll test shortly when I get home :) Thanks for this!

NotExpectYet, 2018-08-13:
Dude! You are awesome, finished in rapid time. I say resolved! 

Can I donate? Feel you deserve it.

> 2018-08-13 15:32:43,811 Running built-in analysis.
> 2018-08-13 15:52:19,114 Result: {'estimatedPrintTime': 136115.7272189984, 'printingArea': {'maxZ': 300.0, 'minX': 0.1, 'maxX': 277.596, 'maxY': 257.595, 'minY': 20.0, 'minZ': 0.2}, 'dimensions': {'width': 277.496, 'depth': 237.59500000000003, 'height': 299.8}, 'filament': {'tool0': {'volume': 0.0, 'length': 177450.94739984273}}}
> 2018-08-13 15:52:19,120 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Groot_Pot.gcode"
> 2018-08-13 15:52:35,146 Subprocess output: 
> 2018-08-13 15:52:35,147 Result: {"estimatedPrintTime": 136140.0}
> 
> 2018-08-13 15:52:35,148 Merged result: {'estimatedPrintTime': 136140.0, 'printingArea': {'maxZ': 300.0, 'minX': 0.1, 'maxX': 277.596, 'maxY': 257.595, 'minY': 20.0, 'minZ': 0.2}, 'dimensions': {'width': 277.496, 'depth': 237.59500000000003, 'height': 299.8}, 'filament': {'tool0': {'volume': 0.0, 'length': 177450.94739984273}}}
> 2018-08-13 15:52:35,153 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" "marlin-calc" "/home/pi/.octoprint/uploads/Groot_Pot.gcode" ""
> 2018-08-13 15:55:39,852 Subprocess output: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Groot_Pot.gcode"
> ...........................................Processed 4557244 Gcodes and 617 Mcodes. 4381760 blocks
> Total time: 147039.868409
> 
> 2018-08-13 15:55:39,854 Result: {"printingArea": {"maxZ": 300.0, "maxX": 277.596, "maxY": 257.595, "minX": 0.1, "minY": 20.0, "minZ": 0.20000000298023224}, "dimensions": {"width": 277.496, "depth": 237.59500000000003, "height": 299.79999999701977}, "firstFilament": 5.063000327129e-05, "progress": [[0, 147039.86840915936], 'filament': {u'tool0': {u'volume': 1254.3021313688337, u'length': 177447.447807008}}}
>

eyal0, 2018-08-13:
I need to figure out how to do some analysis feedback @Thisismydigitalself . The problem is that it is asynchronous. I can start it from the button in the settings but then it goes into a queue and I don't know when it finishes. But even if I did know, how do I communicate that to the user. Right now, a gold star appears next to the gcode if the analysis is done. So that's a start. I can't really report that analysis is done in the settings because you can close those and it keeps going.

Maybe I can have a toast in the corner, like you get when the slicer finishes. I would only show it if analyze is selected manually or on every analysis? Those toasts can get annoying. A busy spinning star is cool, too, but where would I put it? In settings? That might be closed. But maybe there is.nowhere else.

NotExpectYet, 2018-08-13:
Understood :) thanks again much appreciated. Ignore that dude I clipped the extra analysis! It was just extra code to me, I just pasted the relevant parts. It did do it :)

NotExpectYet, 2018-08-13:
I can post the full log if you require!

eyal0, 2018-08-13:
Ok, got it. 3 minutes compared to 20. Pretty sweet! You can disable that built in one and the analyze gcode comments, marlin-calc overwrites all of those and does a better job. I'll give the new code a final glance and make a release.

Thisismydigitalself, 2018-08-13:
**_"I have a dream".._**
Can't resist the urge to speak up and say how i wish for a PTG plugin for Cura. 
it is obvious that this will make more sense to know beforehand and during different slicing profiles how long print will take before uploading it to printer using Cura's API.

Thisismydigitalself, 2018-08-13:
Just finished a print with PTG  1.1.5.

Cura estimation: 1h 42min
PTG initial estimation: 1h 56min
Actual print time:  2h 7min

[image]
[image]

eyal0, 2018-08-13:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/70 for the progress indication.

eyal0, 2018-08-13:
@Thisismydigitalself if you can, please take an image of the table in advanced stats. Total print time accuracy is hard because it.includes a variable heating and cooling time. I want to compare predicted versus actual extruding time.

I also want to know if gcodestat, which is a Smoothieware estimator, is better than mine for Smoothieware prints. Please run your gcode through that, too, or just send me the gcode and I'll do it.

I have never made a cura plugin. Cura has plugins? An easy way would be for the calculator to run after cura runs and put a progress comment in the file. And then PTG can read that out. How do cura plugins work?

Thisismydigitalself, 2018-08-13:
[image]
[image]

https://github.com/Ultimaker/Cura/tree/master/plugins

closed 2018-08-15 by eyal0

eyal0, 2018-08-15:
@Thisismydigitalself The difference between actual extruding time and predicted extruding time isn't great.  Sometimes more, sometimes less.  The inaccuracy is a problem.  Maybe gcodestat would do better.  If you want to run gcodestat yourself or send me gcode files to check myself, we could see if gcodestat is better.  If it is better then I will add gcodestat to PTG for Smoothieware printers.

PR #68 (2018-08-12): Efficient analyze
Fixes #67
commit 2aa4af1c (2018-08-15): Merge pull request #71 from eyal0/efficient_analyze

Improve large gcode file support

This fixes #67