eyal0/OctoPrint-PrintTimeGenius #69: Fails to analyse the gcode

glena, 2018-08-13:
OctoPrint Version: `OctoPrint 1.3.9 running on OctoPi 0.15.1`

PrintTimeGenius Version (if you know): `1.1.5`

What happenned: Run several prints and it never showed the `Approx. Total Print Time` nor the gold star. Also run the gcode analysis and it does not show any result.

Log:
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2283000/plugin_PrintTimeGenius_engine.log)

glena, 2018-08-13:
Ah missed that setting, is it possible to add some kind of feedback when the analysis is running?

Let me check.

glena, 2018-08-13:
That was it. Sorry for the fake issue :P
good job with the plugin BTW

closed 2018-08-13 by glena

eyal0, 2018-10-12:
This is fixed by #70 and #83 and #91