eyal0/OctoPrint-PrintTimeGenius #70: Add indication that analysis is started/queued/finished

eyal0, 2018-08-13:
Maybe in the location where there is a gold star on completed could be an hourglass for a file that is queued for analysis and with a spinning star for analysis in progress.

Maybe something that would say the progress of the analysis as it happens, in percent. But that message doesn't get sent to the web UI so probably can't do that easily. Queued/started/complete should be enough.

LazaroFilm, 2018-10-05:
That would really be awesome!

eyal0, 2018-10-09:
@LazaroFilm I did some work on this.  The gold star should now pulse on/off while a file is being analyzed.  If you want to try it, you can go to Octoprint plugin installation and put in this link:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/analysis_pending.zip

Let me know what you think of it.  It should pulse on the file that is currently being processed.  If you turn off octoprint in the middle of analysis and restart, the pulsing goes away (because the analysis didn't complete).  If you analyze multiple files, only one at a time will pulse because analysis is done one file at a time, in a queue.

Let me know if the pulsing seems like an obvious enough signal that it is working.  I also tried a spinning star instead of a pulsing one but I found it harder to see.  I also made the star a little bigger so that it's easier to see, because I think that it's hard to see it clearly due to the color.

closed 2018-10-12 by eyal0