eyal0/OctoPrint-PrintTimeGenius #72: Cannot build Marlin simulator

doug-a-brunner, 2018-08-16:
**OctoPrint Version:** 1.3.9

**PrintTimeGenius Version (if you know):** 1.1.6

**What did you try:** Attempted to build Marlin analyzer locally so it could run on my OS X OctoPrint machine

**What happened:** [Link](https://github.com/eyal0/Marlin/tree/estimate-bugfix-2.0) provided in `octoprint_PrintTimeGenius/analyzers/README.md` is dead. The closest match I could find (`https://github.com/eyal0/Marlin/tree/estimate-bugfix-2.0.x/`) includes instructions to `cd Marlin/Marlin && make -f Makefile.calc`. `Makefile.calc` does not exist in that branch, and attempting to build with the regular Makefile fails as expected since the Arduino headers are not present. This occurs both on the Mac and on a Linux machine.

**What did you expect to happen:** Successful build of the Marlin simulator.

I would have reported this on the forked Marlin repo, since I imagine the omitted `.x` at the end of the provided URL is just a typo, but issue tracking is not enabled there. Could you add `Makefile.calc` and any other required files to the repo?

eyal0, 2018-08-16:
I no longer use make, I now use scons.  Please try again.  You might need to modify the file SConstruct in that directory.

Please send a PR to update the out-of-date instructions or tell me where they are and I can do it.

After you build for OSX, please provide the output of uname -a and also the binary that you build so that I can add it to the release and everyone can enjoy PTG with OSX support.

Thanks!

eyal0, 2018-08-17:
If you're good at that, be my guest! I'm happy to see a PR. Know that most of the code is verbatim Marlin source and only some of it is mine. I prefer to change as little Marlin code as possible so that future merges will be easy to do. So, if you can find clang flags that will make the compile work, that's better than modifying the Marlin source. Does that reasoning make sense too you?

If you need answers about the source to get it to work, let me know. The most interesting files are calc.cpp and hax.h .  Good luck!

doug-a-brunner, 2018-09-09:
Just remembered to check back on this, thanks for the merge.

Thinking about making it easier to keep the binaries up to date, I took a look at something called [OSXCross](https://github.com/tpoechtrager/osxcross) (does what you think it does), but unfortunately it depends on extracting the SDK from XCode, and Apple does not like this. I'm happy to build new binaries when you need though.