eyal0/OctoPrint-PrintTimeGenius #75: I am wondering

Psychokiller1888, 2018-08-25:
Hi!

First of all, thank you for your effort on this plugin. Now, it's gonna sound bad, I know, but it's not meant to hurt.

I am a CNC programmer, working mostly on Heidenhain, but also used to Siemens, so, in other words, almost gcode.

After programing my CNC, be it manually or through a computer (I tend to prefer to program myself directly on the machine, takes more time, but the results are better), we get a VERY accurate time estimation, that usually is about 10 seconds correct.

On Octoprint, this or any other time estimator, it's BAD. I mean you start with a devil time 3 times higher than what's really gonna be, and it decreases like madness skipping 5-6 seconds every seconds and tends to be correct when the print nears the end. This totally reminds me about the old navigation systems in cars.

Now, that being said and as I said it's not bad meant, I'd like to understand why... An estimation of the time, even though our 3d printers are not based on rulers like CNCs are, is based on the analysis of the code and some constants. A travel speed * the length of a line gives a time, on every axis. I simply can't understand why the estimation can be THAT wrong. A CNC program looks alike a gcode printing file, once dumped to a readable file. We just have other functions and codes, but mainly it's the same.

eyal0, 2018-08-25:
No worries, no hurt.  I know that my code isn't perfect.  If you can provide the PTG logs, we can see what it did and why it might have been so far off.  Here are a few things that I would check with the log:

* PTG doesn't know how long your printer will spend heating up or cooling off.  It will learn over time but it might not have yet learned.
* The estimate is based on Marlin printers.  If you printer isn't using Marlin firmware, it might be off.
* Is PTG even running?  Maybe you didn't wait long enough for the analysis to complete?
* Did PTG get the right details about the speed of your printer?  Maybe those parameters weren't sent to the estimator?

There are probably a few others that I can't think of.  The best thing that you can provide is the PrintTimeGenius log, which you can download from OctoPrint's logging settings, and the actual time that it took to print something, as measured after the fact.  With those, we can start to investigate.  I think that many users get results that are within 1%.

Psychokiller1888, 2018-08-26:
Happy you didn't take it bad!

- Heating up / cooling down shouldn't play a big role, we are talking or a default of maybe max 30 seconds or a minute? between old heaters and new ones and between different temperatures needed
- I'm on Marlin
- I have no idea how long I should wait for the analysis to complete unfortunately? What I know is that while it's printing, time decreases very fast at start, the decrease slows down over time

When you say within 1%, which is totally more than acceptable, do you mean at very start? Or live recalculation over time? I mean, if at effective 10 minutes from the end it finally tells me there's 10 minutes and 37 seconds left, it's no real use. The estimation needs to be within the said 5% right at start. Those 5% can be recalculated once running then

Imma check the logs after work this pm and have my eyes a bit everywhere to check what could lead to this. It's not on one file only, it's on all.

eyal0, 2018-08-26:
Two of those look okay.  One doesn't.

I can't really do much to help without seeing the log.  Are the correct Mcodes being sent?  Without logs, I can't help much.

Psychokiller1888, 2018-08-26:
Sure sure :) Don't trip :) I'm first trying to solve by myself so you don't work for something I'm responsible of, given the fact you gave me four leads to check

eyal0, 2018-09-21:
Haven't heard back in a while. Please reopen this issue if you need more help.

closed 2018-09-21 by eyal0