eyal0/OctoPrint-PrintTimeGenius #77: Support for volumetric extrusion and firmware retract

celogeek, 2018-09-09:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: last

PrintTimeGenius Version (if you know): last

What did you try:
Activate volumetric extrusion and firmware retract

What happenned:
not sure for the volumetric extrusion but the firmware retract seems not take in account

What did you expect to happen:
more accurate time computation with those options activated

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

Here the command called by PTG:
```
"/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/3DBenchy.gcode" "M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z400.00 E400.00
M203 X250.00 Y250.00 Z30.00 E55.00
M201 X2000 Y2000 Z120 E10000
M204 P800.00 R10000.00 T1500.00
M205 B50000 S0.00 T0.00 J0.02"
```

Here my firmware M503
```
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E400.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X250.00 Y250.00 Z30.00 E55.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X2000 Y2000 Z120 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P800.00 R10000.00 T1500.00
Recv: echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B50000 S0.00 T0.00 J0.02
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:  G29 W I0 J0 Z-0.27650
Recv: echo:  G29 W I1 J0 Z-0.12050
Recv: echo:  G29 W I2 J0 Z-0.06600
Recv: echo:  G29 W I3 J0 Z-0.09300
Recv: echo:  G29 W I4 J0 Z-0.20200
Recv: echo:  G29 W I0 J1 Z-0.20350
Recv: echo:  G29 W I1 J1 Z-0.00750
Recv: echo:  G29 W I2 J1 Z0.02650
Recv: echo:  G29 W I3 J1 Z0.12350
Recv: echo:  G29 W I4 J1 Z0.05300
Recv: echo:  G29 W I0 J2 Z-0.23850
Recv: echo:  G29 W I1 J2 Z-0.07300
Recv: echo:  G29 W I2 J2 Z0.01800
Recv: echo:  G29 W I3 J2 Z0.02500
Recv: echo:  G29 W I4 J2 Z0.03100
Recv: echo:  G29 W I0 J3 Z-0.24750
Recv: echo:  G29 W I1 J3 Z-0.07300
Recv: echo:  G29 W I2 J3 Z-0.03200
Recv: echo:  G29 W I3 J3 Z0.07150
Recv: echo:  G29 W I4 J3 Z0.13700
Recv: echo:  G29 W I0 J4 Z-0.37450
Recv: echo:  G29 W I1 J4 Z-0.23550
Recv: echo:  G29 W I2 J4 Z-0.18000
Recv: echo:  G29 W I3 J4 Z-0.18850
Recv: echo:  G29 W I4 J4 Z-0.23150
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H205 B65 F0
Recv: echo:  M145 S1 H240 B110 F0
Recv: echo:PID settings:
Recv: echo:  M301 P26.07 I2.05 D82.75
Recv: echo:  M304 P136.27 I18.48 D251.16
Recv: echo:Retract: S<length> F<units/m> Z<lift>
Recv: echo:  M207 S5.00 W13.00 F3000.00 Z0.10
Recv: echo:Recover: S<length> F<units/m>
Recv: echo:  M208 S0.00 W0.00 F1500.00
Recv: echo:Auto-Retract: S=0 to disable, 1 to interpret E-only moves as retract/recover
Recv: echo:  M209 S0
Recv: echo:Z-Probe Offset (mm): (mm):
Recv: echo:  M851 Z-2.30
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L0.00 U0.00
Recv: ok
```

The firmware retract value is 
M207 S5.00 W13.00 F3000.00 Z0.10
M208 S0.00 W0.00 F1500.00

That means if a G10 is called, the M207 is used (retractation)
And if G11 is called, the M208 is used (detraction)

I join my benchy gcode done by Slic3r

[3DBenchy.gcode.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2364667/3DBenchy.gcode.zip)

The estimated time was 1h08 (with correction x1.04).
And real time was 1h25.

Also, I have readjusted on the fly the M207 AND M208 option, but this can't be taken into account on the fly right?

About the volumetric extrusion, I do prime without it, and before printing I run the M200 command.
is it take into account ?
So by default it's off, and then on ? so the meaning "EXX" change in the file.

eyal0, 2018-09-12:
Thanks for your detailed report.  It's very good!

Volumetric extrusion is turned on in Marlin firmware by the M200 command.  If the latest M200 command is "M200 Dx" and "x" is not 0, then volumetric extrusion is turned on!  But if x is 0, it is turned off.

In your configuration, you have volumetric extrusion *off*, because the latest M200 is "M200 D0".  But if you have a different M200 in your gcode file, it will be observed and used.  So volumetric extrusion should work in that case.

One case where you might have difficulty is if you type "M200 D1.75" into your terminal *after* PTG has analyzed your file.  This can happen if you work like this:

1. Upload a file with no M200 command in it.  (PTG starts analyzing right after upload).
2. Type "M200 D1.75" on the terminal.
3. Print the file.

There is no way for PTG to know at the time of analysis that you are going to later type "M200 D1.75" before printing, so that analysis won't be right!  I think that the built-in OctoPrint analyzer would fail in the same way.

If you put "M200 D1.75" into your gcode, then it will work.

I hope that explanation helps.  Now I'll look at the G10 and G11 issue and get back to you.

eyal0, 2018-09-12:
I have tried adding support for G10 and G11 in marlin-calc.  It works but it only adds a few minutes to the simulation of your benchy.  Your 3dbenchy has 508 G10 in it.  Each one is 5mm and with rate 3000mm/min.  That is only 50.8 seconds.

There is also a zhop of 0.1mm at max z speed, which is 30mm/s.  That is less than 2 seconds (but probably a little more because of acceleration maximum.

It is not adding too much.  In the total simulation, which includes jerk and other things, it added just 167 seconds, which is less than 3 minutes.  But you are seeing a difference of 17 minutes, which is much more!  I don't know why.

eyal0, 2018-09-12:
Ah, I see that you have M200 D1.75 in your gcode.  That is okay, marlin-calc can handle that.  So you should be getting accurate results.

marlin-calc computed that the benchy will have a volume of 9.44.  Official benchy is supposed to be 15.55, but some of benchy is hollow so 9.44 sounds about right.  I think that volumetric seems to be working.

celogeek, 2018-09-12:
Great for volumetric. I have added to my start gcode after prime print the M200 command.
I disable it at the end of the print.
For the 17 min off I have an explanation.
My previous print the retractation was set in the slicer, with a speed of 30ms/s (retraction and detraction).
The default settings in Marlin firmware was 45 retraction and 8 detraction. During the print (after a long start), I readjust on the fly. So I change for 50 retraction and 25 detraction.
The time get closer now but ignore completly the retractation time.

Do you have a release or a branch with the patch ?

Thanks for this by the way ;)

celogeek, 2018-09-12:
PS: I use a nozzle of 0.6 with 0.25 of layer height. I reduce the infill to 10%, which still stay strong thanks to the 0.68 width of the filament that get out of the nozzle.
That's why the benshi use less filament.

celogeek, 2018-09-20:
Any news on this ?

eyal0, 2018-09-21:
Not yet. I'll try tomorrow.