eyal0/OctoPrint-PrintTimeGenius #79: [Feature suggestion] Use Slic3r PE's new M73 gcode

jollino, 2018-09-17:
OctoPrint Version: 1.3.9
PrintTimeGenius Version (if you know): 1.1.6

Not a bug report, rather a feature suggestion. Slic3r Prusa Edition 1.41 and newer generates M73 commands indicating the remaining time:

> Slic3r PE 1.41.0-alpha3 combined with our new firmware has this feature already implemented. Starting with the alpha2 release of Slic3r PE, a new M73 code is added in G-codes and the printer will inform the user about the time remaining. This value is refreshed every minute and works both for the Normal and Stealth (silent) mode. Please note you need firmware 3.3.0 or higher.
The format of the new M73 code is the following:
**M73 Pxxx Qxxx Rxxx Sxxx**
where P is the percentage of the total time already printed in normal mode, R is the time remaining for normal mode in minutes, Q is the percentage of the total time printed in the stealth mode, and S is the time remaining to the end of print in the stealth mode.

I was wondering if there's any plan to eventually use that data, if available in the gcode, for the time estimate. It seems to be quite accurate, and would probably simplify the analysis as the first M73 command encountered with a R parameter is effectively the expected duration of the print.

Thanks!

eyal0, 2018-09-21:
I could use that. I already have the same thing for cura file outputs. However, I have it turned off by default because I have found the marlin-calc output to be more accurate than the comments in the files.

If you want, I can add support for the latest Slic3r PE M73 output. Users could then use that instead of marlin-calc. It would run much faster than marlin-calc but probably be less accurate. In the past, Slic3r PE predictions have not been as good as marlin-calc.

jollino, 2018-09-21:
I think Slic3r's new predictions are probably only accurate with Prusa's own original printers (I have a MK3 and it seems to really nail it!) so I understand the limited appeal for owners of other printers. It would definitely be nice to have it as an option within PrintTimeGenius, if it's not too much hassle. Someone came up with another plugin that seems to alter OctoPrint's own estimation: https://github.com/sysadminsh/OctoPrint-M73ETAOverride but I suppose many would like to have the feature within PTG, especially if they change slicers often.

Speaking of which, slightly OT: is there any way to have PTG analyze all past files at once?

Also, is the plugin currently translatable? I see there's a babel.cfg file and a translation folder, but nothing is in the latter. (I never used babel, however, so I may be missing something.) I'm a translator and would be happy to give back to the project by providing an Italian localization for the plugin.

DesC21, 2018-09-22:
It would be awesome if you could add this, I also have a mk3 and the Slicer estimates are spot an. It would be a great feature for prusa owners.

jollino, 2018-09-22:
Here is one I generated a few days ago from a Thingiverse stl: [Toothpaste_Squeezer.gcode.gz](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2408251/Toothpaste_Squeezer.gcode.gz)

I would say the normal mode is going to be used by most users most of the time so that's the one I would go for.

Thanks!

eyal0, 2018-10-01:
In the sample file that you attached, both normal and stealth have the same values so it doesn't seem to matter.  I will use normal.

eyal0, 2018-10-01:
How do you suggest I use it?  Users without a Prusa printer aren't affected but there may be users that use Slic3r PE but don't have a Prusa printer.  Those users might prefer marlin-calc.

I think that I will put the option to get progress from the file, default false, and users with Prusa printer can try it by turning on the option.  I'll try to explain it well in the options.

jollino, 2018-10-01:
I think the normal and stealth values change for more complex prints; stealth mode essentially slows down the print so that the motors' noise is greatly reduced (kind of pointless if you have the fan blasting air at 100%, but that's another story... :) )

A switch to turn the feature on or off, defaulting to false, sounds like a very sensible approach to me. In theory, a fully configured Slic3r PE with accurate values for jerk, acceleration and so on should be able to guess the remaining time accurately enough even for non-Prusa printer, but that probably requires more tests than most people are likely going to care about. That's probably why this new Gcode can be disabled within Slic3r PE (and indeed I had to manually enable it in my custom profiles after upgrading to 1.40!)

eyal0, 2018-10-03:
@jollino I don't know how to do or use translations but if you think that it would be useful and you know how, I'm happy to try it!

Have you seen evidence that Slic3r PE estimates are better than the marlin-calc ones?  I ask because I'm trying to find the correct default for users.  If it's better, then I want to turn it on for all Slic3r PE users.  If not, then I want to leave it off by default.

There's another issue that Slic3r PE also writes the amount of filament used, volume, etc.  Other slicers like Cura write those, too, and they are usually wrong.  Simplify3d is especially bad with multi-filament.  So I would need to ignore those while not ignoring the M73.

Also, I think that I only want to use the M73 estimates for Prusa printers.  Some people might use Slic3r PE with non-Prusa printers and maybe for them, marlin-calc is more accurate.  How can I detect if a printer is a Prusa printer?  Is there anything in the M503 output that indicates it?  Or M115?

eyal0, 2018-10-03:
Opened this for analyzing all files: https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/84

jollino, 2018-10-03:
I admit I didn't use marlin-calc much because until Prusa came up with this M73 thing, I didn't even know about PTG :) But I gave it a try and the Slic3r PE estimates are literally spot-on to the minute, whereas PTG, which I briefly used, is close but not as spot-on. It's certainly way better than default estimate and Slic3r's PE estimate are not as dramatically better, however given the opportunity to use them on a compatible printer, it makes sense to do so. Right now I'm using this https://github.com/sysadminsh/OctoPrint-M73ETAOverride which completely overrides the default calculation, and has the nice side effect of providing the same accurate ETA even in other plugins (such as https://github.com/fabianonline/OctoPrint-Telegram).

M115 does say it's an original Prusa:
Send: M115
Recv: FIRMWARE_NAME:Prusa-Firmware 3.3.1 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok
(I'm a little surprised to see no "field separator" there; seems a little nasty to parse!)

M503 returns a bunch of stuff but nothing that explicitly identifies it as a Prusa printer:
Send: M503
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z400.00 E280.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X200.00 Y200.00 Z12.00 E120.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X1000 Y1000 Z200 E5000
Recv: echo:Acceleration: S=acceleration, T=retract acceleration
Recv: echo:  M204 S1250.00 T1250.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E2.50
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P22.86 I2.61 D49.91
Recv: echo:PID heatbed settings:
Recv: echo:   M304 P126.13 I4.30 D924.76
Recv: echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
Recv: echo:   M207 S3.00 F2700.00 Z0.00
Recv: echo:Recover: S=Extra length (mm) F:Speed (mm/m)
Recv: echo:   M208 S0.00 F480.00
Recv: echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
Recv: echo:   M209 S0
Recv: echo:Filament settings: Disabled
Recv: ok

Thank you so much for your work on this, and please do keep me updated about translations. I'm working on localizing Octopod for iOS as well these days. :)

eyal0, 2018-10-11:
Okay, I did some work.  If you can, please try to install this plugin (you can put the URL into OctoPrint's plugin install dialog):

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/slic3r_pe_progress.zip

After that, I suggest that you reboot OctoPrint.  Then go into the options for PTG and enable Slic3r PE M73.  You might also want to delete all the rows in the compensation table in the advanced section of the options.

Please then analyze a file and look at the PTG logs.  You should see something like:

   Running: "/home/eyal/private/OctoPrint/venv/bin/python" "/home/eyal/private/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/eyal/.octoprint/uploads/Toothpaste_Squeezer.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining

You can see there where it is doing an analysis with the slic3r_pe print_times extracted.  That will override marlin-calc.  (But you still need marlin-calc because that is what calculates the filament length and object size, and those are important to OctoPrint.)

Let me know after you try it how it works.  Thanks!

jollino, 2018-10-11:
Okay, here's the result. It's a slightly different file from the one I had uploaded here as I thought I'd seize the opportunity to debug a filament I've been having some issues with! The object is the same, I just changed some parameters in the slicer.

```
2018-10-11 11:52:01,373 Not running built-in analysis.
2018-10-11 11:52:01,380 Disabled: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Toothpaste_Squeezer_GREY.gcode"
2018-10-11 11:52:01,385 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Toothpaste_Squeezer_GREY.gcode" ""
2018-10-11 11:52:04,458 Sarge output: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Toothpaste_Squeezer_GREY.gcode"
Processed 15763 Gcodes and 507 Mcodes. 15843 blocks
Total time: 934.528213

2018-10-11 11:52:04,460 Result: {"printingArea": {"maxZ": 5.0, "maxX": 164.538, "maxY": 127.05, "minX": 60.0, "minY": -3.0, "minZ": 0.20000000298023224}, "dimensions": {"width": 104.53800000000001, "depth": 130.05, "height": 4.799999997019768}, "firstFilament": 0.00267506546990817, "progress": [[0, 934.5282132365015], [0.00182798675089803, 912.0226599593622], [0.00267506546990817, 893.3452581994618], [0.04463663820766051, 833.2923067990695], [0.0632799739641155, 773.2707420719734], [0.11372906255607905, 713.2684755708444], [0.1801144848940163, 653.2625031265328], [0.2667218205653072, 592.8291277542083], [0.35124871873640795, 532.8179732989519], [0.4409371701791792, 471.70338296491826], [0.5300979083086712, 411.1974794683524], [0.6114722136889138, 351.1680941634273], [0.6998984113852247, 290.69838977593565], [0.7826018851683664, 230.67774223097604], [0.8758915617576656, 170.2402505681473], [0.9081641964005998, 110.14456460871509], [0.9575411297018952, 50.13088555630577], [0.9862992240941428, 5.094982007075714], [0.9867037536080687, 0.0], [1, 0]], "estimatedPrintTime": 934.5282132365015, "lastFilament": 0.9862992240941428, "filament": {"tool0": {"volume": 8.570387839647937, "length": 1212.4618230554108}}}

2018-10-11 11:52:04,463 Merged result: {u'printingArea': {u'maxZ': 5.0, u'maxX': 164.538, u'maxY': 127.05, u'minX': 60.0, u'minY': -3.0, u'minZ': 0.20000000298023224}, u'dimensions': {u'width': 104.53800000000001, u'depth': 130.05, u'height': 4.799999997019768}, u'progress': [[0, 934.5282132365015], [0.00182798675089803, 912.0226599593622], [0.00267506546990817, 893.3452581994618], [0.04463663820766051, 833.2923067990695], [0.0632799739641155, 773.2707420719734], [0.11372906255607905, 713.2684755708444], [0.1801144848940163, 653.2625031265328], [0.2667218205653072, 592.8291277542083], [0.35124871873640795, 532.8179732989519], [0.4409371701791792, 471.70338296491826], [0.5300979083086712, 411.1974794683524], [0.6114722136889138, 351.1680941634273], [0.6998984113852247, 290.69838977593565], [0.7826018851683664, 230.67774223097604], [0.8758915617576656, 170.2402505681473], [0.9081641964005998, 110.14456460871509], [0.9575411297018952, 50.13088555630577], [0.9862992240941428, 5.094982007075714], [0.9867037536080687, 0.0], [1, 0]], u'estimatedPrintTime': 934.5282132365015, u'lastFilament': 0.9862992240941428, u'filament': {u'tool0': {u'volume': 8.570387839647937, u'length': 1212.4618230554108}}, u'firstFilament': 0.00267506546990817}
2018-10-11 11:52:05,435 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Toothpaste_Squeezer_GREY.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
2018-10-11 11:52:06,585 Sarge output: 
2018-10-11 11:52:06,588 Result: {"progress": [[0, 997.0], [0.0026172755393473386, 960.0], [0.042093881262983725, 900.0], [0.061991866827667386, 840.0], [0.105638472277562, 780.0], [0.14720007786432748, 720.0], [0.23219233705520764, 660.0], [0.31081401679557874, 600.0], [0.39557511626421554, 540.0], [0.4731824306444794, 480.0], [0.5505798763295486, 420.0], [0.6329989628228252, 360.0], [0.7108648111029664, 300.0], [0.7950358449648242, 240.0], [0.8761957192869331, 180.0], [0.9058145794870079, 120.0], [0.950872475872704, 60.0], [1, 0]], "estimatedPrintTime": 997.0}

2018-10-11 11:52:06,625 Merged result: {u'printingArea': {u'maxZ': 5.0, u'maxX': 164.538, u'maxY': 127.05, u'minX': 60.0, u'minY': -3.0, u'minZ': 0.20000000298023224}, u'dimensions': {u'width': 104.53800000000001, u'depth': 130.05, u'height': 4.799999997019768}, u'progress': [[0, 997.0], [0.0026172755393473386, 960.0], [0.042093881262983725, 900.0], [0.061991866827667386, 840.0], [0.105638472277562, 780.0], [0.14720007786432748, 720.0], [0.23219233705520764, 660.0], [0.31081401679557874, 600.0], [0.39557511626421554, 540.0], [0.4731824306444794, 480.0], [0.5505798763295486, 420.0], [0.6329989628228252, 360.0], [0.7108648111029664, 300.0], [0.7950358449648242, 240.0], [0.8761957192869331, 180.0], [0.9058145794870079, 120.0], [0.950872475872704, 60.0], [1, 0]], u'estimatedPrintTime': 997.0, u'lastFilament': 0.9862992240941428, u'filament': {u'tool0': {u'volume': 8.570387839647937, u'length': 1212.4618230554108}}, u'firstFilament': 0.00267506546990817}
```

The first M73 occurrences in the file are:
```M73 Q2 S16
M73 P2 R16
```

jollino, 2018-10-11:
Ok, managed to get the filament to work :D The print time estimation shown is slightly off: OctoPrint reports one minute less than what the printer shows on the display from the M73. It may just be a minor issue however, as I have a relatively slow start-up gcode to avoid oozing during the bed leveling procedure, and that does take around one minute, and that's obviously not taken into account in the M73 thing, though the first M73 appears *after* that so it shouldn't matter.

eyal0, 2018-10-11:
@jollino The outputs looks okay to me.  Can you provide more lines from the log?  Just after the lines that you've included, there should be information about compensation that PTG did.

See, PTG will try to use your previous heat-up time statistics to adjust the total print time so that your print time is more accurate.  However, it might be using old marlin-calc adjustments together with the new M73 so that could lead to some errors.  For example, in the file that you sent me previously, The first M73 says 21 minutes but the total time is 21 minutes and 11 seconds.  That means that slic3r PE is expecting your printer to heat up in just 11 seconds.  That's crazy!  PTG is looking at your previous heat-up times to compensate.  The result is shown in the logs.

Would you please slice a few files with slic3r PE for me and send them?  I don't need them to be large but they should all be different objects.  If you can send me a few outputs, I can try to understand better how slic3r PE is working and improve the gcode analysis even further.

Thanks!

jollino, 2018-10-11:
That was actually all it showed, but I had disabled PTG after installing the other plugin as it did a good job of using M73 directly (minus the start-up procedure). I'm going to keep it going for a while as I do have a few things to print in the coming days, so it should learn from my habits. :) I'll report in a few prints!

Slic3r PE's calculation is for the actual print only, I think, which is why they add the first M73 when the print actually starts. The new version (currently in beta) should add one at the beginning, but it's likely not taking any preparation into account; it would just show the time sooner.

Thank you very much for all your work, and please do let me know if I can help with translation!

eyal0, 2018-10-11:
Maybe because it was your first use of PTG and there was no compensation data available?

Also, the M73 is only in minutes.  It doesn't have any accounting of seconds, so maybe it's a rounding error?

Again, I would appreciate if you are able to send me a variety (3-5) of files sliced with Slic3r PE, so that I can do more research.

jollino, 2018-10-11:
It did have a couple of lines of compensation data, but I removed them as you had suggested. I didn't take a note of them, unfortunately.

Attached are a few files I sliced recently, ranging from very short to a few hours long (my bad, I had misread your earlier post!)
[first_layer_test.gcode.gz](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2469030/first_layer_test.gcode.gz)
[painting_palette_ii.gcode.gz](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2469031/painting_palette_ii.gcode.gz)
[Paper_Towel_Hanger.gcode.gz](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2469032/Paper_Towel_Hanger.gcode.gz)
[USB_Cable_Hanger.gcode.gz](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2469033/USB_Cable_Hanger.gcode.gz)
[Woman_Body_-_Polygon_reduction_1.gcode.gz](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2469034/Woman_Body_-_Polygon_reduction_1.gcode.gz)

eyal0, 2018-10-12:
Thanks for those.

If you see a difference between my results and that of M73, it might be due to some weird rounding.  I noticed that the file shows "estimated printing time" and also the M73 lines.  But the estimated printing time is up to 60 seconds greater than the M73.  Anyway, it's only a 60 second difference so not a very big deal.  Prusa should fix that.  I opened an issue about it: https://github.com/prusa3d/Slic3r/issues/1317

I see that the very first M73 lines up with the start of printing in all examples.  That's great, I can use that for making my code more accurate.  Thanks!

eyal0, 2018-10-12:
Thanks again.  I am done.  Please test by installing this in the OctoPrint settings:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip

You can put the URL into "From URL...":

https://plugins.octoprint.org/assets/img/help/install_plugin_from_repo.png

After you are done, go to OctoPrint advanced options and reset the analyzers please.  I added a new one for M73.  After you reset, you will need to enable Slic3r PE M73.  You can see the option in the settings.  Let me know if you don't find it.

This version has three new features:
* The M73 Slic3r support #90 
* Analyze while heating/cooling #83 
* Make the gold star flashing while analysis is in progress #70 

Whatever you test, if it works or not, I will be happy to hear about.  Thank you.

eyal0, 2018-10-12:
I will close the issue after a release is made and testing is done.

eyal0, 2018-10-12:
@DesC21 @chuckdaniel @calania @glena @Thisismydigitalself  Also you might want to test.  I would like to hear what works and what doesn't, please.  Details: https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/79#issuecomment-429357634

DesC21, 2018-10-17:
@eyal0 first of all, thanks for all your work!

I was finally able to test the M73 feature. I noticed the same as @jollino, the estimate in Octoprint is slightly off, about 3-4min. more, than what is shown on the printer display. But the estimate will get closer to what is shown on the display as the print is progressing. I only did three prints yet (reset everything as you explained before). The first run was a lot worse but I think the estimates are getting better and better. I will take a close look at this with my next prints and if I notice anything else I will report back.

eyal0, 2018-10-18:
@DesC21 Could you provide the log and also the gcode that you print?  I'm copying the time to print directly from the file so I don't see why it should be different than what the M73 files show.  If you could also provide screenshots of OctoPrint where it shows the wrong number, maybe that would help, too.

calania, 2018-10-18:
@eyal0 I have used the new version now for a couple of days and everything is working fine for me! As I don't use Slic3r I haven't tested the new m73 command but analyzing while heating works fine for me and I also don't have any problems with the new animation.

eyal0, 2018-11-13:
I think that this is working now.

closed 2018-11-13 by eyal0