eyal0/OctoPrint-PrintTimeGenius #80: No MIPS support?

ClusterM, 2018-09-20:
```
Can't find: /usr/share/octoprint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.mips
```
Is it possible to add MIPS processor support?

ClusterM, 2018-09-21:
I compiled it.
[marlin-calc.mips.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2403524/marlin-calc.mips.zip)

It works on my Onion Omega2 without any problems.

eyal0, 2018-09-21:
Good work. You used the SConstruct file?

Are you able to add the options needed for compiling into the SConstruct file so that it will work with scons and build a binary? If you send me a good PR, I will add it.

Which OctoPi machine are you using that is MIPS based?

If you submit a PR with the new file and the correct filename for it, I will add it to

ClusterM, 2018-09-21:
Yes, i'm using SConstruct but i added this lines to it:
```
env = Environment(
    CXX='mipsel-linux-gnu-g++',
    CPPFLAGS= '-Wall -static -std=gnu++14 -MMD ' + debug_flags,
    LINKFLAGS='-Wall -static -std=gnu++14 ' + debug_flags)
env.VariantDir('mips', 'src', duplicate=0)
add_program(env, 'mips')
```

There is statically linked version of binary: 
[marlin-calc.mips.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2404286/marlin-calc.mips.zip)

I'm using Onion Omega2: https://en.wikipedia.org/wiki/Omega2_(computer)

eyal0, 2018-09-24:
Nevermind, I got it done with your example.  Thanks!

commit 8861020b (2018-09-24): Add marlin-calc.mips

This fixes #80
closed 2018-09-24 by eyal0