eyal0/OctoPrint-PrintTimeGenius #82: Not Getting Gold Star Until After Printing

rebeltaz, 2018-09-29:
OctoPrint Version: 1.3.9
PrintTimeGenius Version (if you know): 1.2

Maybe I am misunderstanding what PrintTimeGenius does and does not do. I was under the impression that the file was analyzed ahead of printing and would then show the corrected time estimate, along with the gold star. What I get, however, is that everything works as it did before except that once the print is complete, **then** I get the gold star and the updated "estimate". That kind of defeats the whole purpose for this plugin. I checked the log (which is quite large - 11.8k). I did a search for the word 'error', but didn't find anything.

Am I expecting more than I should or am I doing something wrong or is there something else going on?

eyal0, 2018-09-30:
There was probably a message in the log indicating that file analysis was cancelled because you started printing.

You need to wait after uploading and before you start printing so that the analysis can complete. Alternatively, allow PTG to analyze while printing. That is available in the options.

rebeltaz, 2018-09-30:
Ohhh... ok. I am passing the files directly to OctoPrint from Cura and the print begins automatically. I will disable the "Begin Printing Automatically" option. I didn't know that. I did see the "analyze while printing" option, but I didn't want to risk affecting the print quality. Thank you for that information.

closed 2018-09-30 by rebeltaz

eyal0, 2018-10-01:
You should consider trying the Analyze while printing option.  If you're actually printing while you analyze, then the analysis might cause the printer to work poorly during the analysis.  However, the start of your print might include a few minutes of heating time and analysis won't affect the printer just waiting around and heating!

eyal0, 2018-10-12:
@rebeltaz I will soon add a feature that will analyze while printing but only while you're waiting for the printer to heat up.  If analysis takes less time than heating, it will complete successfully.  If it takes longer then it will cancel as before.

If you want to test it out, let me know and I'll send you a link.

rebeltaz, 2018-10-12:
I've been using the "analyze while printing" feature because my printer takes forever to heat and then to run the auto-leveling sequence, but I'd be more than happy to help test that for you.

eyal0, 2018-10-12:
@rebeltaz Okay, thanks, please read here: https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/79#issuecomment-429357634