eyal0/OctoPrint-PrintTimeGenius #88: Interrupts printing on large models during analysis

chuckdaniel, 2018-10-08:
First off - thank you for the tool you've created.   Appreciate the work that goes into it.

I'm running OctoPrint 1.3.9 running on OctoPi 0.15.1 and printing the "Deadpool Knife Holder" from Thingiverse.    Sliced with Simplify3D at .1mm layer height and uploaded.    I selected the gcode file to print from the files section of Octoprint and the analysis started.   

I started the print and a very interesting thing happened.     The printer would start and stop leaving little blobs all over the plate at each stop.   This was during the skirt printing.   At soon as the analysis was completed, printing continued on and worked smoothly.     

This has never been an issue for me with smaller prints.    I believe the analysis was too much for the CPU of the Pi 3+ to handle along with printing.   In the future, I will wait until analysis is complete before printing.    

I'd recommend hiding the print button, giving a warning not to print until complete or reducing the analysis speed to keep folks from experiencing this.

Thanks for considering.

eyal0, 2018-10-08:
Analysis should automatically cancel itself if printing begins, unless you specified otherwise in the options.  Can you check your options and logs for that?

eyal0, 2018-10-18:
No answer for 10 days.  Re-open if needed.

closed 2018-10-18 by eyal0