eyal0/OctoPrint-PrintTimeGenius #92: [Feature Request] Include a ignore commands list (M600)

luke321, 2018-10-16:
Hello,

I love PrintTimeGenius, it works perfectly for normal prints.
The only problem I have are prints with the M600 Filament change Gcode.
If I do not change the Filament immediately, PrintTimeGenius also learns the time the printer is waiting.
Therefore all my future predictions with the M600 command are way off.

Would it be possible to add an Ignore list in the settings?

Regards
Lukas

eyal0, 2018-10-17:
Interesting.  So that I understand: M600 causes your printer to pause and wait for a filament change.  And then later on, the print finishes with possibly a very long time because it waited a long time.  And then, on future prints, the compensation is multiplying a big factor on to the printing because of this, right?

What I would need to do to solve this is make is so that the total print time that is recorded at the end of the printing ignores all that waiting time.  Maybe I could make it so that whenever an M600 occurs, I start to count up the "waiting period" until the printer continues?  And then, at the very end of all the printing, I take the total printing time and subtract all the waiting periods.

Does that sound like a good plan?

luke321, 2018-10-17:
Hello,

you are spot on with your analysis and also the solution is probably the only easy way to fix this.
Maybe you can simply take a time stamp when the M600 command is send to the printer and one when the printer starts to request new commands?

On a side note, does PrintTimeGenius estimate the time for certain commands? So would every other GCode with M600 also use the wrong estimate?

Here is one of the files. It normally prints in about 2.5 hourse, once I let the printer sit 4 hours waiting for the filament change and on the next print start PrintTimeGenius suggested over 6 hours print time.

[HT_FLEX_Curtis_Leistungsabdeckung.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2488432/HT_FLEX_Curtis_Leistungsabdeckung.zip)

eyal0, 2018-10-17:
PrintTimeGenius does a Marlin simulation.  That means that the actual Marlin code converts Gcode lines into "blocks" where each block is the profile of the movement that will be done.  Part of that is knowing the old direction of travel and the new direction of travel and then figuring out how to make a smooth acceleration profile between then so that the printer's moves aren't too jerky.  That's the jerk settings.

Right now, the simulation ignores gcode that it doesn't recognize, so the M600 is ignored and the time value of it is 0.  But M600 actually includes moving the head out of the way, unloading filament, then reloading it, etc. I *could* put that into the simulation so that it would add that in but I'm not sure if that's the right thing to do.

What I need is to watch the gcode commands as they're being executed and look for the M600 and start the timer there, stop it when the printer keeps going, and then subtract that from the total time.  So if I'm going to subtract the *entire* M600 time, that means that in my simulation, I might need to also remove the M600 entirely.  That is, leave it as time=0.

Hmm.  Ideally I would subtract off not the entire time that the M600 is running, just the part of the M600 that isn't the motion.  Just the waiting part.  I'll think about it a little and see what can be done.

luke321, 2018-10-18:
This sounds very promising, I can run tests for you if you like.

I think it will be very difficult to only subtract the waiting part, the color changing movements are probably different for every printer.

eyal0, 2018-10-18:
The color changing movements are different for each printer but that is actually not a problem. The code in Marlin that converts gcode into stepper motor pulses and timings is already a party of the simulation. I would just need to add M600 support.

What I'm missing is a way to know when the printer has settled down and asked the user to change filament and then when the user is done. Presumably there is something in the terminal to tell me. If you could put here a copy paste of the OctoPrint terminal from a little before an M600 until a little after, so that I can see what it looks like, that would help me in development.

Thanks!

eyal0, 2018-10-18:
I think that something like this line might appear in the terminal when I need to pause the timer:

`Insert filament and press button (or M108)`

I don't know what line I should be looking for that indicates when to resume the timer.  If you send me a log, it'll help a lot.

luke321, 2018-10-18:
I will upload a log tomorrow, I have a print running till than.

luke321, 2018-10-21:
Hello,

I included a log file, whenever the printer is waiting for the user it sends "paused for user", "processing" is send when the printer is moving the head, unloading and loading the filament during M600.

[M600 Terminal.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2498861/M600.Terminal.txt)

eyal0, 2018-11-21:
Sorry that I haven't worked on this in a while.  :sob: 

I see a lot of messages about "paused for user" and "processing".  When are those displayed?  I guess that the first processing is while the print head moves out of position and removes the filament.  Then it's paused waiting for you to put it back in, then processing... what's the next pause?

It would help if I knew what each of those processing and pause sections were doing!

luke321, 2018-11-22:
Hello,

no problem ;)

all those "paused for user" and "processing" messages are send periodically by the printer.

1. processing -> printhead moves to color change point
1. paused for user -> printer waits for button press to unload filament
2. processing -> printer unloads
2. paused for user -> printer waits for button press to load filament OR extrude more filament

so basically you can have infinite "processing" and "paused for user" steps after the first two, often the filament has not the correct color after one extrusion so I choose to extrude more.

last processing -> printer resumes to printing position and continues the print