eyal0/OctoPrint-PrintTimeGenius #93: Stars are missaligned in File Manager

DominikPalo, 2018-10-17:
OctoPrint Version: 1.3.9
PrintTimeGenius Version: 1.2
FileManager Version: 0.1.3

I don't really know if this issue is more related to the [FileManager](https://github.com/Salandora/OctoPrint-FileManager) plugin or to the PritTimeGenus plugin, but when I have installed both of them, files in [FileMananager](https://github.com/Salandora/OctoPrint-FileManager) are displayed as you can see on the screenshot (the star icons are misaligned):

[image]

It is possible to somehow fix it? Or should it be reported to the FileManager project?

eyal0, 2018-10-17:
I think that this is due to an unfortunate interaction between File Manager and my plugin.  I set it up so that the gold star appears right after the gcode filename.  Maybe the best solution is to make it so that it only appears outside the File Manager and not in the File Manager window?

eyal0, 2018-10-17:
I'll need to install File Manager to test it.

eyal0, 2018-10-18:
Okay, I trust you!  I'll patch it in and release it soon.

closed 2018-10-18 by eyal0