eyal0/OctoPrint-PrintTimeGenius #100: Bug with a multiple process to calculate complete gcode

Khaossa, 2018-10-21:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.9

PrintTimeGenius Version (if you know): 1.3

What did you try: 

What happenned: When I put to print a gcode with a lot of process slice by Simplify3D don't calculate all the process. And estimating a ridiculus time.

What did you expect to happen: Calculate all the processestime

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

I attach a screen shoot, the gcode, factory file and the log to analyce it.

[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2499565/plugin_PrintTimeGenius_engine.log)

[TPU Azul.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2499567/TPU.Azul.zip)

[image]

eyal0, 2018-10-22:
Yes, I see, it's very crazy.  In the log, I see that before compensation the prediction is more than 9 hours but after it is only 36.5 minutes.  That's crazy!

I will try to figure it out.  Thanks for the bug report.

eyal0, 2018-10-22:
In the file that you sent me, there are 340 lines that start with the letter "M".  But in the log that you sent me, it says:

`390 Mcodes`

So maybe the file that you sent is the not the same as the file that was analyzed?  The number of gcode is correct, however.  Do you have an idea why that might be?

Khaossa, 2018-10-22:
Yes, the file I send you is the file that is analyzed. I slice twice from simplify3D and this is the reason that I have send you a factory file...

And you can se how when it goes at the layer 61 of 124... The process percentage is upper than 2010% of the process complete.

I suppose the bug is that only analyze the first of all the process. I have separate each piece type in a different process for easy cancelation if is needed. Without call El complete printing gcode.

eyal0, 2018-10-22:
I will show you the problem that I see:

I run this:

```
./octoprint_PrintTimeGenius/analyzers/analyze_progress.py  marlin-calc ../Marlin/Marlin/Angel-TPU_Azul.gcode 
```

and I see this output:

```
Running: "/home/eyalsoha/private/OctoPrint-PrintTimeGenius/octoprint_PrintTimeGenius/analyzers/marlin-calc.x86_64" "../Marlin/Marlin/Angel-TPU_Azul.gcode"
..........Processed 1034122 Gcodes and 340 Mcodes. 1002070 blocks
Total time: 48799.654592
{"printingArea": {"maxZ": 19.832000732421875, "maxX": 251.369, "maxY": 230.068, "minX": 69.096, "minY": 87.712, "minZ": 0.15199999511241913}, "dimensions": {"width": 182.273, "depth": 142.356, "height": 19.680000737309456}, "firstFilament": 0.00021416189544228, "progress": [[0, 48799.6545924097], [0.00020719643573615, 48794.33056106982], [0.00021416189544228, 48791.25481550457], [0.00217630763683992, 48731.190067566415], [0.00258480932348614, 48668.90517861417], [0.00263207741820835, 48606.40493551044], [0.00269975673565152, 48546.23194827264], [0.00281702596771894, 48486.174234743914], [0.0029965338347725, 48425.41859716037], [0.00307679256452078, 48365.312777054554], [0.0031381648487972, 48302.69478763657], [0.00320889372073857, 48242.60060774239], [0.00349426964451967, 48182.400858700916], [0.00411246285694602, 48122.40006051117], [0.00518341095027098, 48062.388050173744], [0.00662051350755115, 48002.38653255288], [0.0081055772401213, 47942.3760336767], [0.0094706687264097, 47882.36417343705], [0.01081136377671245, 47822.19428561108], [0.01219804472268703, 47762.178100970596], [0.01359578030371761, 47702.01046109148], [0.01494645571419631, 47641.99693848252], [0.01576345908748874, 47581.753497846876], [0.01675071770225494, 47521.741866724624], [0.01774501108478406, 47461.43753558835], [0.01879132016392464, 47401.43481487591], [0.01978682643744736, 47341.20910747962], [0.02070924736509828, 47281.20406922692], [0.02102570795234406, 47221.11289038196], [0.02134099030262462, 47160.740982620846], [0.02165052008419266, 47100.45925232893], [0.02235992269933812, 47040.176854492966], [0.02441057482924054, 46979.93744654026], [0.02664558639017095, 46919.9326889247], [0.02857938513634662, 46859.90111574829], [0.03089008109527799, 46799.786541327354], [0.0331598159926539, 46739.62338693326], [0.03447504033209094, 46679.610616852224], [0.03566194080440455, 46619.60610275177], [0.03684398971274375, 46559.31806220142], [0.03805161329403382, 46499.2773419927], [0.03922281549148735, 46439.15796553582], [0.04039796824817718, 46379.04299300692], [0.04160926515647646, 46319.028893131464], [0.04280080926859416, 46259.02634729596], [0.04395090716275895, 46199.01883351201], [0.04515596634594825, 46138.99271275076], [0.04561790454436988, 46078.673847825994], [0.04591766188993229, 46018.642227236705], [0.04654295917817833, 45958.475037624194], [0.04859603709006797, 45898.46650071953], [0.05070899716301331, 45838.4481231497], [0.05254940330268138, 45778.20679511483], [0.05491294665490555, 45718.192905634525], [0.05717557747646177, 45658.08782196327], [0.0579597288308417, 45598.06647915], [0.05849132162632464, 45537.93779704429], [0.05878161842213694, 45477.92380851263], [0.05927377493331352, 45417.771379384976], [0.0612530051107068, 45357.59538408763], [0.06308339623617053, 45297.58713699052], [0.06542209265003998, 45237.48710984704], [0.06767727285549263, 45177.48002387114], [0.06953961499513077, 45117.470586443735], [0.07119819144784323, 45057.18292315208], [0.07240096346515887, 44997.16043590409], [0.07354565533089504, 44936.99861407311], [0.07476492266572371, 44876.94940421047], [0.07597864535601019, 44816.88257302995], [0.07710989145873152, 44756.704004359155], [0.07847078980758489, 44696.57061937954], [0.0796174916069676, 44636.34888905689], [0.08086905649625456, 44576.27928020058], [0.08203793687380603, 44516.06136864986], [0.0824671963234571, 44456.014791239366], [0.08276757744153052, 44395.907596633195], [0.08355408526984087, 44335.8991847469], [0.08584208284017764, 44275.84685204466], [0.08894607881698355, 44215.58460662801], [0.09222175085042718, 44155.57592921582], [0.09549388817297516, 44095.5353885544], [0.0980501772310974, 44035.29799093197], [0.09966356483079096, 43975.28490315428], [0.10055836649781606, 43914.84217441869], [0.10079910803303249, 43854.82812484403], [0.10185082314060172, 43794.727335264666], [0.10452614878623918, 43734.495172827345], [0.10762362980570803, 43674.4938044787], [0.11078941391513104, 43614.35098469171], [0.11377468519067721, 43554.13084890271], [0.11632820192652833, 43493.83139270817], [0.1175560981144258, 43433.8141548103], [0.11879262315539206, 43373.6363915011], [0.11997217697368727, 43313.63393695163], [0.12119140965448756, 43253.39731159304], [0.1223023484965731, 43193.20847624997], [0.12347708540492228, 43133.18794777646], [0.12465570356445098, 43072.95335898808], [0.12584977742064107, 43012.942064579714], [0.126949696281699, 42952.88404760139], [0.12734201453708768, 42892.82611903283], [0.12760659804383553, 42832.713813460825], [0.12823210325625192, 42772.44465639781], [0.13099212334727142, 42712.44455272666], [0.1342064924044954, 42652.428388418724], [0.1371904814809912, 42592.10541887952], [0.14035654282264132, 42532.08648884968], [0.14252886524621652, 42471.95501892303], [0.1437576624388521, 42411.9509800921], [0.1444082225137935, 42350.800675930295], [0.14473345057022163, 42290.767766880206], [0.14542982327069295, 42230.38308300061], [0.14844018406279255, 42170.3733591601], [0.15149521389748521, 42110.33624042385], [0.154739939 [... cut ...]
That's wrong.  It should be a number that is much larger.  Much closer to 1, not close to 0.

I see that it is only a problem with the TPU file.  In your log you have other files that are okay.

I notice that the file that you sent me has a little unicode character in it.  On line 6 of the gcode, I see `TPU-A30` but the `-` is strange.  It's small.  Not normal.  It has caused the entire file to be unicode.  Maybe this is confusing for the gocde parsing?  Would you try please to remove that comment line and then analyze the file again?

Thanks.