eyal0/OctoPrint-PrintTimeGenius #106: 20 min print estimated as 3 hours

mark-hahn, 2018-11-11:
OctoPrint Version: Version 1.3.9
PrintTimeGenius Version (if you know): Latest
What did you try: Normal print
What happenned:  20 min print estimated as 3 hours

I've enclosed the gcode file (as .txt).
[lens test.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2569765/lens.test.txt)

Settings ...
[image]

eyal0, 2018-11-12:
I ran marlin-calc on my computer just now and it also said about 20 minutes, so I'm not sure why yours didn't say that.

Could you provide the actual command line that you ran?  Please send me the PTG log file where it shows the command that was run and also any compensation that was done after the run.  Thanks!

eyal0, 2018-11-13:
Not sure if this is happening to you but OctoPrint used to have a bug that if you upload a file that overwrites a previously existing file, it doesn't recalculate the print time. Not sure if that is your issue.