eyal0/OctoPrint-PrintTimeGenius #107: Time is completely wrong (almost 2 hours more than it took)

enigma2k, 2018-11-19:
Hello!

I have a problem with the time estimations by PrintTimeGenius. I'm using the new Cura 3.6 as slicer. 
In Cura it shows me a print time of 2h 48min. In Octoprint without Genius it shows me 2h 30min. and with Genius activated it shows me 4h 57min. 
The actual print time was 3h 10min. So Genius was almost off by 2 hours and Cura was only off by 22minutes.
Even in the middle of printing Genius showed me that it would take 1h more than it actually took. 

I have attached a log and gcode (renamed it to txt) of the file.

I have all the standard settings activated and didn't make any changes.
activated:
- Display precise durations instead of fuzzy ones
- Allow analysis while heating (only takes effect after saving)
- Marlin firmware simulation (replaces Octoprint built-in, faster and more accurate)

OctoPrint Version:
1.3.9
PrintTimeGenius Version (if you know):
1.3
What did you try:
reinstalling PrintTimeGenius but it did not make any difference

[plugin_PrintTimeGenius_engine (2).log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2593388/plugin_PrintTimeGenius_engine.2.log)

[CE3_supportps4progcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2593390/CE3_supportps4progcode.txt)

eyal0, 2018-11-21:
According to the log, the first estimate was 9647.216013464738 seconds.  That's 2 hours, 40 minutes, and 47 seconds.  That seems pretty close to the right answer.

However, it then compensated because it saw that in previous prints it was off by like 60%!  The scaling factor was 1.657, which is really high, so it scaled it up to over 4 hours.  Crazy!

The reason that the compensation was so high is because PTG thinks that previous prints were way off.  It shows that one print that was predicted to take 592 seconds actually took 1259.  Another that was predicted to take 18012 seconds actually took 29953seconds.  That's really far off!

Do you know why those might have happened?  You can see a table of those previous results in the PTG options under "Advanced".  Did something happen that would cause a print to take way longer than predicted?  Like, maybe you run out of filament in the middle and the printer pauses?  Or you have some sort of delay in your printing code?

eyal0, 2018-11-21:
Also, if any of those rows in the Print History are not typical, you can delete them.  The print history always stores up 5 successful prints.  If it's empty then it doesn't compensate at all.  In your case, that would give a prediction of 2 hours and 40 minutes, similar to Cura but not perfect.

I notice in the log that there is no gcode given to the marlin-calc.  That means that PTG tried to calculate the print time without knowing basics about the printer, such as max acceleration and jerk.  If you run M503 in the terminal, PTG will notice it and grab all that information.  This will make the estimate more accurate.

Is your printer using Marlin firmware?

enigma2k, 2018-11-22:
I have now restarted my Chrome browser and the Advanced options showed up. I deleted all 5 prints under "Compensation values" and now PTG is showing me the same time as Cura for the first print. I'll let PTG learn the compensation for the future prints and see if it is now working correctly.

Just a side note about: there is a usability bug for the Advanced options. Right now the little arrow next to "Advanced" is showing down when no advanced options show up and right when the options show up. It should be, however, exactly opposite.

eyal0, 2018-11-27:
I'm not familiar with the arrow pointing the wrong way.  I'll take a look.  It might just be a fluke that is solved by reloading.

tcurdt, 2018-12-02:
I just got here because a print estimated at around 3h finished in around 5h.
Looking for some advice on how to improve that.

eyal0, 2018-12-03:
Please send the gcode file that you have analyzed and also your log file.  You might need to rename the files to upload them.

tcurdt, 2018-12-03:
@eyal0 IIUC the log gets truncated, right? So I fear I will have to keep track of this on upcoming prints. 

Would be nice if there was a feedback button to send the gcode and the log. It could appear if the estimation is too far off.

eyal0, 2018-12-03:
The log files are one per day and only the last 3 days are saved.

When a new bug is opened, there is a note about attaching logs and gcode, but by reopening a bug that message doesn't show.

eyal0, 2019-11-13:
I did a big update to marlin-calc and computations should be more accurate now.  Re-open this if you're still seeing bad estimates.

closed 2019-11-13 by eyal0