eyal0/OctoPrint-PrintTimeGenius #108: Unable to generate analysis

tslilc, 2018-11-22:
OctoPrint Version: 1.39

PrintTimeGenius Version (if you know): 1.3

What did you try: Analyse a file generated by Cura

What happenned: Programme crashed

What did you expect to happen: Time estimate generated

Log and file attached.

eyal0, 2018-11-24:
I see in your logs that you had M203 E0.0.  That means that the maximum feedrate for extruding was 0!  That's why it was calculating the print time as infinite.  On some of the previous log entries, the max extruder speed was 50.

if you type M503 in the terminal, it'll dump out all the settings.  You can check if the M203 in the settings really says E0.

tslilc, 2018-11-25:
Of course! I'm so sorry, thanks for looking into this. What a stupid mistake, thanks again!

closed 2018-11-25 by tslilc