eyal0/OctoPrint-PrintTimeGenius #109: PTG with Cura 3.6 API

Thisismydigitalself, 2018-11-24:
OctoPrint Version:
1.3.9
PrintTimeGenius Version (if you know):
1.3
What did you try:
Sending gcode to Octoprint 1.3.9 via Cura 3.6 API.
What happenned:
PTG star keeps blinking but never calculate print time. when loading gcode manually baypassing Cura API - all's good.

What did you expect to happen:
PTG should calculate ETA regardless if  gcode is sent to octoprint by API or copy/paste.
If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

eyal0, 2018-11-27:
Could you send me the log file so that I could see what's happening?

Thisismydigitalself, 2018-11-28:
Hi Eyal, log attached. i hope it's the right one. i manually selected and analyzed then exported the log. i think after printing is done OP GUI showed the ETA - not before. just let me know if you need me to conduct more tests.
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2624144/plugin_PrintTimeGenius_engine.log)

Thisismydigitalself, 2018-12-02:
Guy, please help those of us that uses Cura API and add manual trigger botton to start Marlin Calc cause it wont automatically start when gcode is sent to octoprint using API. i used to copy/paste the gcode as i like PTG but i prefer using API - much easier workflow between Cura and Octoprint.

Thank you.

eyal0, 2018-12-03:
I've never used Cura with OctoPrint so I don't really know how it works.  What do you think is happening?  With a normal upload in OctoPrint, I think that analysis works.  What is special about the Cura upload...?

Thisismydigitalself, 2018-12-03:
Hi, When manualy loading a gcode to octoprint print wont start automatically and you have time to let  PTG finish it's thingy and then you manually press print. Wheb loading  gcode to octoprint using Cura API, print starts immidiatly. i believe that's the difference and PTG wont finish its calculations before print starts and it seems like when octoprint starts a print - it also stops PTG. Does this makes sense? i don't want to add pause to octoprint, which i believe i can cause my start commands are not part of Cura's gcode but added by Octoprint but PTG needs to somehow tell octoprint that it is still calculating (when file sent by API, yes?..) and for print to wait until it finishes. I hope this makes sense. i think PTG should have a checkbox like "Allow PTG to finish calculation before  commencing print" ..

Thank you.

eyal0, 2018-12-03:
I see.  First, PTG options let PTG keep working even while it is heating.  So it only cancels after the printing actually begins, like when it starts extruding filament.  If heating takes a long enough time then that should be enough. But, if not, then you're right, it will cancel.  If you are pre-heating your printer then it will probably not be enough time.

First, check that it's working like that.  Assuming that that works and *still* PTG is cancelling the simulation, I guess that means that the simulation is running longer than the heating time.  So for that, you would need a way to pause.  I'm not totally sure how to do that!  I will have to think of a way and add an option.

So, please check that you are able to calculate while heating before the extruding starts.  And if you see that it isn't enough for finishing the simulation, then I'll look into the pausing.

Thisismydigitalself, 2018-12-03:
> If you are pre-heating your printer then it will probably not be enough time.

You got that right. i am pre-hearing that's probably why even though "Allow analysis while heating" has no effect in my case. below you can see my octoprint gcode scripts. it would have been great if PTG was a command line on top of this list which will start the analysis and wait for it to finish before continue to next commands. i wonder if octoprint supports running a windows process from within start codes.. that would be nice!

[image]

Thisismydigitalself, 2018-12-04:
Should this be asked on OctoPrint git?

fieldOfView, 2020-01-12:
Resurrecting this issue, is there any way I can check if the analysis is done through the REST API?  Then I could make the Cura-OctoPrintPlugin wait until analysis is done.

Perhaps some way I can see it in the `GET /api/files/(string: location)/(path: filename)` endpoint, or an API command in the PrintTimeGenius plugin?

eyal0, 2020-01-12:
Sure.  If you use a URL like this:

http://localhost:5000/api/files/local/Flexi-Dragon_Body_unix.gcode

And then parse the json output and look for the `gcodeAnalysis.progress` in there, it'll only be there after PTG analysis is done.

Is there a way for Cura to access that?  Would you be making a plugin for Cura or for OctoPrint?

fieldOfView, 2020-01-13:
I will extend the (Cura-)plugin that does the communication between Cura and OctoPrint to periodically check for `gcodeAnalysis.progress` if PTG is detected, and delay starting the print.

fieldOfView, 2020-01-17:
I have implemented this in the Cura OctoPrint Connection plugin. A development snapshot can be found here:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/97#issuecomment-575646168

Since it is solved from the Cura side of things, I think this issue can be closed.

closed 2020-01-18 by eyal0