eyal0/OctoPrint-PrintTimeGenius #111: Windows Installed PrintTimeGenius Not working

Atomfusion1, 2018-12-23:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

OctoPrint Version: Version 1.3.10 .. Installed in **Windows** using Python 2.7

PrintTimeGenius Version (if you know):  PrintTimeGenius Plugin (1.3)

What did you try: Install into Octoprint 

What happenned: No GCode gets a Star, will not predict time at all until Gcode is started then it gives a time completely off 

What did you expect to happen: I expect to be given a Star next to gcode and see the calculated time 

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
LOG: 

2018-12-22 18:39:35,134 Not running built-in analysis.
2018-12-22 18:39:35,138 Disabled: "d:\octoprint\venv\scripts\python.exe" "d:\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_gcode_comments.py" "C:\Users\Atomfusion\AppData\Roaming\OctoPrint\uploads\charmander_MK3Fast.gcode"
2018-12-22 18:39:35,138 Running: "d:\octoprint\venv\scripts\python.exe" "d:\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_progress.py" marlin-calc "C:\Users\Atomfusion\AppData\Roaming\OctoPrint\uploads\charmander_MK3Fast.gcode" "M92 X100.2 Y100.2 Z400 E288"
2018-12-22 18:39:35,641 Failed to run '"d:\octoprint\venv\scripts\python.exe" "d:\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_progress.py" marlin-calc "C:\Users\Atomfusion\AppData\Roaming\OctoPrint\uploads\charmander_MK3Fast.gcode" "M92 X100.2 Y100.2 Z400 E288"'
Traceback (most recent call last):
  File "d:\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\__init__.py", line 298, in _do_analysis
    process = psutil.Process(sarge_job.processes[0].pid)
  File "d:\octoprint\venv\lib\site-packages\psutil\__init__.py", line 338, in __init__
    self._init(pid)
  File "d:\octoprint\venv\lib\site-packages\psutil\__init__.py", line 378, in _init
    raise NoSuchProcess(pid, None, msg)
NoSuchProcess: psutil.NoSuchProcess no process found with pid 9264
2018-12-22 18:39:35,641 Disabled: "d:\octoprint\venv\scripts\python.exe" "d:\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_gcode_comments.py" "C:\Users\Atomfusion\AppData\Roaming\OctoPrint\uploads\charmander_MK3Fast.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
2018-12-22 20:07:51,440 Not running built-in analysis.
2018-12-22 20:07:51,440 Disabled: "d:\octoprint\venv\scripts\python.exe" "d:\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_gcode_comments.py" "C:\Users\Atomfusion\AppData\Roaming\OctoPrint\uploads\Gear_Bearing.gcode"
2018-12-22 20:07:51,440 Running: "d:\octoprint\venv\scripts\python.exe" "d:\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_progress.py" marlin-calc "C:\Users\Atomfusion\AppData\Roaming\OctoPrint\uploads\Gear_Bearing.gcode" "M92 X100.2 Y100.2 Z400 E288"
2018-12-22 20:07:51,953 Failed to run '"d:\octoprint\venv\scripts\python.exe" "d:\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_progress.py" marlin-calc "C:\Users\Atomfusion\AppData\Roaming\OctoPrint\uploads\Gear_Bearing.gcode" "M92 X100.2 Y100.2 Z400 E288"'
Traceback (most recent call last):
  File "d:\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\__init__.py", line 298, in _do_analysis
    process = psutil.Process(sarge_job.processes[0].pid)
  File "d:\octoprint\venv\lib\site-packages\psutil\__init__.py", line 338, in __init__
    self._init(pid)
  File "d:\octoprint\venv\lib\site-packages\psutil\__init__.py", line 378, in _init
    raise NoSuchProcess(pid, None, msg)
NoSuchProcess: psutil.NoSuchProcess no process found with pid 8768
2018-12-22 20:07:51,954 Disabled: "d:\octoprint\venv\scripts\python.exe" "d:\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_gcode_comments.py" "C:\Users\Atomfusion\AppData\Roaming\OctoPrint\uploads\Gear_Bearing.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining

DOCaCola, 2018-12-24:
The primary issue here is that the script is not designed to run under windows. The plugin is missing the windows binaries for marlin-calc as well.

diyquad, 2020-01-01:
any update on this? same issue here

eyal0, 2020-01-02:
I still have yet to find someone that is able to work on this and I don't have a window computer that I could use.  If someone with programming ability wants to try, I can assist.

diyquad, 2020-01-03:
> I still have yet to find someone that is able to work on this and I don't have a window computer that I could use. If someone with programming ability wants to try, I can assist.

i could try, i have octoprint on windows;  happy to help, let me know how :)

eyal0, 2020-01-04:
If you've never compiled anything on Windows, it'll be.... a learning experience. 😜 But if you have, then not too hard! You need to get the marlin-calc code which is available on GitHub as eyal0/marlin-calc. It's here:

https://github.com/eyal0/Marlin

Meanwhile, now that GitHub has added some cool workflow features, I'll see if that might help.

eyal0, 2020-01-04:
@diyquad I was able to cross compile in on Linux.  I think that it worked!  Try to run the attached file.  You might want to rename it so that it ends with ".exe".  You can run it from the command line.  I think that if you press the "windows" button and then type "cmd" and hit enter, you'll be on a command-line.  From there, you can run the program.  You need to find in what folder it is on your computer but you might run it something like this:

```
"c:\Downloads\marlin-calc.windows.exe"
```

And it should spit out `Read up all remaining commands` plus some other progress info that you can ignore.  Next, try it with a real file.  I don't know where the files are on your computer but maybe something like this:

```
"c:\Downloads\marlin-calc.windows.exe" "C:\Users\diyquad\AppData\Roaming\OctoPrint\uploads\charmander_MK3Fast.gcode"
```

Obviously, change the path above to point to the real location of marlin-calc.  Also change the path of the gcode file to point to a real gcode file.  This time you should get a real analysis.  It's a lot of lines of numbers and a summary at the end.  You can compare the total number of seconds with your expectations.

After we get past those steps we'll try the next ones.  Thanks for your help on behalf of all the windows users.

diyquad, 2020-01-04:
seams to work :)
let me know what you think
you can contact by mail (@gmail) if you need some tests

```

Progress: 0.94734336513981743, 608.14685061573982239, 872.16108214037501511
Progress: 0.94774698321362782, 608.14685061573982239, 872.30822435029870121
Progress: 0.94803676029226092, 608.26763918995857239, 877.32855587619519611
Progress: 0.94843002918469144, 608.26763918995857239, 877.44367176580044543
Progress: 0.94870945708194487, 608.36181643605232239, 877.46401156958938827
Read up all remaining commands
Progress: 0.94911307515575516, 608.36181643605232239, 877.54721699790388811
Progress: 0.94940285223438825, 608.42938235402107239, 877.56754026058843010
Progress: 0.94979612112681888, 608.42938235402107239, 877.61872117998859721
Progress: 0.95007554902407221, 608.47033694386482239, 877.63906098377754006
Progress: 0.95047916709788260, 608.47033694386482239, 877.65949101107196384
Progress: 0.95096557862273090, 608.48474124073982239, 877.86481796481996298
Progress: 0.95176246558897193, 603.48474124073982239, 878.26539978664982300
Processed 3030 Gcodes and 15 Mcodes. 2872 blocks
Total time: 878.265400
Analysis: {"estimatedPrintTime": 878.26539978664982300, "printingArea": {"maxX": 119.00000000000000000, "maxY": 129.00000000000000000, "maxZ": 9.89999961853027344, "minX": 81.00000000000000000, "minY": 91.00000000000000000, "minZ": 0.30000001192092896}, "dimensions": {"width": 38.00000000000000000, "depth": 38.00000000000000000, "height": 9.59999960660934448}, "filament": {"tool0": {"length": 601.48474124073982239, "volume": 4.25164509973859861}}}
```

eyal0, 2020-01-04:
Cool.  Next step, I need to figure out how to identify Windows computers.

What do you get if you type in something like this into the same command line:

```
"d:\octoprint\venv\scripts\python.exe" -c "import platform; print(platform.system()); print(platform.machine())"
```

Assuming that is where your python.exe is, you might need to adjust.  On my computer, the output is:

```
Linux
x86_64
```

What's yours please?

eyal0, 2020-01-04:
Assuming that the system will say "Windows" and the machine will say "x86_64", I made this:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/windows.zip

If it works out like that, `Windows` and `x86_64` then you can try to install the link above.  You should be able to copy and paste that right into OctoPrint's plugin installation dialog.

diyquad, 2020-01-04:
> Cool. Next step, I need to figure out how to identify Windows computers.
>
> What do you get if you type in something like this into the same command line:
>
> ```
> "d:\octoprint\venv\scripts\python.exe" [...]
it says

Windows
AMD64

but my CPY is intel I5-7400

eyal0, 2020-01-05:
Ah.  That's okay that is says `AMD64`, even though if you have Intel.  Naming the 64-bit version after AMD was traditional or something.

I adjusted for the `AMD64` thing.  Can you try the link above now?  See if you files get gold stars when they are analyzed, which happens automatically, and when you upload, and when you force analysis in the settings.

Put the PTG logs here so we can keep going.  We're getting close!

diyquad, 2020-01-05:
cool ;)
It have install this version, but we got a python error 

[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4023020/plugin_PrintTimeGenius_engine.log)

Also, the plugin is asking for update when loading the web interface

eyal0, 2020-01-05:
Okay, it seems to have failed to run this command:

```
"d:\imprimante3d\octoprint\venv\scripts\python.exe" "d:\imprimante3d\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_progress.py" marlin-calc "C:\Users\Fixe.DESKTOP-H15S5OG\AppData\Roaming\OctoPrint\uploads\CFFFP_20mm-hollow-box.gcode" ""
```

This is something that should have worked.  Can you try to run it from a command line?  (Press the Windows button, type "cmd", and then put in the command above.)

I have fixed it so that it won't ask for the web interface update now.  Same install link as before.

We will look into fixing the "Cost Estimation" plugin later.  (I have never used it.)

eyal0, 2020-01-05:
I just uploaded a version that should fix the bug that you were seeing.  Please install again.

diyquad, 2020-01-05:
> Okay, it seems to have failed to run this command:
>
> ```
> "d:\imprimante3d\octoprint\venv\scripts\python.exe" [...]

I did it before update to new version, here is the result
Read up all remaining commands
Processed 6764 Gcodes and 98 Mcodes. 6684 blocks
Total time: 1631.736973
{"printingArea": {"maxZ": 11.899999618530273, "maxX": 76.87100219726562, "maxY": 77.17400360107422, "minX": 18.67799949645996, "minY": 25.990999221801758, "minZ": 0.30000001192092896}, "dimensions": {"width": 58.193002700805664, "depth": 51.18300437927246, "height": 11.599999606609344}, "firstFilament": 0.00195990324770309, "progress": [[0, 1631.7369733578662], [0.00177825367840378, 1625.6364226290104], [0.00195990324770309, 1625.432692984763], [0.07179460213964071, 1565.392221648185], [0.0888744419055996, 1505.0466723290456], [0.1248267158713921, 1444.975875289714], [0.16510989798942607, 1384.5620278251574], [0.20618660191018862, 1324.4176752691426], [0.2400594663326864, 1264.1852057199824], [0.2851133397707391, 1203.425629800228], [0.3257263592646061, 1143.3844687319724], [0.3652303603353825, 1083.2089634107429], [0.4048825492126925, 1022.8878526284198], [0.4467145329215943, 962.6979630298283], [0.48940218170693234, 902.2175872759168], [0.5312341654158341, 842.0410386698475], [0.566435939845311, 781.9437006538296], [0.6051846611279482, 721.6659508703138], [0.6423845808197176, 661.4922477231056], [0.6724284635314588, 600.5491519336738], [0.7109668537338547, 540.4110176542285], [0.7493761771370115, 479.78297318479304], [0.7834450318842797, 419.07787270359677], [0.8102335631041043, 358.3435101680293], [0.8362190120175531, 298.0987180863774], [0.861401378624626, 237.69986426913033], [0.8871956174651281, 177.50287122098644], [0.9227798120404983, 116.95687806624187], [0.949769113836917, 56.540435263838845], [0.9773559471112938, 0.40058182182974633], [0.9777240265017161, 0.0], [1, 0]], "estimatedPrintTime": 1631.7369733578662, "lastFilament": 0.9773559471112938, "filament": {"tool0": {"volume": 7.008950063321779, "length": 991.5635986328125}}}

After that i have updated the plugin with new version.... and everything seams good!!
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4023535/plugin_PrintTimeGenius_engine.log)

I have the gold stars next to my files that have been analyzed

congrats and Thanks a lot ;)

eyal0, 2020-01-05:
Pretty cool.  Okay, I'll soon publish a release with the new windows support.  And then you'll be able to uninstall the custom version that I gave you and install the regular version.

I'll write again when it's ready.

PR #174 (2020-01-05): Add support for Windows
This fixes #111
closed 2020-01-05 by eyal0