eyal0/OctoPrint-PrintTimeGenius #112: Not detecting direct uploads to .uploads through SMB

s1mpleman, 2019-01-03:
I'm guessing this is a similar issue to what I'm having so I'll put it here instead. 

I've now decided to start uploading the print files directly to .uploads through SMB. I've noticed now though I've got to start a manual calculation from the settings to get it to detect files. 

Is this something solvable or do you think it's down to octoprint itself?

_Originally posted by @s1mpleman in https://github.com/eyal0/OctoPrint-PrintTimeGenius/issue_comments#issuecomment-451070081_

eyal0, 2019-01-03:
Probably just OctoPrint. OctoPrint is written in Python and I believe that there are python modules to monitor directories and notice when files are uploaded. I don't know if OctoPrint uses one or not but it would be easy to find out in less than an hour's work. I'll do that.

If it does have that module then we need to find out why it isn't working for you and repair it.

If it doesn't, then a different solution is needed. One solution is to modify OctoPrint to have such a module. Another solution is to have a plug-in that does it. (Maybe that plugin exists already, have you searched for it?) A third solution is for you to upload properly using the OctoPrint API. That might be difficult for you.

I'll check OctoPrint code to see what it is doing. Meanwhile, can you search if some plugin already does it in the plugin repository? Also, can you explain your workflow for me so that we can determine if there is a way for you to do a proper upload through the API? Thanks!

eyal0, 2019-01-03:
BTW, how did you do that "original posted by..." thing? I've never seen that before.

eyal0, 2019-01-03:
https://github.com/foosel/OctoPrint/blob/112a8b9c8bf7f5895f6a29dbe3d98ea4d9e0f400/src/octoprint/server/__init__.py#L687

Looks like you're supposed to upload to a folder called "watched" that is next to the uploads folder. Want to try that?

https://forum.e3d-online.com/threads/setting-up-samba-on-the-raspberry-pi-to-make-file-transfers-simple.1763/

Let me know how it works for you. Good luck!

s1mpleman, 2019-01-04:
Yeah literally just figured that out now man! Thanks :) 

to change a comment into a issue, you use the little drop down after posted, and you can make an issue from that. 

Feel free to close, this works perfectly for me now!

s1mpleman, 2019-01-04:
My work flow was cura upload didn't work with your plugin, and S3D doesn't have an auto upload to octoprint. I don't care about starting a print from the slicer, but having just a click save and it's ready for octoprint is handy. 

So Slice, Save and Print again with no messing about uploading :). I can parse multiple files ready for a long haul print week aha.

closed 2019-01-04 by eyal0