eyal0/OctoPrint-PrintTimeGenius #113: PrintTimeGenius coinciding with UI slowness

Commod0re, 2019-01-05:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.10

PrintTimeGenius Version (if you know): 1.3

What did you try:
- delete many old gcode files
- make sure everything is up to date
- check all possible OctoPrint settings
- reset PrintTimeGenius settings
- reinstall PrintTimeGenius
- disable PrintTimeGenius

What happened: only disabling PrintTimeGenius helped

What did you expect to happen: some improvement

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

here's page load time with PrintTimeGenius enabled:
[image]

and here's page load time with PrintTimeGenius disabled:
[image]

any idea what's going on here? it seems like everything was working fine until recently. I'm not sure what changed, but the UI has been very slow to load lately, and I finally tracked it down to this plugin being loaded. I'll attach the plugin log file here as well in case it can help

[plugin_PrintTimeGenius_engine.log.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2729091/plugin_PrintTimeGenius_engine.log.txt)

eyal0, 2019-01-05:
I do see some errors in your log.  They are caused by PTG having the wrong idea about your printer's extruder acceleration speed.  it seems to think that you are set to max acceleration of 0!  That's impossible.  Can you send the output of the M503 command?  I want to see the M203 output in there.

This is the second time that someone has reported something like this before so it's making me suspicious that there is something flaky in my software.

Commod0re, 2019-01-05:
Sure thing, here's my M503

```
Send: M503Recv: echo:Steps per unit:Recv: echo:  M92 X100.00 Y100.00 Z400.00 E141.50Recv: echo:Maximum feedrates (mm/s):Recv: echo:  M203 X500.00 Y500.00 Z12.00 E120.00Recv: echo:Maximum acceleration (mm/s2):Recv: echo:  M201 X9000 Y9000 Z500 E10000Recv: echo:Acceleration: S=acceleration, T=retract accelerationRecv: echo:  M204 S1000.00 T1500.00Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)Recv: echo:  M205 S0.00 T0.00 B0.00 X10.00 Y10.00 Z0.20 E2.50Recv: echo:Home offset (mm):Recv: echo:  M206 X0.00 Y0.00 Z0.00Recv: echo:PID settings:Recv: echo:   M301 P23.11 I2.18 D61.20Recv: echo:PID heatbed settings:Recv: echo:   M304 P126.13 I4.30 D924.76Recv: echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)Recv: echo:   M207 S3.00 F2700.00 Z0.00Recv: echo:Recover: S=Extra length (mm) F:Speed (mm/m)Recv: echo:   M208 S0.00 F480.00Recv: echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveriesRecv: echo:   M209 S0Recv: echo:Filament settings: Disabled
```

Commod0re, 2019-01-05:
would that really affect those page loads like that, though? I didn't have a print running, and no gcode files were being analyzed by PTG at the time

Commod0re, 2019-02-26:
I deleted some handmade gcodes and octoprint and printtimegenius both started working right again, so I think this was my fault

closed 2019-02-26 by Commod0re