eyal0/OctoPrint-PrintTimeGenius #116: PTG not working on an Orange PI

JoshuaSG, 2019-01-18:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.10

PrintTimeGenius Version (if you know): 1.3

What did you try: Try analyzing a file with PTG

What happenned: Nothing but an error in the log file

What did you expect to happen: A gold star when printing the file

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

`2019-01-18 10:21:01,962 Not running built-in analysis.
2019-01-18 10:21:01,997 Disabled: "/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/octoprint/.octoprint/uploads/Google_Home_Mini_holder_REMIX_P75X3IN_COMMAND_STRIP.gcode"
2019-01-18 10:21:02,000 Running: "/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/octoprint/.octoprint/uploads/Google_Home_Mini_holder_REMIX_P75X3IN_COMMAND_STRIP.gcode" "M92 X80.00 Y80.00 Z400.00 E93.00
M203 X500.00 Y500.00 Z15.00 E50.00
M201 X1000 Y1000 Z100 E5000
M204 P500.00 R1000.00 T500.00
M205 E5.00 S0.00 T0.00 X18.00 Y18.00 Z0.40"
2019-01-18 10:21:02,532 Failed to run '"/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/octoprint/.octoprint/uploads/Google_Home_Mini_holder_REMIX_P75X3IN_COMMAND_STRIP.gcode" "M92 X80.00 Y80.00 Z400.00 E93.00
M203 X500.00 Y500.00 Z15.00 E50.00
M201 X1000 Y1000 Z100 E5000
M204 P500.00 R1000.00 T500.00
M205 E5.00 S0.00 T0.00 X18.00 Y18.00 Z0.40"'
Traceback (most recent call last):
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 314, in _do_analysis
    raise Exception(results_err)
Exception: Running: "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.aarch64" "/home/octoprint/.octoprint/uploads/Google_Home_Mini_holder_REMIX_P75X3IN_COMMAND_STRIP.gcode" "M92 X80.00 Y80.00 Z400.00 E93.00
M203 X500.00 Y500.00 Z15.00 E50.00
M201 X1000 Y1000 Z100 E5000
M204 P500.00 R1000.00 T500.00
M205 E5.00 S0.00 T0.00 X18.00 Y18.00 Z0.40"
Can't find: /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.aarch64

2019-01-18 10:21:02,534 Disabled: "/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/octoprint/.octoprint/uploads/Google_Home_Mini_holder_REMIX_P75X3IN_COMMAND_STRIP.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
2019-01-18 10:21:30,373 Not running built-in analysis.
2019-01-18 10:21:30,376 Disabled: "/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/octoprint/.octoprint/uploads/xyzCalibration_cube.gcode"
2019-01-18 10:21:30,377 Running: "/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/octoprint/.octoprint/uploads/xyzCalibration_cube.gcode" "M92 X80.00 Y80.00 Z400.00 E93.00
M203 X500.00 Y500.00 Z15.00 E50.00
M201 X1000 Y1000 Z100 E5000
M204 P500.00 R1000.00 T500.00
M205 E5.00 S0.00 T0.00 X18.00 Y18.00 Z0.40"
2019-01-18 10:21:30,957 Failed to run '"/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/octoprint/.octoprint/uploads/xyzCalibration_cube.gcode" "M92 X80.00 Y80.00 Z400.00 E93.00
M203 X500.00 Y500.00 Z15.00 E50.00
M201 X1000 Y1000 Z100 E5000
M204 P500.00 R1000.00 T500.00
M205 E5.00 S0.00 T0.00 X18.00 Y18.00 Z0.40"'
Traceback (most recent call last):
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 314, in _do_analysis
    raise Exception(results_err)
Exception: Running: "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.aarch64" "/home/octoprint/.octoprint/uploads/xyzCalibration_cube.gcode" "M92 X80.00 Y80.00 Z400.00 E93.00
M203 X500.00 Y500.00 Z15.00 E50.00
M201 X1000 Y1000 Z100 E5000
M204 P500.00 R1000.00 T500.00
M205 E5.00 S0.00 T0.00 X18.00 Y18.00 Z0.40"
Can't find: /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.aarch64

2019-01-18 10:21:30,958 Disabled: "/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/octoprint/.octoprint/uploads/xyzCalibration_cube.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
2019-01-18 11:56:39,412 Not running built-in analysis.
2019-01-18 11:56:39,416 Disabled: "/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/octoprint/.octoprint/uploads/truck_egg.gcode"
2019-01-18 11:56:39,417 Running: "/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/octoprint/.octoprint/uploads/truck_egg.gcode" "M92 X80.00 Y80.00 Z400.00 E93.00
M203 X500.00 Y500.00 Z15.00 E50.00
M201 X1000 Y1000 Z100 E5000
M204 P500.00 R1000.00 T500.00
M205 E5.00 S0.00 T0.00 X18.00 Y18.00 Z0.40"
2019-01-18 11:56:40,733 Failed to run '"/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/octoprint/.octoprint/uploads/truck_egg.gcode" "M92 X80.00 Y80.00 Z400.00 E93.00
M203 X500.00 Y500.00 Z15.00 E50.00
M201 X1000 Y1000 Z100 E5000
M204 P500.00 R1000.00 T500.00
M205 E5.00 S0.00 T0.00 X18.00 Y18.00 Z0.40"'
Traceback (most recent call last):
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 314, in _do_analysis
    raise Exception(results_err)
Exception: Running: "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.aarch64" "/home/octoprint/.octoprint/uploads/truck_egg.gcode" "M92 X80.00 Y80.00 Z400.00 E93.00
M203 X500.00 Y500.00 Z15.00 E50.00
M201 X1000 Y1000 Z100 E5000
M204 P500.00 R1000.00 T500.00
M205 E5.00 S0.00 T0.00 X18.00 Y18.00 Z0.40"
Can't find: /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.aarch64

2019-01-18 11:56:40,736 Disabled: "/home/octoprint/OctoPrint/venv/bin/python" "/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/octoprint/.octoprint/uploads/truck_egg.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
`

JoshuaSG, 2019-01-18:
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/2774399/plugin_PrintTimeGenius_engine.log)

eyal0, 2019-01-18:
`Can't find: /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.aarch64`

This means that it was looking for the aarch64 file.  That is a weird architecture name, I have never heard of it.  What kind of device are you using for OctoPrint?  Like, Raspberry Pi or what?

It would help if you could run "uname -a" on that device and paste the output here.  Thanks!

JoshuaSG, 2019-01-18:
uname -a output is
`Linux octopi 4.14.70-sunxi64 #274 SMP Wed Sep 19 12:09:30 CEST 2018 aarch64 aarch64 aarch64 GNU/Linux`

It is an [OrangPi Zero+](http://www.orangepi.org/OrangePiZeroPlus/) running [Armbian](https://www.armbian.com/)

After looking around I looked into compiling the Marling Timing Simulator from your repo, but I'm not good with Linux. I attempted `scons aarch64/marling-calc.aarch64` but I get the error `*** Do not know how to make File target 'aarch64/marling-calc.aarch64'`. I also tried just `scons` but get the error `arm-linux-gnueabi-g++: not found`

eyal0, 2019-01-18:
Ah, Orange Pi, it's in the title.  Duh!  Okay, I didn't make it for that architecture!  But, it might be compatible with the RPi version so here's what we'll try:

1. Try to run one of the other ones.  Just type:

"/home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv6l" "/home/octoprint/.octoprint/uploads/Google_Home_Mini_holder_REMIX_P75X3IN_COMMAND_STRIP.gcode" "M92 X80.00 Y80.00 Z400.00 E93.00
M203 X500.00 Y500.00 Z15.00 E50.00
M201 X1000 Y1000 Z100 E5000
M204 P500.00 R1000.00 T500.00
M205 E5.00 S0.00 T0.00 X18.00 Y18.00 Z0.40"

Notice how I used the `armv6l` version in there.  You could also try the `armv7l` one, too.

If that works, cool.  Just copy that file to the new file name in the same directory and we're done.

2. If that doesn't work or if you want to compile a version built specifically for the Orange Pi, you can compile marlin-calc for Orange Pi.  If you're savvy with compiling code then you can try that.  I can walk you through that if you are interested.

eyal0, 2019-01-18:
The scons command that you wrote is correct but you need to add that target to the SConstruct file.  It's pretty easy.  I think that you'll need to make a duplicate of the lines 51 through 63 of this file:

https://github.com/eyal0/Marlin/blob/918fa08ce87cf9dcd9789152bbeeadbb7845dfe4/Marlin/SConstruct#L51

And change the CXX to `g++` if you compile on the OrangePi itself.

If you compile on your desktop, maybe you'll need to run:

`sudo apt-get install g++-7-aarch64-linux-gnu`

If you find that the compiling works and solves your problem, let me know and I'll make aarch64 a permanent part of this project.

JoshuaSG, 2019-01-19:
I was unable to use the armv61 or armv71, gives error `Not executable`.

I'm closer to compiling, but at some point it switches to trying to compile armv6 again. Here is the output

pi@octopi:~/marlin/Marlin/Marlin$ scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
g++ -o aarch64/feature/fwretract.o -c -Wall -static -std=gnu++11 -march=native -mlittle-endian -MMD -O2 src/feature/fwretract.cpp
g++ -o aarch64/module/planner.o -c -Wall -static -std=gnu++11 -march=native -mlittle-endian -MMD -O2 src/module/planner.cpp
g++ -o aarch64/module/calc.o -c -Wall -static -std=gnu++11 -march=native -mlittle-endian -MMD -O2 src/module/calc.cpp
g++ -o aarch64/marlin-calc.aarch64 -Wall -static -std=gnu++11 -march=native -mlittle-endian -O2 aarch64/module/planner.o aarch64/module/calc.o aarch64/feature/fwretract.o
arm-linux-gnueabi-g++ -o armv6l/feature/fwretract.o -c -Wall -static -std=gnu++11 -march=armv6 -mlittle-endian -MMD -O2 src/feature/fwretract.cpp
sh: 1: arm-linux-gnueabi-g++: not found
scons: *** [armv6l/feature/fwretract.o] Error 127
scons: building terminated because of errors.

Here are the lines I added to SConstruct
env = Environment(
    CXX='g++',
    CPPFLAGS= '-Wall -static -std=gnu++11 -march=native -mlittle-endian -MMD ' + debug_flags,
    LINKFLAGS='-Wall -static -std=gnu++11 -march=native -mlittle-endian ' + debug_flags)
env.VariantDir('aarch64', 'src', duplicate=0)
add_program(env, 'aarch64')

eyal0, 2019-01-19:
`scons` when run without arguments will attempt to build every possible version.  But you just want the aarch64 one.  You should run:

`scons aarch64/marlin-calc`

I think.

eyal0, 2019-01-19:
I tried bulding it but I don't have an OrangePi for testing.  Want to try?

You can just install this in the install plugins dialog of OctoPrint:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/aarch64.zip

Then try to upload a new file and see if it gets analyzed.  Check the log if it finds the right file to run and runs it.

If you can confirm for me that it works, then I'll put it in the next release.

JoshuaSG, 2019-01-22:
Thanks, everything looks good with the aarch64 version you uploaded! I get the gold star now and the log file looks good.

RX14, 2019-02-17:
I tested this on a 64bit raspi3 and it works well, seems like this could be merged

RX14, 2019-07-17:
Please could the aarch64 branch be merged?

eyal0, 2019-07-18:
@RX14 Did you send a PR?

eyal0, 2019-07-18:
@RX14 Ah, I see that I have the aarch64 branch and it isn't merged.  Maybe I was waiting for more confirmation that it works?  I don't remember.  Anyway, no harm in putting it in.

Can you check the output of this command on your device:

```
python -c 'import platform; print platform.machine()'
```

Just type that in and confirm that the output is aarch64.  Thanks.

eyal0, 2019-07-18:
Fixed in version 1.3.4

closed 2019-07-18 by eyal0