eyal0/OctoPrint-PrintTimeGenius #119: Analyse all files in a folder

Ruko2010, 2019-02-21:
OctoPrint Version:
0.16.0

PrintTimeGenius Version (if you know):
1.3 (I think)

What did you try:
Uploaded files *not* using the file Upload by octoprint.

What happenned:
Of cause, no analysation of my files.

What did you expect to happen:
time analysation of my uploaded files.

Feature Request:
I'm using the Gitfiles Plugin to transfer the gcode files to my octopi, and of cause, they are not analysed, because they were not uploaded with the octoprint upload dialog.
My request is now to have a button "Analyse All", which analyses all gcode-files in the current folder that has not been analysed so far.
Is it possible to implement such a feature?

eyal0, 2019-02-21:
Gitfiles isn't properly updating files.

"uploads" is the folder where all the gcode files are put.  But users aren't supposed to copy stuff directly into there.  The right way to do it is to put files into the "watched" folder.  OctoPrint monitors that folder and moves stuff from there into uploads.

Because Gitfiles is putting the files into uploads by itself, OctoPrint isn't quite finding out about it properly so PTG isn't getting the list.  If @OutsourcedGuru wanted to repair this, maybe it would work to have git fetch all the gcode into a temporary directory and then compare that directory with the contents of the "uploads" directory.  For all files that are not matching, copy them from the temporary directory into the "watched" directory.  Then let OctoPrint find the files and process them correctly.

I think that this would solve the problem.  Maybe it would be possible to do it without a temporary directory, too, by just comparing file hashes.  Up to Gitfiles to figure this out.

Ruko2010, 2019-02-21:
Thanks @eyal0 for this explanation. I can change the gitfolder to another one, can you give me the path to the "watched" folder and I will try if it is working, when I put the home directory of my gitfiles to this watched folder.
Maybe I can also write a little script to copy the files from the gitfolder to the watchfolder, but I need the path to the watch folder 😃.

eyal0, 2019-02-22:
It's literally called "watched".  It's next to the uploads folder.  On OctoPi, maybe it's ~/.octoprint/watched .  Any file that you add into there should be treated as if it is uploaded.

I don't know if it will work well together with Octoprint-Gitfiles.  That why I suggested that @OutsourcedGuru consider adjusting Gitfiles to use it.

OutsourcedGuru, 2019-02-22:
I'll take a look at it. And there could be another way of doing this as well. I think foosel herself admits that there's a bug in the watched folder code at the moment so...

Ruko2010, 2019-02-22:
Thanks @eyal0 and @OutsourcedGuru for this Cross Plugin Help 😃. I like the PTG Plugin a lot, because it has a good time estimate, but I also need the gitfiles plugin. I slice my files on my working laptop, that stays at work most of the time and at home I mostly use my ipad to start prints via the octopi.
So getting both plugins to work together would be awesome.

OutsourcedGuru, 2019-02-22:
I myself use PTG on my production printer; I've just been coding lately on a test rig so I wouldn't have noticed otherwise.

OutsourcedGuru, 2019-02-24:
Just wondering, does PTG support dragging/dropping two files at once and does it analyze both or just one?

OutsourcedGuru, 2019-02-24:
In my mind, deleting the particular folder's `.metadata.json` file should result in OctoPrint—upon startup—rebuilding that and doing the analysis _for all files in that folder_.  It's clear that OctoPrint is adding those to the analysis queue as seen here:

```
2019-02-23 15:59:13,134 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2019-02-23 15:59:13,161 - octoprint.filemanager.analysis - INFO - Starting analysis of local:gitfiles/PLA/BatteryTool.gcode
2019-02-23 15:59:13,197 - octoprint.plugins.PrintTimeGenius - INFO - Not running built-in analysis.
2019-02-23 15:59:13,207 - octoprint.plugins.PrintTimeGenius - INFO - Disabled: "/Users/me/sites/OctoPrint/venv/bin/python2.7" "/Users/me/sites/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/Users/me/Library/Application Support/OctoPrint/uploads/gitfiles/PLA/BatteryTool.gcode"
2019-02-23 15:59:13,208 - octoprint.plugins.PrintTimeGenius - INFO - Running: "/Users/me/sites/OctoPrint/venv/bin/python2.7" "/Users/me/sites/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/Users/me/Library/Application Support/OctoPrint/uploads/gitfiles/PLA/BatteryTool.gcode" ""
2019-02-23 15:59:13,209 - octoprint.filemanager - INFO - Added 4 items from storage type "local" to analysis queue
2019-02-23 15:59:13,724 - octoprint.filemanager.analysis - INFO - Analysis of entry local:gitfiles/PLA/BatteryTool.gcode finished, needed 0.56s
2019-02-23 15:59:14,728 - octoprint.filemanager.analysis - INFO - Starting analysis of local:gitfiles/PLA/5milCamRing.gcode
2019-02-23 15:59:15,259 - octoprint.filemanager.analysis - INFO - Analysis of entry local:gitfiles/PLA/5milCamRing.gcode finished, needed 0.53s
2019-02-23 15:59:16,266 - octoprint.filemanager.analysis - INFO - Starting analysis of local:gitfiles/PLA/Braille_a-z.gcode
2019-02-23 15:59:17,322 - octoprint.filemanager.analysis - INFO - Analysis of entry local:gitfiles/PLA/Braille_a-z.gcode finished, needed 1.06s
2019-02-23 15:59:18,336 - octoprint.filemanager.analysis - INFO - Starting analysis of local:gitfiles/PLA/microSD-clip.gcode
2019-02-23 15:59:18,882 - octoprint.filemanager.analysis - INFO - Analysis of entry local:gitfiles/PLA/microSD-clip.gcode finished, needed 0.55s
```

Unfortunately it only results in a single file being processed.
[image]

Disabling PTG, deleting the `.metadata.json` file and restarting OctoPrint again:

```
2019-02-23 16:57:15,363 - octoprint.filemanager.analysis - INFO - Analysis of entry local:gitfiles/PLA/BatteryTool.gcode finished, needed 1.96s
2019-02-23 16:57:16,369 - octoprint.filemanager.analysis - INFO - Starting analysis of local:gitfiles/PLA/5milCamRing.gcode
2019-02-23 16:57:16,370 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /Users/me/sites/OctoPrint/venv/bin/python2.7 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.01 --throttle-lines=100 /Users/me/Library/Application Support/OctoPrint/uploads/gitfiles/PLA/5milCamRing.gcode
2019-02-23 16:57:21,837 - octoprint.filemanager.analysis - INFO - Analysis of entry local:gitfiles/PLA/5milCamRing.gcode finished, needed 5.47s
2019-02-23 16:57:22,846 - octoprint.filemanager.analysis - INFO - Starting analysis of local:gitfiles/PLA/Braille_a-z.gcode
2019-02-23 16:57:22,848 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /Users/me/sites/OctoPrint/venv/bin/python2.7 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.01 --throttle-lines=100 /Users/me/Library/Application Support/OctoPrint/uploads/gitfiles/PLA/Braille_a-z.gcode
2019-02-23 16:57:29,758 - octoprint.filemanager.analysis - INFO - Analysis of entry local:gitfiles/PLA/Braille_a-z.gcode finished, needed 6.91s
2019-02-23 16:57:30,764 - octoprint.filemanager.analysis - INFO - Starting analysis of local:gitfiles/PLA/microSD-clip.gcode
2019-02-23 16:57:30,766 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /Users/me/sites/OctoPrint/venv/bin/python2.7 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.01 --throttle-lines=100 /Users/me/Library/Application Support/OctoPrint/uploads/gitfiles/PLA/microSD-clip.gcode
2019-02-23 16:57:32,130 - octoprint.filemanager.analysis - INFO - Analysis of entry local:gitfiles/PLA/microSD-clip.gcode finished, needed 1.37s
```

[image]

So OctoPrint by itself was able to rebuild the `.metadata.json` file with all four files rather than just a single one.

OutsourcedGuru, 2019-02-24:
Re-enable PTG, delete a single paragraph from `.metadata.json`, restart OctoPrint and PTG is able to do its analysis for this one so now I have two stars.

OutsourcedGuru, 2019-02-24:
I'm thinking that your code assumes that only one instance of itself can run at a time. Maybe you need a mutex on that [analyze_gcode_comments.py](https://github.com/eyal0/OctoPrint-PrintTimeGenius/blob/master/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py) somewhere around Line 227:

```
return json.dumps(analysis)
```

If four of these are running at once (in the current situation), then all four are stepping on each others' toes and just blindly overcopying the same `.metdata.json` file. In my case this is the Braille file since it's the largest of the four (think "reverse race condition" as in "the slowest writes history").

Does that make sense?

OutsourcedGuru, 2019-02-24:
I've been trying to exercise the Blueprint plugin endpoint and it doesn't seem to be consistent. I called it from Postman having set X-Api-Key and Content_Type as expected. It worked once but doesn't seem to be behaving for any of the other files.

http://localhost:5000/plugin/PrintTimeGenius/analyze/local/gitfiles/PLA/microSD-clip.gcode

OutsourcedGuru, 2019-03-01:
@Ruko2010 What happens if you adjust your Settings -> Folders -> Watched Folder to `/home/pi/.octoprint/uploads/gitfiles`?

[image]

Ruko2010, 2019-03-05:
Sorry for the late response, but I was on vaction. I'll try it at home in the evening today. But I'm pretty sure, when I'll get the latest git pull, the files will be added to the watched folder and then beeing copied to the uploads folder. I asume they will be deleted from the watched folder, so doing another pull will either result in merge conflicts, or will add the files to the watched folder again. But I'll try it and give you a propper response in the evening 😉

OutsourcedGuru, 2019-03-05:
I would agree with the first part of your reasoning "they will be deleted from the watched folder (moved to `uploads`". Okay, so now the `uploads/gitfiles` folder will be missing your gcode files. When you `git pull` again it may just bring the full set of files in again and not complain. It's worth a test.

Ruko2010, 2019-03-05:
Ok, not what I expected...I tryed to left out gitfiles for the test, so here is what I've done.

1. set watched folder to ~/.octoprint/uploads/gitfiles
2. Cloned my repo into ~/.octoprint/uploads/gitfiles
3. added a file to the repo from my desktop pc and pushed it to master
4. octoprint is copying the new files from watched to upload and is running the PTG script
5. octoprint is deleting the files from the ~/.octoprint/uploads/gitfiles folder and no mergeconflicts after deleting the files from the repo (from the desktop).
6. Adding more files to the repo is working fine!
7. Even adding files with the original upload button is working!

So, everything is working like expected 😃. Thanks for the help!

closed 2019-03-05 by Ruko2010

OutsourcedGuru, 2019-03-06:
Glad to hear it. As I was exercising everything with mine and a fork of PTG running at the same time, this idea came to me as a workable solution without adding any coding otherwise.
👍