eyal0/OctoPrint-PrintTimeGenius #123: [Feature Request] Manual Analysis on Folder / Automated Gap Scan

jefft8910, 2019-04-21:
I'm leveraging WinSCP scripts to keep a local folder synchronized with multiple instances of OctoPrint.  Unfortunately, when sideloading content outside of the Web UI or API set, PTG doesn't appear to trigger analysis on uploaded files.

While you can go into the plugin page and trigger analysis on individual files successfully, this becomes burdensome when performing large scale uploads, or renaming directories - and the file list can get pretty long, especially when printing a large catalog commercially.

I'd like to request either a configuration flag that allows automated scanning for files that have not been analyzed, or a manual trigger on a directory (+ subdirectories, optionally?) so PTG can "catch up" on missing analysis.

This behavior may need to be throttled to a configurable max concurrency, obviously.

eyal0, 2019-07-09:
I think that I fixed it in the latest version.  Can you try to install this: https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip (from the normal OctoPrint install dialog, you can put a URL).

New features:
* You can select multiple items to analyze.
* They are sorted so that the unanalyzed are at the top and then alphabetically.
* I fixed the button, it was broken.

eyal0, 2019-07-09:
After you comment on the behavior, I'll make a new version.

jefft8910, 2019-07-09:
Thanks!  The batch analyze is a great feature but I'm still having trouble with it.  Either selecting things and clicking analyze is doing nothing, or I have items that aren't analyzed but aren't showing up in the list.

How do you determine eligible files for analysis?  I feel like I need to re-trigger that somehow and try again.

eyal0, 2019-07-09:
It's the same list that OctoPrint uses for showing the list of files.

Are your files stored in sdcard or octopi? I think that sdcard files can't be analyzed at all.

jefft8910, 2019-07-09:
Should be the octopi - I'm not trying to analyze the SD cards.  Some of the stuff works, some doesn't.  I'm having spotty issues on two Octoprint instances.  Happy to provide whatever debug logs are necessary.  Here's an example of the remote sync script:

#Resources Can be Found here - https://winscp.net/eng/docs/scripting

option batch abort
option confirm off
open sftp://pi:raspberry@x.x.x.x -hostkey="ssh-ed25519 256 <<hash>>"
synchronize remote "C:\Users\<<path>>\Dropbox (Personal)\!SDCardFiles" /home/pi/.octoprint/uploads -delete
exit

eyal0, 2019-07-09:
Is that script to uplaod files into the upload directory?  You should not do that.  You should upload into the watched directory:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/119#issuecomment-466131587

Maybe that will work better for you?