eyal0/OctoPrint-PrintTimeGenius #125: Gold star is not showing, instead shows "â˜…"

Brain-Gamer, 2019-05-19:
OctoPrint Version: 1.3.11

PrintTimeGenius Version (if you know): 1.3

What did you try:
Updated Octoprint to the newest version

What happenned:
The golden star which displays if GCODE has been analyzed by PTG is not showing but instead it shows "â˜…"

What did you expect to happen:
The golden star should still be displayed as usual

---

I don't know if this is a problem caused by PTG or if some change in the Octoprint Core is the cause of this?
After a quick look in the inspector it seems that the css class `print-time-genius:before` has a wrong value for the content attribute.

---

[image]

eyal0, 2019-05-20:
The gold star is unicode so maybe the unicode support for the font that you are using is not complete?  It used to work okay?

Brain-Gamer, 2019-05-20:
Before updating Octoprint itself the gold star was displayed without any issues since I installed PTG.

I just used a different machine to view Octoprint and there I can see the gold star without any issues. Maybe it will be fixed if I clean the cache of the browser on my other machine.

Brain-Gamer, 2019-05-20:
Yeah it seems to have fixed itself and displays as it did before. I'll close this now

closed 2019-05-20 by Brain-Gamer