eyal0/OctoPrint-PrintTimeGenius #127: Strange values for print time

csetera, 2019-05-29:
OctoPrint Version: 1.3.11

PrintTimeGenius Version (if you know):  1.3

What did you try:  Printing a couple of different prints. The GCODE for the prints that are acting up did not always show this kind of behavior, but I'm not sure what triggered this to happen.  Latest Octoprint?

What happened:  See the attached screenshot.
[image]

What did you expect to happen:  Semi-valid and non-negative percentages :-)

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

eyal0, 2019-05-29:
Please check the log files for errors or just send me the log file and I'll read them.

csetera, 2019-05-29:
There are definitely some exceptions in the log file.  However, rather than trying to guess what might be helpful, I thought it better to just send you the entire log file.

[plugin_PrintTimeGenius_engine.log.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3234043/plugin_PrintTimeGenius_engine.log.zip)

eyal0, 2019-06-03:
Possible duplicate of #128 ?  Did you slice using Cura latest?

csetera, 2019-06-03:
I believe it is unrelated.  These were GCODE files sliced with Cura 4.0 that I've used before.  It is just recently they started showing strange print time values with the same GCODE.

eyal0, 2019-06-04:
Perhaps you can attach the gcode that failed anyway, just to be sure?  Thanks.

csetera, 2019-06-04:
My Ubuntu server running Octoprint has decided not to recognize my printer at the moment.  Once I get that sorted, I will find a GCODE file that is misbehaving and attach it.

csetera, 2019-06-06:
[CCR10_SmartTemperatureTower_PLA_180-225.gcode.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3262539/CCR10_SmartTemperatureTower_PLA_180-225.gcode.zip)
This is printing right now with the strange timing and percentages noted by this ticket.

eyal0, 2019-06-07:
Could you send the log again?  Would you like to try the new version of PTG that I linked in #128 ?

eyal0, 2019-06-07:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/128#issuecomment-498686690

csetera, 2019-06-07:
I will try the release from #128 and see if that helps.  I will capture the log if it does not.  I don't think that it is related to the Cura 4.1 issue, as I ran into that and downgraded immediately since it made my MKS Gen L board hang with the feedrate.

csetera, 2019-06-07:
I'm still seeing the same behavior with the plugin from #128 for the same GCODE file.  I cleared the logs before starting and after running the print, the PTG log file is empty.  I did cancel the print relatively early since I didn't care about the result, but would there be some kind of delay in logging that could lead to an empty log file?

eyal0, 2019-06-07:
PTG writes to the log when it's doing its math, not when you're printing.  Maybe you'd need to remove the file and upload it again?  Just to be safe, I would even rename it a little before uploading.  Then PTG will start logging.

PTG starts a new log file each day.  Maybe deleting the log file while it's running has confused it?

eyal0, 2019-06-07:
I tried your file and it worked okay for me:

```
Analysis: {"estimatedPrintTime": 7211.68256179539639561, "printingArea": {"maxX": 179.72999999999998977, "maxY": 170.46600000000000819, "maxZ": 101.40000152587890625, "minX": 124.10399999999999920, "minY": 129.53899999999998727, "minZ": 0.20000000298023224}, "dimensions": {"width": 55.62599999999999056, "depth": 40.92700000000002092, "height": 101.20000152289867401}, "filament": {"tool0": {"length": 5465.47175455469732697, "volume": 38.63314437821615854}}}
```

7211 seconds estimated print time.

csetera, 2019-06-07:
[plugin_PrintTimeGenius_engine.log.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3266300/plugin_PrintTimeGenius_engine.log.zip)

I renamed the file and re-uploaded it and I'm still seeing the behavior.  Looking at the logs, there are some interesting things.  The merged result shows:

```u'estimatedPrintTime': 7163.3821014960495, u'lastFilament': 0.9996319992144281, ```

While the compensated result shows:
``` u'estimatedPrintTime': -40722.13917365838, u'lastFilament': 0.9996319992144281, ```

I think this is due to:
```2019-06-07 09:09:50,607 Time spent printing, actual vs predicted: 3932.51691122/3175.62835113, 8037.97892787/6174.45385359, 1234.49925849/864.325571116, 0.0/36.0878445475, -246.29988162/7.54187632979```
```2019-06-07 09:09:50,607 Average scaling factor: -5.73784089862```

While I don't know exactly where the negative scaling factor comes from, I wonder if this is due to prints that get canceled?  If I have a predicted long-running print that fails early and I cancel it, does that value get out of whack over time?  Clearly only a guess/theory.  If you point me in the right direction, it would be interesting to clear those statistics and try again to see how that changes things.

csetera, 2019-06-24:
Just another update on this issue.  I was having normal/reasonable values being calculated again for a bit of time.  Just now, it has started going negative again.  Interestingly, this seemed to be right after I was forced to cancel a print of the same file soon after it started.  It does seem like there is something going on with the plugin when dealing with canceled jobs.

eyal0, 2019-06-24:
Ah, okay, it might be the compensation.  I don't remember if the compensation knows to ignore the cancelled print.  In the PTG options, you can remove the cancelled prints.  Only the 5 most recent prints are saved.  I don't remember if it saves all prints or just the successful ones.  Can you look at the compensation table in the PTG options and check?

csetera, 2019-06-24:
These numbers seem reasonable.  Is this what you are suggesting I delete from?

[image]

eyal0, 2019-06-27:
Yes, I agree that the compensation values look reasonable.  Hmm...  I can't see where that negative value came from.  The values are stored in the .octoprint directory in the yaml file.  The negative numbers are definitely a problem but I don't understand how they happened in the first place.

csetera, 2019-06-29:
In theory, if I see that issue again, would deleting the compensation values clear things up?  If so, that is good enough for me to get by.

eyal0, 2019-07-03:
Yes.  Delete them.  Don't forget to save after you delete them.

lmilekic, 2022-02-25:
I'm also seeing this issue. My prints all look like this:
[image]
I already tried deleting all the compensation values but that doesn't seem to have fixed it. What else can I try/look at?

lmilekic, 2022-02-25:
Actually PrintTimeGenius might be innocent here. In the logs I see the merged result still has a positive print time estimate. Still open to ideas on where to look though.

eyal0, 2022-02-25:
I see a little green circle where it says print time left.  Usually it's a gold star when the print time genius is enabled.  If you hover your mouse cursor over the green circle, does it say that the estimate is from print time genius or does it say that the estimate is from somewhere else?

lmilekic, 2022-02-27:
Interesting, it says
[image]
In the files list the file has a gold star next to it.

lmilekic, 2022-03-01:
Yep, though I have also seen this on files I have not printed before. Here's something I'm printing right now that I haven't printed before:
[image]

I'm using Prusa Slicer, which outputs a normal time as its time estimate. Not sure if that could be related.

eyal0, 2022-03-01:
Can you provide the log file or look by yourself to see if anything weird is going on in there?

lmilekic, 2022-03-01:
Sure thing, I took a look at octoprint.log and didn't see anything too interesting. There was some exception when POSTing to spaghetti detective, but that seems unlikely to be related.
[octoprint.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8164133/octoprint.log)
Other than that nothing jumped out. Any ideas on where else to look?

eyal0, 2022-03-01:
Each plugin can have its own log and PrintTimeGenius does have a log.  Perhaps look in that log?

The snippet of log that you posted looks reasonable and good.

lmilekic, 2022-03-01:
Pretty sure I found the issue and I am just a doofus. When I was first setting up octoprint I installed this plugin: https://github.com/kanocz/octopi_eta_override. Once I uninstalled it I'm seeing normal things. Gold stars showing up and everything :)
[image]

eyal0, 2022-03-01:
Good to know!  I see that the ETA plugin is making a PrintTimeEstimator, which is something that PTG also does.  So probably they were interfering with one another.

Happy printing!