eyal0/OctoPrint-PrintTimeGenius #128: Error while trying to analyze a file

extesy, 2019-05-30:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version:
1.3.11

PrintTimeGenius Version (if you know):
1.3

What did you try:
Uploaded a file, clicked "analyze" in settings and downloaded a log

What happened:
```
2019-05-29 19:41:00,680 Not running built-in analysis.
2019-05-29 19:41:00,684 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Flexi-Dragon_Body.gcode"
2019-05-29 19:41:00,685 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Flexi-Dragon_Body.gcode" "M200 D3.00
M200 D0
M92 X80.30 Y80.30 Z400.50 E129.30
M203 X300.00 Y300.00 Z5.00 E25.00
M201 Z100 E1000
M204 P800.00 R800.00 T800.00
M205 B20000 E1.00 S0.00 T0.00 X7.00 Y7.00 Z0.40"
2019-05-29 19:41:14,754 Failed to run '"/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Flexi-Dragon_Body.gcode" "M200 D3.00
M200 D0
M92 X80.30 Y80.30 Z400.50 E129.30
M203 X300.00 Y300.00 Z5.00 E25.00
M201 Z100 E1000
M204 P800.00 R800.00 T800.00
M205 B20000 E1.00 S0.00 T0.00 X7.00 Y7.00 Z0.40"'
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 318, in _do_analysis
    raise Exception(results_err)
Exception: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Flexi-Dragon_Body.gcode" "M200 D3.00
M200 D0
M92 X80.30 Y80.30 Z400.50 E129.30
M203 X300.00 Y300.00 Z5.00 E25.00
M201 Z100 E1000
M204 P800.00 R800.00 T800.00
M205 B20000 E1.00 S0.00 T0.00 X7.00 Y7.00 Z0.40"
...Processed 369701 Gcodes and 123 Mcodes. 342511 blocks
Total time: inf
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 90, in <module>
    main()
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 63, in main
    result.update(json.loads(line))
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

2019-05-29 19:41:14,756 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Flexi-Dragon_Body.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
```

What did you expect to happen:
Successful analysis.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
gcode: [Flexi-Dragon_Body.gcode.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3235244/Flexi-Dragon_Body.gcode.zip)

HimbeersaftLP, 2019-05-30:
Can confirm this

hellofaduck, 2019-05-30:
Same error, I think this happens when I updated to latest OctoPrint recently, but not sure 100%
How to fix that?

remy561, 2019-05-31:
I also have this issue.
Since that first crash the log isn't updated anymore and stays empty after a deletion.

eyal0, 2019-05-31:
Found this in the gcode:

```
G1 F17987547480000 Z0.9
```

That feedrate is super high.  Anyone know why it's in the file like that?

extesy, 2019-05-31:
This is because of current bug in Cura: https://github.com/Ultimaker/Cura/issues/5753

eyal0, 2019-06-03:
Okay.  I should still be able to work around this by handling a bad feedrate.

eyal0, 2019-06-04:
I have a fix ready.  Can someone try this?

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/cura4.1.zip

You can install that by copying that link and pasting it into the plugins page in OctoPrint.

After someone confirms that it has solved the problem, I'll make a proper release with this fix in it.  Thanks.

remy561, 2019-06-08:
Tested your fix, seems to work just fine.
Thanks! :)

CDrawe, 2019-06-17:
My test was unsuccessfully. No changes at all.

vadsh, 2019-06-19:
> I have a fix ready. Can someone try this?
>
> https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/cura4.1.zip
>
> You can install that by copying that link and pasting it into the plugins page in [...]

No luck with this.
Here is log:
2019-06-19 08:37:54,002 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/surfTest06.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
2019-06-19 09:05:57,780 Not running built-in analysis.
2019-06-19 09:05:57,783 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/surfTest07.gcode"
2019-06-19 09:05:57,785 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/surfTest07.gcode" "M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z400.00 E420.00
M203 X100.00 Y100.00 Z0.00 E25.00
M201 X5000.00 Y5000.00 Z70.00 E5000.00
M204 S500.00 P4000.00 R3000.00 T4000.00
M205 B50000.00 E5.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 J0.05"
2019-06-19 09:06:00,761 Failed to run '"/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/surfTest07.gcode" "M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z400.00 E420.00
M203 X100.00 Y100.00 Z0.00 E25.00
M201 X5000.00 Y5000.00 Z70.00 E5000.00
M204 S500.00 P4000.00 R3000.00 T4000.00
M205 B50000.00 E5.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 J0.05"'
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 318, in _do_analysis
    raise Exception(results_err)
Exception: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/surfTest07.gcode" "M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z400.00 E420.00
M203 X100.00 Y100.00 Z0.00 E25.00
M201 X5000.00 Y5000.00 Z70.00 E5000.00
M204 S500.00 P4000.00 R3000.00 T4000.00
M205 B50000.00 E5.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 J0.05"
Junction deviation enabled
Processed 740 Gcodes and 30 Mcodes. 751 blocks
Total time: inf
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 90, in <module>
    main()
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 63, in main
    result.update(json.loads(line))
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

2019-06-19 09:06:00,764 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/surfTest07.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining

eyal0, 2019-06-19:
Bummer! Upload that gcode and I'll try it here.

CDrawe, 2019-06-20:
May be it helps...

Here is my log and the gcode file as well.

019-06-20 02:35:31,552 Not running built-in analysis.
2019-06-20 02:35:31,554 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Tools/Druckerzubehor/AC_Chiron_Light_System_Flat.gcode"
2019-06-20 02:35:31,556 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Tools/Druckerzubehor/AC_Chiron_Light_System_Flat.gcode" "M220 S50
M92 X80.00 Y100.00 Z400.00 E400.99
M203 X100.00 Y100.00 Z0.00 E80.00
M201 X350 Y350 Z50 E30000
M204 P4000.00 R3000.00 T4000.00
M205 B20000 E20.00 S0.00 T0.00 X20.00 Y20.00 Z0.40
M200 D1.75
M200 D0"
2019-06-20 02:35:34,593 Failed to run '"/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Tools/Druckerzubehor/AC_Chiron_Light_System_Flat.gcode" "M220 S50
M92 X80.00 Y100.00 Z400.00 E400.99
M203 X100.00 Y100.00 Z0.00 E80.00
M201 X350 Y350 Z50 E30000
M204 P4000.00 R3000.00 T4000.00
M205 B20000 E20.00 S0.00 T0.00 X20.00 Y20.00 Z0.40
M200 D1.75
M200 D0"'
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 318, in _do_analysis
    raise Exception(results_err)
Exception: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Tools/Druckerzubehor/AC_Chiron_Light_System_Flat.gcode" "M220 S50
M92 X80.00 Y100.00 Z400.00 E400.99
M203 X100.00 Y100.00 Z0.00 E80.00
M201 X350 Y350 Z50 E30000
M204 P4000.00 R3000.00 T4000.00
M205 B20000 E20.00 S0.00 T0.00 X20.00 Y20.00 Z0.40
M200 D1.75
M200 D0"
.Processed 67171 Gcodes and 65651 Mcodes. 130803 blocks
Total time: inf
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 90, in <module>
    main()
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 63, in main
    result.update(json.loads(line))
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

2019-06-20 02:35:34,595 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Tools/Druckerzubehor/AC_Chiron_Light_System_Flat.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining

[AC_Chiron_Light_System_Flat.gcode.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3308361/AC_Chiron_Light_System_Flat.gcode.zip)

vadsh, 2019-06-20:
> Bummer! Upload that gcode and I'll try it here.

Here is gcode.
https://yadi.sk/i/iiqy1o6F7kWm0w

eyal0, 2019-06-20:
I am able to see the same problem as you.  It seems that you have the line `M203 X100.00 Y100.00 Z0.00 E80.00` in your log.  Do you see that?

According to [this page](http://marlinfw.org/docs/gcode/M203.html) it means that you have set the maximum z speed of your printer to 0.  This is causing the calculated print time to be infinity.

The M503 command can output the actual settings on your printer.  Would you like to run that and send me the output?  Let's see if Z is really set to 0 in there.

vadsh, 2019-06-21:
> The M503 command can output the actual settings on your printer. Would you like to run that and send me the output? Let's see if Z is really set to 0 in there.

The problem was found. It is due to Cura.

Latest Cura has this weird bug with huge Z speed in every Gcode [https://github.com/Ultimaker/Cura/issues/5753](url)

Proposed solution was to set Maximum Z speed setting. Which I did. But this lead to these two lines in Gcode
M203 Z20 - at the beginning of gcode
M203 Z299792458000 - at the end of gcode

And this leads to such M503 result
M203 X300.00 Y300.00 Zovf E30.00
Hence the problem

Thanks @eyal0

eyal0, 2019-06-24:
I know about this one and I put a workaround into PTG to protect against that huge Cura default.  It was fixed in #130 .

closed 2019-06-24 by eyal0

Poeschl, 2019-07-17:
Sry, to warm this one up again, but still getting the the same error on latest version of Cura (4.1.0) and your plugin (1.3.3).

eyal0, 2019-07-18:
@Poeschl Can you open a new bug and fill out all the proper info, like logs and version and the rest?  I can't really fix it without that info.