eyal0/OctoPrint-PrintTimeGenius #129: marlin-calc produces estimate of 'inf'

zaventh, 2019-06-03:
When running `marlin-calc` on the attached gcode file, the result for the estimated print time is `inf`:

```bash
$ ./marlin-calc.x86_64 ~/enclosure_window_parts_calc.gcode
...
Progress: 0.99972918233508801, 2637.76028151021728263, inf
Processed 136171 Gcodes and 24 Mcodes. 133906 blocks
Total time: inf
Analysis: {"estimatedPrintTime": inf, "printingArea": {"maxX": 151.62999999999999545, "maxY": 151.40100000000001046, "maxZ": 5.00000000000000000, "minX": 61.64499664306640625, "minY": 66.36899999999999977, "minZ": 0.30000001192092896}, "dimensions": {"width": 89.98500335693358920, "depth": 85.03200000000001069, "height": 4.69999998807907104}, "filament": {"tool0": {"length": 2632.76028151021728263, "volume": 18.60988632574211366}}}
```

Is this normal? Other gcode files seem to be processed correctly.

Sliced with Cura 4.1 and using the master branch of this repo.

[enclosure_window_parts_calc.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3248983/enclosure_window_parts_calc.zip)

eyal0, 2019-06-03:
Can you check if this is a duplicate of #128 ?  Thanks.

zaventh, 2019-06-03:
Sorry yes, this appears to be related to that Cura bug. Setting a max z rate of 5 mm/s resolves the issue.

closed 2019-06-03 by zaventh