eyal0/OctoPrint-PrintTimeGenius #131: [bug] Analyze Dropdown in settings isn't sorted

TheKrush, 2019-06-04:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.11

PrintTimeGenius Version (if you know): 1.3

What did you try: No options to change sorting

What happened: The dropdown appears to be sorted based on folders, but the files within that folder are not sorted in alphabetical order.

What did you expect to happen: That the dropdown be completely sorted, to aid in finding specific files.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

eyal0, 2019-06-05:
I think that I would need to add sorting here:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/blob/master/octoprint_PrintTimeGenius/static/js/PrintTimeGenius.js#L74

There is no sorting currently.  I think that your idea is good.

closed 2019-07-09 by eyal0

eyal0, 2019-07-09:
I think that I fixed it in the latest version.  Can you try to install this: https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip (from the normal OctoPrint install dialog, you can put a URL).

New features:
* You can select multiple items to analyze.
* They are sorted so that the unanalyzed are at the top and then alphabetically.
* I fixed the button, it was broken.

eyal0, 2019-07-09:
After you comment on the behavior, I'll make a new version.

PR #139 (2019-07-09): Sort file list in settings
This fixes #131